JMRI/JMRI #10: Git support in NetBeans 8.0.2

bobjacobsen, 2015-09-23:
Having trouble with the Git support in NetBeans 8.0.2.

JMRI GitHub users should be set up like:

    % git remote -v
    origin	https://github.com/JMRI/JMRI (fetch)
    origin	https://github.com/bobjacobsen/JMRI (push)

with split fetch/pull and push.  This appears in the git .config (which NetBeans shows as config) as

    [branch "master"]
    [remote "origin"]
    	url = https://github.com/JMRI/JMRI
    	fetch = +refs/heads/*:refs/remotes/origin/*
    	pushurl = https://github.com/username/JMRI

NetBeans seems to only store the "url" term, It also seems to insist on putting the user name in (though I don't know if that's a problem specifically), and in the process rewrites the .config:

    [branch "master"]
    [remote "origin"]
    	url = https://bobjacobsen@github.com/bobjacobsen/JMRI
    	url = https://bobjacobsen@github.com/JMRI/JMRI
    	pushurl = https://github.com/bobjacobsen/JMRI
    	fetch = +refs/heads/*:refs/remotes/origin/*

The work-around is to always use the full "Pull..." or "Push..." and fill out the form, but that's a pain.

bobjacobsen, 2015-09-23:
This workaround works just once, and the gets overwritten:
* Quit NetBeads
* On the command line, go to the directory you just created and do:
   * git remote set-url origin https://github.com/JMRI/JMRI
   * git remote set-url --push origin https://github.com/username/JMRI
where "username" is your GitHub user name.
* Restart NetBeans

rhwood, 2015-09-24:
I think you are making this way to complex. The default push AND pull should be to your fork, not to or from the JMRI repo.

I am having no issues with Git and NetBeans using this workflow:
1) Fork JMRI/JMRI in GitHub website or GitHub Desktop.
2) Clone your fork using GitHub Desktop or whatever tool you choose to your computer.
3) In NetBeans, use “Team->Remote->Pull from Upstream” to pull from the same branch upstream (JMRI/master -> your-fork/master) or “Team->Remote->Pull” if you need to pull from another branch (JMRI/master -> your-fork/your-branch).

> This workaround works just once, and the gets overwritten:
>
> 	• Quit NetBeads
> 	• On the command line, go to the directory you just created and do:
> 		• git remote set-url origin [...]

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

rhwood, 2015-09-24:
I don't agree with the assertion that
> JMRI GitHub users should be set up like:
> 
> % git remote -v
> origin  https://github.com/JMRI/JMRI (fetch)
> origin  https://github.com/bobjacobsen/JMRI (push)

I think they should be set up like:

````
[remote "origin"]
	url = https://github.com/rhwood/JMRI.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master
[remote "JMRI"]
	url = https://github.com/JMRI/JMRI.git
	fetch = +refs/heads/*:refs/remotes/JMRI/*
[branch "travis-ci"]
	remote = origin
	merge = refs/heads/travis-ci
[branch "multi-node-profiles"]
	remote = origin
	merge = refs/heads/multi-node-profiles
````
(Note that in this example there are two feature branches in the fork that are private to the fork)

bobjacobsen, 2015-09-24:
Setting aside that it's hard to get NetBeans to do this, what advantage is there to your proposed setup?

When like this: 
  origin https://github.com/JMRI/JMRI (fetch)
  origin https://github.com/bobjacobsen/JMRI (push)
users with just one on-computer repository have a very simple workflow:  "git pull" makes them up-to-date with the main JMRI/JMRI repository. "git push" moves their changes to their GitHub repository _and_ updates it to current contents.  Any issues with conflicts (and there will be issues with compatibility, there always are) are resolved when the user merges to their local on-computer repository, where they have the tools to do that. This avoids having to deal with conflicts in a GitHub to GitHub pull.

With the config you're recommending, they can do that same thing by remembering to do "git pull JMRI" instead.  "git pull" is harmless, but also ineffective in that case; there's nothing new in the user's GitHub repository.  But the user does have to remember to add the " JMRI" argument.

I'm not sure which (if either) can be easily configured into NetBeans.  A quick experiment (not conclusive) seems to indicate that asking it to do "git pull JMRI" makes the next default push go to the JMRI remote too; not really progress.

rhwood, 2015-09-24:
I think I am recommending that we encourage people working with our code to think of their forks of master more as "just another branch" instead of a "strange second copy of JMRI/master".

What happens when a user pulls JMRI/master into a local copy with changes and then commits and pushes to fork/master? Do the pulled changes show as the user's own work in the same commit or as a separate commit like in a merge?

I had not caught the change in default URI in NetBeans. Seems like we need to file RFIs against NetBeans to decouple the default URI for pushes from the last used URI for pulls.

pabender, 2015-09-24:
On Sep 24, 2015, at 7:18 AM, Randall Wood <notifications@github.com> wrote:

> What happens when a user pulls JMRI/master into a local copy with changes and then commits and pushes to fork/master? Do the pulled changes show as the user's own work in the same commit or as a separate commit like in a merge?
> 
The change history comes with the file.  After a push to a fork on github, the file shows the updates with the original commit information.

Paul

rhwood, 2015-10-01:
Per discussions on the developers mailing list, it appears that Eclipse has the same problems NetBeans has: both IDEs seem to change the working branch for pulls and pushes to the same URL when a push is made.

mattharris, 2015-10-01:
Indeed.

I decided to give up on using Git directly in the IDE and, on my Windows
box, use TortoiseGit.

GitHub desktop doesn't always work for me as it doesn't seem to like the
authenticated proxy that I am sometimes behind yet TortoiseGit handles that
OK.

rhwood, 2015-10-01:
@mattharris Have you reported the problems you are having with the GitHub Desktop to GitHub?

mattharris, 2015-10-01:
<mode="redfaced">
Nope.
</mode>

mattharris, 2015-10-01:
A bit more digging throws this up:

https://help.github.com/desktop/faq/articles/can-i-log-in-behind-a-proxy-server/

Basically, they don't support it...

bobjacobsen, 2015-10-01:
This might be one of those applications for access via SSH:

https://help.github.com/articles/which-remote-url-should-i-use/

(note page is incorrectly optimistic about proxies, so perhaps it’s optimistic about SSH too)

If you want to try that from behind your proxy, here’s some info on changing the URL:

https://help.github.com/articles/changing-a-remote-s-url/

and here’s a hybrid approach:

https://help.github.com/articles/using-ssh-over-the-https-port/

Bob

> On Oct 1, 2015, at 3:21 AM, Matthew Harris <notifications@github.com> wrote:
> 
> A bit more digging throws this up:
> 
> https://help.github.com/desktop/faq/articles/can-i-log-in-behind-a-proxy-server/
> 
> Basically, they don't support it...

--
Bob Jacobsen
jacobsen@mac.com +1-510-486-7355 AIM, Skype JacobsenRG

closed 2015-11-29 by bobjacobsen

githeko, 2017-04-10:
Am trying to get Netbeans to work with Bitbucket. In Android Studio its a simple as copy and paste a URL. In git it looks like you need multiple PhDs to get there. After a whole evening mucking around I am no nearer the truth than when I started. Perhaps I should change my career and go back to growing potatoes....

dheap, 2017-04-11:
> @devel-bobm Personal opinion - I don't do anything "Git"-ish within NetBeans' integrated stuff other than "revert local modifications".  Instead I use a command-line GIT for pulls, merges, commits, reverts, branch creation/checkout, etc.  I find it very easy to know exactly what GIT commands I'm executing when I know what I'm typing, so I know that I'm following the JMRI-recommended actions.  The same cannot be said for pulldown menu choices within NetBeans...

Agreed. I have exactly the same approach.

AlanUS, 2017-04-22:
I'm using NetBeans local commit only, and GitHub Desktop + GitHub on-line for all the rest. Works pretty well, only push buttons, so no need to use command line if you don't like it.

shosigh, 2023-06-26:
lodu lalit

PR #288 (2015-11-29): Improve developer Git docs
Add discussion of testing a pull request. Improve NetBeans doc, closes #10
commit a979fa3d (2015-11-29): Update NetBeans doc to current practice

Improve discussion of pull, push and pull-requests to current practice. Fix some links.

Closes #10