JMRI/JMRI #30: Keyword expansion

bobjacobsen, 2015-09-26:
For some (possibly incorrect) background, see:  http://jmri.org/help/en/html/doc/Technical/GitFAQ.shtml#CVSCookies  (Sorry, don't know how to get it to automatically open at the item)

This Issue comes from the discussion in pull  https://github.com/JMRI/JMRI/pull/29  where Paul Bender said:

> I'm not ready to say the SVN keywords should be deprecated yet.
>
> It looks like we can get them to work using got attributes.
>
> https://git-scm.com/book/en/v2/Customizing-Git-Git-Attributes
>
> I need to look at that page more closely later to get the proper keyword expansions.
>
> Other than the ID keyword, the one we use most frequently is revision....

and Randall replied:

> Note that per the documentation:
> 
>  * keyword expansions in git are not included in the commits, so it's retained only locally in the working copy
>  * it can get ugly if you require certain applications (i.e. bash) to expand keywords, and not everyone is capable of using those applications.

My feeling on this is that there _are_ certain files, esp. decoder files, that people will pass around outside Git (as they did with SVN).  When they finally come back to us with useful and valuable changes, having a _reliable_ internal version identification can be very helpful.  But an unreliable one is worse than none at all.  So I think we should not use them unless we can be sure that they work reliably when un-sophisticated (e.g. non-programmer) users pick them up any of the possible sources:  A JMRI distribution, our website, Git or GitHub, or anywhere else they usually do this.

I don't know how to achieve that with the existing tags and Git.

It _might_ be possible to pull the existing tags and replace them with something new.  I'm certainly open to that.  Would be easy to e.g. insert something in every XML file as a one time operation.

bobjacobsen, 2015-09-26:
This might be of interest:
http://thomas-cokelaer.info/blog/2014/12/moving-from-svn-to-git-id-equivalent/

rhwood, 2015-10-01:
Per https://github.com/eclipse/jgit/blob/master/README.md git attributes are not supported in the Java implementation of git. This precludes their use within Eclipse or NetBeans (and likely also IntelliJ)

bobjacobsen, 2015-10-15:
Closing this. No plans to use attributes now.

closed 2015-10-15 by bobjacobsen