JMRI/JMRI #41: ci-tests cannot be run in NetBeans

rhwood, 2015-09-27:
Prior to adding the "quiet" macro, the ci-tests could be run in NetBeans (and provide output integrated into NetBeans). Now I get this error:
````
ant -f /Users/rhwood/Developer/JMRI/JMRI/JMRI ci-test
init:
/Users/rhwood/Developer/JMRI/JMRI/JMRI/build.xml:277: The following error occurred while executing this line:
/Users/rhwood/Developer/JMRI/JMRI/JMRI/build.xml:220: 
javax.script.ScriptException: TypeError: org.apache.tools.ant.module.bridge.impl.NbBuildLogger@6cfdb295 has no such function "setMessageOutputLevel" in <eval> at line number 2
	at jdk.nashorn.api.scripting.NashornScriptEngine.throwAsScriptException(NashornScriptEngine.java:455)
	at jdk.nashorn.api.scripting.NashornScriptEngine.evalImpl(NashornScriptEngine.java:439)
	at jdk.nashorn.api.scripting.NashornScriptEngine.evalImpl(NashornScriptEngine.java:401)
	at jdk.nashorn.api.scripting.NashornScriptEngine.evalImpl(NashornScriptEngine.java:397)
	at jdk.nashorn.api.scripting.NashornScriptEngine.eval(NashornScriptEngine.java:152)
	at javax.script.AbstractScriptEngine.eval(AbstractScriptEngine.java:264)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.apache.tools.ant.util.ReflectUtil.invoke(ReflectUtil.java:108)
	at org.apache.tools.ant.util.ReflectWrapper.invoke(ReflectWrapper.java:81)
	at org.apache.tools.ant.util.optional.JavaxScriptRunner.evaluateScript(JavaxScriptRunner.java:103)
	at org.apache.tools.ant.util.optional.JavaxScriptRunner.executeScript(JavaxScriptRunner.java:67)
	at org.apache.tools.ant.taskdefs.optional.Script.execute(Script.java:52)
	at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
	at sun.reflect.GeneratedMethodAccessor315.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:106)
	at org.apache.tools.ant.Task.perform(Task.java:348)
	at org.apache.tools.ant.taskdefs.Sequential.execute(Sequential.java:68)
	at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
	at sun.reflect.GeneratedMethodAccessor315.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:106)
	at org.apache.tools.ant.Task.perform(Task.java:348)
	at org.apache.tools.ant.taskdefs.MacroInstance.execute(MacroInstance.java:396)
	at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
	at sun.reflect.GeneratedMethodAccessor315.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:106)
	at org.apache.tools.ant.Task.perform(Task.java:348)
	at org.apache.tools.ant.Target.execute(Target.java:435)
	at org.apache.tools.ant.Target.performTasks(Target.java:456)
	at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1393)
	at org.apache.tools.ant.Project.executeTarget(Project.java:1364)
	at org.apache.tools.ant.helper.DefaultExecutor.executeTargets(DefaultExecutor.java:41)
	at org.apache.tools.ant.Project.executeTargets(Project.java:1248)
	at org.apache.tools.ant.module.bridge.impl.BridgeImpl.run(BridgeImpl.java:286)
	at org.apache.tools.ant.module.run.TargetExecutor.run(TargetExecutor.java:555)
	at org.netbeans.core.execution.RunClassThread.run(RunClassThread.java:153)
Caused by: <eval>:2 TypeError: org.apache.tools.ant.module.bridge.impl.NbBuildLogger@6cfdb295 has no such function "setMessageOutputLevel"
	at jdk.nashorn.internal.runtime.ECMAErrors.error(ECMAErrors.java:57)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:213)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:185)
	at jdk.nashorn.internal.runtime.ECMAErrors.typeError(ECMAErrors.java:172)
	at jdk.nashorn.internal.runtime.linker.NashornBottomLinker.linkBean(NashornBottomLinker.java:103)
	at jdk.nashorn.internal.runtime.linker.NashornBottomLinker.getGuardedInvocation(NashornBottomLinker.java:71)
	at jdk.internal.dynalink.support.CompositeGuardingDynamicLinker.getGuardedInvocation(CompositeGuardingDynamicLinker.java:124)
	at jdk.internal.dynalink.support.LinkerServicesImpl.getGuardedInvocation(LinkerServicesImpl.java:149)
	at jdk.internal.dynalink.DynamicLinker.relink(DynamicLinker.java:233)
	at jdk.nashorn.internal.scripts.Script$\^eval\_.:program(<eval>:2)
	at jdk.nashorn.internal.runtime.ScriptFunctionData.invoke(ScriptFunctionData.java:636)
	at jdk.nashorn.internal.runtime.ScriptFunction.invoke(ScriptFunction.java:229)
	at jdk.nashorn.internal.runtime.ScriptRuntime.apply(ScriptRuntime.java:387)
	at jdk.nashorn.api.scripting.NashornScriptEngine.evalImpl(NashornScriptEngine.java:437)
	... 42 more
BUILD FAILED (total time: 1 second)

````

rhwood, 2015-09-27:
Actually, I can't run any build tasks in NetBeans because of this error.

mattharris, 2015-09-27:
Yup, I've found the same.

Commenting out all uses of the quiet macro allows normal service to resume (at least in my local environment)

bobjacobsen, 2015-09-27:
Sorry about the trouble.

It looks like NetBeans 8 includes a not-quite-complete logger implementation that it injects to Ant.

I’ve updated the build file to check for the Ant one before acting.  Pull request #43 is building in Travis and then I’ll merge it.

Bob

> 
> Yup, I've found the same.
> 
> Commenting out all uses of the quiet macro allows normal service to resume (at least in my local environment)

commit 0171f67c (2015-09-27): Merge pull request #43 from bobjacobsen/master

Fix conflict between quiet element in build.xml and NetBeans
Fixes #41.
closed 2015-09-27 by rhwood