JMRI/JMRI #67: Line Ends in Git

bobjacobsen, 2015-10-03:
See discussion by @devel-bobm in pull request #66 for some key background.

We set up file-level attributes in SVN to use local line-ends, keeping Linux/Mac line-ends in the repository.  (I think we did this file-by-file mostly for historical reasons) The SVN command was 

    svn propset svn:eol-style native <file-path-name>

We probably need to do something about line ends for Git.

GitHub's help as a page on dealing with line endings: https://help.github.com/articles/dealing-with-line-endings/

It shows how to issue a command on _each_ user machine to "properly handle line endings" (That references Mac OS X on my Mac OS X machine; could somebody on Windows please check what it says on a Windows-resident browser?)

There is a per-repository setting, but the process to set it up is a bit fierce. 

How do we want to handle this?

Bob

devel-bobm, 2015-10-03:
I suspect that you should be able to get the Windows-related info from the GitHub's "page on dealing with line endings" regardless of O/S you are on.  With my Windows/Firefox browser, I see a set of O/Ses just underneath the primary heading.  If I select "All" for operating system, it presents information for "all" operating systems.  Doing so shows the following under "global settings for line endings":

`$ git config --global core.autocrlf input`
`# Configure Git on OS X or Linux to properly handle line endings`

`$ git config --global core.autocrlf true`
`# Configure Git on Windows to properly handle line endings`

Since this has an effect on <ALL> Git repositories on the machine, I suspect that this will not be a good solution for all developers.

I suspect we will have to bite the bullet and take the per-repository route, as this ought to make the process "transparent" to operating system and development environment.  Yes, there will probably be some annoyingly-large "pulls" from JMRI/JMRI immediately after this, but that's something we would have to deal with.

Bob M.

bobjacobsen, 2015-10-03:
Thanks, hadn't noticed the selector at the top of that page.

I agree that messing at the user-machine level probably isn't the right solution.  

Before we make a repository-level change, though, I want to make sure that the git-migration group @davidzuhn @mattharris @kevin-dickerson @rhwood @plocher have had a chance to weigh in, so I'm referencing them

(Still not clear on which discussions are best handled here vs jmri-developers, but as this is a git-related one, I figure here for this item)

rhwood, 2015-10-04:
I would make the repository-level change. That said, note that both Eclipse and NetBeans (and likely IntelliJ IDEA) ignore git attributes, so users who insist on using these tools to commit changes will still need to make the per-machine changes.

bobjacobsen, 2015-11-09:
So do we really need to do something here?  It's been a month, with commits from multiple people, and except for the #66  pull request that kicked this off, everything has been OK.  Does Git's automatic behavior do what we want?

devel-bobm, 2015-11-09:
I do not currently have an opinion, as I haven't made a pull request to JMRI/JMRI in more than a month.  

I do have a few things in the works, but I do not anticipate making a pull-request from my repository any time sooner than a week from now.

bobjacobsen, 2015-11-24:
I'm closing this as "apparently not a problem at the current time".  We can re-open if anybody has trouble with this issue in the future.

closed 2015-11-24 by bobjacobsen