JMRI/JMRI #103: Release build process needs work in GitHub.

rhwood, 2015-10-19:
See https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md for the current proposal.

rhwood, 2015-10-19:
I would do the following:
- [ ] Use `release-X.Y.Z` for the branch instead of `Developer Release-x.y.x`; this would follow the naming that currently exists.
- [ ] Tag releases, when released, only as `X.Y.Z` or `vX.Y.Z`; these are both common idioms for tagging releases in GitHub already.
- [ ] Use GitHub releases
- - See https://github.com/JMRI/JMRI/releases/tag/Release-2-12 for an example of an existing release
- - See https://help.github.com/categories/releases/ for documentation
- - See https://github.com/JMRI/JMRI/releases/new to start a new release

mattharris, 2015-10-19:
The branch title is proposed as `release-x.y.z` - the `Develop release-x.y.z` is the description to use in the commit message/pull request for generation of the branch.

rhwood, 2015-10-19:
Sorry about misreading that.

mattharris, 2015-10-19:
No problem

But I do think that it would be worth considering the use of GitHub releases to hold additionally copies of the binaries as well as on SF.net

For the tagging, making use of either of the common GitHub idioms possibly has some benefits - if we were to choose to follow this, I'd probably go for the `vX.Y.Z` form.

bobjacobsen, 2015-10-20:
I've updated the https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md doc to reflect the above.  Thanks for the suggestions!

I've also put together a 1st version of the build.xml for the "ant make-test-release-branch" operation.  You can see it here: https://github.com/JMRI/JMRI/blob/master/build.xml  The relevant part is at the end.

I've used that to create a sample "release-4.1.3" test release branch for JMRI 4.1.3, visible here: https://github.com/JMRI/JMRI/tree/release-4.1.3   Note that the release.properties file has had release.is_branched=true automatically set.  The release number in the master branch release.properties has also been set to 4.1.4; we can set that back if we're not using this branch.

Finally, I've set up the first Jenkins job for this:  http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.1.3/
I'll check it further when it's done.

bobjacobsen, 2015-10-20:
Jenkins was OK on the second try.

I added some instructions for creating a Github list near the bottom of  https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md  and will try it in more detail when we decide to release 4.1.3; are we ready to do that soon?

rhwood, 2015-10-20:
If we change how we ship the OS X build from a .dmg to a .zip, we could have travis-ci automatically populate the release. See http://docs.travis-ci.com/user/deployment/ and http://docs.travis-ci.com/user/deployment/releases

Alternately, Jenkins could do this as well. See http://stackoverflow.com/questions/24585609/upload-build-artifact-to-github-as-release-in-jenkins and https://developer.github.com/v3/repos/releases/#upload-a-release-asset

pabender, 2015-10-20:
On 10/20/2015 04:56 AM, Randall Wood wrote:
>
> If we change how we ship the OS X build from a .dmg to a .zip, we 
> could have travis-ci automatically populate the release. See 
> http://docs.travis-ci.com/user/deployment/ and 
> http://docs.travis-ci.com/user/deployment/releases
>

I don't see anything on the page above that references file types.

As I understand the travis-ci engine, as you have it configured, it's 
doing the builds on a Linux virtual machine.  It's possible to build dmg 
files on a Linux machine, and the only difference between the ones 
created on a Mac and the ones created under Linux is that the MacOS 
tools are capable of changing the size of the file so that it doesn't 
include any extra free space, but the Linux tools aren't (or were not at 
the time we created this) capable of doing the truncation required.

There is a script to handle the disk image creation in the scripts 
directory (diskimage.sh) that David Z. wrote from something much more 
crude that I originated.

Paul

rhwood, 2015-10-21:
I was under the impression we had to build the DMGs on OS X, not Linux which is why I made that comment.

bobjacobsen, 2015-10-24:
Anybody know how to create a pull request for a completely _new_ branch via the GitHub web site?  

I ended up pushing directly to the main JMRI/JMRI, instead of using a pull request, because I couldn't see how to generate a pull request from a new branch (in my fork) into GitHub _without_ having it merge onto an existing branch.

rhwood, 2015-10-24:
> Anybody know how to create a pull request for a completely new branch via the GitHub web site?

It can’t be done because PRs are essentially the diffs between two branches. You need to create the new branch first, and then submit the PR against it.

silverailscolo, 2015-10-24:
Hi Bob,
What I have learned to do via Github Web is:
Start by creating a new branch from either my own updated Fork of JRMI/JMRI called silveraislcolo/JMRI (enter a new name in the branch pop up list) or the original and then 
Commit changes to this Fork in Github Desktob or Github Web. When ready, I create a Pull Request to JMRI/JMRI/master.

It is indeed a different experience for one of the owners/contributors. They won’t get the Propose… screen when they click the pencil icon but instead, get 2 options to commit changes, as pictured in the attached image:
1. Either Commit directly to the master branch (where I clicked on the pencil) or
2. Create a new branch etc.

Is that what you are looking for? ( I tried this on one of my own - unlocked - repos)

Egbert

rhwood, 2015-10-25:
Can we ensure that the following test milestone is set when branching for a release? This would allow PRs to be filed while working on features that can't go into the current test release, and give us a history of which PRs were targeted for which release.

I waited to submit PR #113 until after 4.1.3 was released because I knew I couldn't get it in time, but I would have liked to open the PR much earlier for commentary.

bobjacobsen, 2015-10-25:
Re mark PRs for a future release:  this goes to the branching model. Rather than leave a PR as open-but-please-don't-pull-yet on the master branch, I think I'd prefer a model where the longer development work is on a branch in JMRI/JMRI. That can have its own PR discussions, etc, and can get merged when ready.

rhwood, 2015-10-28:
I was thinking more that every PR, even short lived ones, should be set against the milestone when the developer thinks it will be in JMRI. That way, the release notes can refer to "bugs fixed", "changes made", or something similar merely by including a link to https://github.com/JMRI/JMRI/milestones/<release-version>

bobjacobsen, 2015-10-28:
(Re milestones) Interesting suggestion!  I've never used those before.

I added the 4.1.4 milestone to some of my recent commits to see what it looks like.

The standard display (https://github.com/JMRI/JMRI/milestones/4.1.4), which you get by clicking on the link in Randall's comment above then the 4.1.4 entry, just shows the _open_ items.  That's a bit unfortunate if we want this to show as a pseudo-auto-release note. You can get the list of closed items by an extra click on "closed" (https://github.com/JMRI/JMRI/issues?q=milestone%3A4.1.4+is%3Aclosed), but I don't see an easy target to get "all" (though you can get it by with the right link: https://github.com/JMRI/JMRI/milestones/4.1.4 )

I think it's OK to encourage people to add milestones, either when creating the PR or when actually pulling it.  Maybe that'll encourage better PR titles, perhaps even better habits on commits, branches, etc.  

But I'm still not sure about the idea of having a _future_ milestone on a PR mean that we shouldn't pull it until then.  I think GitHub issues are better for that kind of thing than pull _requests_.

bobjacobsen, 2015-11-09:
Going for the 2nd from-GitHub release this coming weekend.

Updated instructions are here: https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md

They still need a lot of work, but hopefully one more pass during a real release build will help.

One major flaw that needs fixing, in the process raising a question:  The process starts with a lot of update-and-commit steps, e.g. for decoder index, etc, that are still in the "commit to master" model from SVN.  I think I'd rather have those be "do all those updates on a branch, and pull them into JMRI/JMRI with a pull request" so that we're creating a record.   The question is whether that should be the start of the release-n.n.n branch, which gets merged to master before carrying on for the final release steps?  Or some other one? The changes are really just routine development work, only temporally associated with the release.  I'm leaning toward creating a "bobj-release-prep" temporary branch and leaving it at that.

On the question of milestones above:  I've added 4.1.4 to a bunch of the include pull requests, and a link to the list is in the release note.  I'll add 4.1.4 to the rest of them.  We can ask on the JMRIusers list if the release summary is helpful.  (I think we'll need better pull request titles for it to be really helpful, but perhaps that will come)  

I'm still of two minds on opening pull requests before they're ready to pull (c.f. #113 ). It's a handy way to discuss code, but I'm concerned that somebody will pull them too soon while trying to be helpful. Perhaps we should open a separate item about that?

bobjacobsen, 2015-11-24:
Instructions updated while doing the 4.1.4 build. Next step is for somebody else to use for 4.1.5, then we can close this.

rhwood, 2015-12-16:
I have a suggestion for possibly another release branching / tagging strategy:

- After the release of a Production version, a branch is created from master for the next production version (the next-production branch). Releases are done by tagging the next-production branch at release time and then building from the branch at the tagged state without creating a new per-release branch.
- For the first few test releases (a.t.1 - a.t.M), the next-production branch gets updated by merging master into the branch prior to building a release.
- Releases closer to the production release (a.t.(M+1) - a.t.N) have changes cherry-picked from master based on milestones associated with PRs.
- Critical updates to production releases would occur on the branch for that release.
- Release tags include a "Release Candidate" and be in the form v4.3.1rc1, so that if multiple builds need to be done for a single release, each build is tagged with an "rc" number.

For the next production release (4.4.0), this would mean having a branch "release-4.4" or something similar, with releases 4.3.1-4.3.N, 4.4.0, and if needed, 4.4.N, all built off the same branch.

I am suggesting this because I think it might be easier to submit PRs for specific releases against a longer-lived branch, instead of having short-lived branches for each release, and while it is not entirely trivial to push a PR against two different branches, it's not difficult either.

bobjacobsen, 2015-12-31:
The basic test release process is working, I think.  See $383 for the most recent example.

But we do need to think a bit about how to handle the production-release run up before we get to 4.4.  I'll close this for now, and we can open another one in a few months with more experience.

closed 2015-12-31 by bobjacobsen