JMRI/JMRI #148: Improvements to decoder validation

bobjacobsen, 2015-11-05:
I want to tighten up the XML Schema for the decoder definition.  Specifically, I want to make sure that constructs like this (simplified):

```xml
<variable>
  <decVal/>
  <splitVal/>
</variable>
```

don’t validate.  They don’t work, in any meaningful sense, and worse they fail silently.  So I’d like them to fail vocally in the validation step.

Right now, the modifiers like “label” and “qualifier” can appear before or after the variable type:

```xml
<variable>
   <label/>
  <decVal/>
  <label/>
</variable>

<variable>
  <decVal/>
  <label/>
</variable>

<variable>
   <label/>
  <decVal/>
</variable>
```

all validate and work assuming that the labels are different, e.g. different languages. The schema _can_ check for “can’t have multiple labels in the same language” (which doesn’t work, and a quick check found a couple) if they’re always before or always after. It can’t check (and doesn’t now) if they can be in either place.  

For a long time, labels had to be after, and most of them are.  (The change to allow them before was inadvertent).  I’m planning to restore that constraint, so that label, comment and tooltip have to be after the variable type.  I’ll fix any existing files that don’t validate with that, and commit them all together.    Then I’ll (slowly) work through solutions to duplicate labels, and eventually commit those along with a constraint on that too.

But that’s not why I’m writing.  I’m writing because I’d like to _also_ constrain qualifiers to be either before or after the variable type.  I need to do that to restore the “just one variable type element” constraint due to a part of the schema vocabulary called the “Unique Particle Attribution Constraint”, but we don’t have to go into detail on that.  

What matters is choosing either “before” or “after”.  

There are a few files that have the qualifier before.  I like the way that looks, because first you see the conditions where the variable exists, then you see it’s type, and the details follow.  That seems easiest to read.

But there are more files with the qualifier after the variable type. That might be historical, but it looks a bit harder to read, at least to me.

Does anybody have a strong preference on this?  I’m happy to edit and commit all that files that don’t validate either way, it’s probably only an hour of editing to do that (if there are too many, I might just do a XSLT macro to fix it)

Thanks for your thoughts.

Bob

KenC57, 2015-11-05:
I'd think the 'cleaner reading' is an important point. It should help to
keep these easy for more people to work on and not mess up.

 

So making the validate correctly picky then helps them insure the best
results of their efforts. The two goals seem to work well together.

 

So the only thing I'd suggest is that we may want to place a notice in the
dev and user lists so people working on any decoder files are aware of a
change in progress and might hold off until you finish.

 

Then we should schedule a dev release as soon as this is all cleaned up so
it gets into as many hands as possible. Granted my thoughts are based on we
have a significant number of folk who work on decoder xml's but not part of
the normal Git base (getting this message) and instead submit patch
files/zips to submit their work.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2015-11-06:
I'm generally opposed to requiring order in the schema when that order does not matter to the software using a conforming XML document. However, it seems that labels are after the variable type in a sequence in the schema (unless [decoder.xsd](https://github.com/JMRI/JMRI/blob/master/xml/schema/decoder.xsd) is not the schema we are discussing).

To enforce a requirement on uniqueness in labels would not using [```xs:unique```] for all ```text``` elements within the label (https://msdn.microsoft.com/en-us/library/ms256146.aspx) constraints work while only allowing one label element per variable? (Or am I misunderstanding the problem?)

Is the ```xs:choice``` around value types (decVal, splitVal, etc…) in the decoder schema not working to enforce a requirement that only one value type be defined per variable?

bobjacobsen, 2015-11-06:
I'm not in favor of over-constrained schema either.  But some constraints are difficult to express in XML Schema.

Boiling down the existing schema:
```xml
  <xs:element name="variable" minOccurs="1" maxOccurs="unbounded" >
    <xs:complexType>
      <xs:sequence minOccurs="0" maxOccurs="unbounded">

        <xs:choice minOccurs="1" maxOccurs="1">
          <xs:element name="enumVal" type="EnumValType"/>
          <xs:element name="decVal" />
          <xs:element name="hexVal" />
          <xs:element name="longAddressVal" />
          <xs:element name="shortAddressVal" />
          <xs:element name="splitVal"/>
          <xs:element name="compositeVal"/>
          <xs:element ref="xi:include" minOccurs="0" maxOccurs="1" />
        </xs:choice>

        <xs:sequence minOccurs="0" maxOccurs="unbounded">
          <xs:element name="label" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="comment" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="tooltip" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="qualifier" type="QualifierType" minOccurs="0" maxOccurs="unbounded" />
        </xs:sequence>
      </xs:sequence>
    </xs:complexType>
  </xs:element>
```

The outer sequence can be invoked multiple times, which allows e.g. label (on 1st turn of outer sequence), then variable and another label on a 2nd turn of the outer sequence.  And that's fine.  But it also allows e.g. decVal to appear twice: once on the first turn of the outer sequence, and another on the next turn.  There's no good way to avoid that when the outer sequence can be used more than once.

A single pass approach might look like:

```xml
  <xs:element name="variable" minOccurs="0" maxOccurs="unbounded" >
    <xs:complexType>
      <xs:sequence minOccurs="1" maxOccurs="1">
 
        <xs:sequence minOccurs="0" maxOccurs="unbounded">
          <xs:element name="label" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="comment" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="tooltip" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="qualifier" type="QualifierType" minOccurs="0" maxOccurs="unbounded" />
        </xs:sequence>

       <xs:choice minOccurs="1" maxOccurs="1">
          <xs:element name="enumVal" type="EnumValType"/>
          <xs:element name="decVal" />
          <xs:element name="hexVal" />
          <xs:element name="longAddressVal" />
          <xs:element name="shortAddressVal" />
          <xs:element name="splitVal"/>
          <xs:element name="compositeVal"/>
          <xs:element ref="xi:include" minOccurs="0" maxOccurs="1" />
        </xs:choice>

        <xs:sequence minOccurs="0" maxOccurs="unbounded">
          <xs:element name="label" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="comment" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="tooltip" minOccurs="0" maxOccurs="unbounded" />
          <xs:element name="qualifier" type="QualifierType" minOccurs="0" maxOccurs="unbounded" />
        </xs:sequence>
      </xs:sequence>
    </xs:complexType>
  </xs:element>
```

This would allow labels & qualifiers before _or_ after the variable type definition, or even both.

But unfortunately, it's not a valid XML Schema because of the “Unique Particle Attribution Constraint” mentioned above.  You basically can't have an element of the same name (e.g. label) in two places (e.g. before and after the choice element) and allow the uncertainty of an optional include in between.  From the Schema formalism point of view, there's no guarantee that the included document will have a enumVal, splitVal or similar, so once it's expanded it might not be possible to tell whether a <label> element is from the 1st sequence or the 2nd.  (In our case, we can ensure that by the way we check every XML document and document fragment against the Schema, but the Schema itself can't ensure that)

bobjacobsen, 2015-11-08:
I've been proceeding with this, with the "qualifier at the front" choice.  The new schema threw a lot of errors at first, but after I moved the qualifers to the front (I ended up writing a simple decoderPromoteQualifier.xsl XSLT transform to automate it) I was able to find errors like:

```xml
      <variable item="Analog Power Conversion" CV="124" mask="XXVXXXXX" default="0">
        <xi:include href="http://jmri.org/xml/decoders/parts/enum-OffOn.xml"/>
        <enumVal>
          <xi:include href="http://jmri.org/xml/decoders/zimo/Choice-Ligth1.xml"/>
          <xi:include href="http://jmri.org/xml/decoders/zimo/Choice-Ligth2.xml"/>
        </enumVal>
        <label>Polarity Dependent DC stopping method</label>
        <label xml:lang="it">Fermata in DC dipendente da polarità</label>
        <label>Light Effects Output 1</label>
        <label xml:lang="it">Effetti Luci uscita 1</label>
      </variable>
```

where the variable type is defined twice:  Once by an include of a standard variable, and then (ineffectively) by a local definition.

I've been handling these on the "make it still do the same thing" principle.  The code takes the 1st one, so that's what I've been keeping.

bobjacobsen, 2015-11-08:
The update for this is now in PR #156

closed 2015-11-08 by bobjacobsen