JMRI/JMRI #159: LocoNet throttle conflict

bobjacobsen, 2015-11-09:
Martin Jahner opened a [bug report](https://sourceforge.net/p/jmri/bugs/534/) on the SF.net bug tracker that says:

```
I've encountered an issue in the way JMRI throttles update the slots in a Digitrax command station. 
If two throttles are selected to control the same engine/slot, and one of them releases the address,
both throttles suddenly lose control and the slot is set back to "Common". The fact that one 
throttle can affect the other is already a bit of a concern, but the more important problem is this:
Once this has happened, no JMRI throttle is able to take control of that slot anymore. 
Selecting the same address again on either throttle (or a new one) will not set the slot 
back to "In Use". This also affects WiFi throttles.
Only Digitrax throttles on LocoNet are not affected, and they will be able to set the slot back to "In Use".

The attached logfile shows a brief test session where I am selecting the same engine on 
two separate throttles first separately, then together, and then release them to cause the issue. 
I put comments with "INFO" into the log to describe what I did and what happened.

I hope this is enough information for someone with knowledge of the internals to debug and 
[monitorLog_DETAIL.txt](https://github.com/JMRI/JMRI/files/29746/monitorLog_DETAIL.txt)
hopefully fix this. I'll be happy to provide more information and/or test things if needed.
```

I'm pulling this over to GitHub partly to see how these issues work for JMRI bug tracking, and partly to get it a bit more attention.

SteveRawlinson, 2016-01-17:
There are a number of ways to prevent this problem. The simplest fix is in pull request #650 

The underlying issue is that if a throttle request is made for a decoder address which already has a throttle open we "return" the existing throttle, but we make some changes as if a new throttle had been created. For example, we update a useActiveCount in the appropriate Addresses object. When the throttle is released all 'copies' of it get the address unset, and the slot status is changed to COMMON but the useActiveCount only gets decremented once. Requesting that address again does not send a slot request to the command station (the slot is already in use, right?) and all subsequent changes on the throttle are ignored by the command station and silently dropped.

My works by suppressing the increment of the useActiveCount if we find an existing throttle with the same address. This throws up another problem: following a release the slot is updated by the panel before all the listeners are reaped, the slot variable gets set to null part way through notifyChangedSlot() causing a NPE. My fix is not very sophisticated but works: I check for a null slot and return before any further dereferences.

There are other ways of resolving this problem, one of which is not to allow multiple panels (or wiThrottles) to have the same address.

rhwood, 2016-01-17:
> There are other ways of resolving this problem, one of which is not to allow multiple panels (or wiThrottles) to have the same address.

I #650 is the best way to fix this.

bobjacobsen, 2016-01-19:
@SteveRawlinson's PR #650 is believed to fix this. It's been merged and will be in the next development packages (http://builds.jmri.org/jenkins/job/Development/job/Packages/) 2962 or later.

closed 2016-01-19 by bobjacobsen