JMRI/JMRI #193: Roster & Locale Preferences Problem

dheap, 2015-11-16:
I encounter a number of problems migrating any existing profile from V4.1.3 or earlier to a build showing  V4.1.3ish-Rd2d3fee.

I created a test profile under 4.1.3 that was unremarkable except for the following:
- LocaleCountry="AU"
- programmer defaultFile="Advanced"
- defaultUserLocation="/Ours/Our Documents/Dave/Dropbox/JMRI/"
- roster directory not set (implicitly uses defaultUserLocation).

After migration and restart, i have the following:
- [ ] apps-gui.locale=en-US (changed).
- [ ] programmer defaultFile not set (changed).
- jmri-implementation.user-files=/Ours/Our Documents/Dave/Dropbox/JMRI/ (unchanged).
- [ ] jmri-jmrit-roster.directory=profile\: (changed & roster no longer visible in main window).

Using Preferences to correct:
- LocaleCountry can be set to "AU".
-  [ ] Programmer Format shows "        ". Changing it produces e.g. jmri-jmrit-symbolicprog.defaultFile=Basic but this has no effect. Format stays at pre-migration setting and ProgrammerFormat continues to show "        ".
- roster directory Reset produces jmri-jmrit-roster.directory=preference\: (roster now visible in main window).

rhwood, 2015-11-16:
Locale issues should be fixed by #195 .

dheap, 2015-11-17:
Tested and it is fixed, thanks.

dheap, 2015-11-17:
Programmer format issues fixed with [#203](https://github.com/JMRI/JMRI/pull/203), thanks.

Probably need to trigger a restart request after changing Programmer format so changes take effect.

rhwood, 2015-11-17:
Since the Programmer format changes took effect immediately everywhere but the Roster window, #206 makes that change take effect immediately in the Roster window as well. This change is effective even without hitting the Save button (although it will not persist across restarts of JMRI unless the Save button is clicked).

dheap, 2015-11-17:
Confirmed it works well with, #206 <https://github.com/JMRI/JMRI/pull/206> ,
thanks.
>

rhwood, 2015-11-18:
@dheap Can you test #220 with DEBUG logging turned on the Roster and RosterConfigManager classes? I would like to see what those classes report when you use a non-default user files location.

dheap, 2015-11-18:
Here they are.

[1-migrate-session.txt](https://github.com/JMRI/JMRI/files/38270/1-migrate-session.txt)
[2-restart-session.txt](https://github.com/JMRI/JMRI/files/38271/2-restart-session.txt)
[3-reset-session.txt](https://github.com/JMRI/JMRI/files/38272/3-reset-session.txt)
[4-fixed-session.txt](https://github.com/JMRI/JMRI/files/38273/4-fixed-session.txt)
[messages.txt](https://github.com/JMRI/JMRI/files/38274/messages.txt)

rhwood, 2015-11-19:
I think #223 fixes the last of these issues. @dheap can you confirm or deny?

dheap, 2015-11-19:
No luck, unfortunately.
[223-messages-filtered.txt](https://github.com/JMRI/JMRI/files/38511/223-messages-filtered.txt)

[223-messages.txt](https://github.com/JMRI/JMRI/files/38512/223-messages.txt)

rhwood, 2015-11-19:
@dheap What did you do? Your 223 test shows the user doing something different than your earlier tests. Line 182 in 223-messages.txt shows that the roster directory is being manually set to something equivalent to the ```preferences:``` directory, but not, critically, to the ```preferences:``` directory (it was set by an external process to the spelled out preferences: directory minus the trailing ```/``` — at no point does ```FileUtil.getPreferencesPath()``` or ```FileUtil.getProfilePath()``` return a directory without the trailing ```/```). My understanding is that if a user does this, I should respect their decision to maintain that directory, even when the ```preferences:``` directory is changed; this appears to be working as expected. When you looked at the profile.properties file, what did it show?

dheap, 2015-11-19:
I'm replying via email because this comment either is deleted or edited:

Ignore everything prior to line 172. I should have deleted them. Lines 154-170 is me running a JMRI 4.1.3 distribution (note version and program:) for the sole purpose of duplicating "4.1.4 Created" to "Test 223-2" for testing. Profile "Test 223" was already screwed by the migration and I had forgotten to configure logging.

Lines 172 onwards are using an ant build DecoderPro and are of interest.
> @dheap What did you do? Your 223 test shows the user doing something different than your earlier tests. Line 182 in 223-messages.txt shows that the roster directory is being manually set to [...]
I did nothing. I just ran an ant build DecoderPro and it did the migration of "Test-223-2". The only two things I did was to OK the migration warning and Quit.
> My understanding is that if a user does this, I should respect their decision to maintain that directory, even when the preferences: directory is changed; this appears to be working as expected.
> 
I had done nothing. that copied 4.1.3 profile had no reference to roster directory in either   ProfileConfig.xml or profile.properties.
> When you looked at the profile.properties file, what did it show?
> 
Nothing.

I will do another sequence and zip up the profile at each stage.

rhwood, 2015-11-19:
Please use a real 4.1.3 or 4.0.1; the logs indicate you used a 4.1.3ish version, which is not reliable for testing.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

dheap, 2015-11-19:
Ran 4.1.3 from /Applications to create (via duplication of existing) profile "Fresh Test"
[Fresh_Test.zip.txt](https://github.com/JMRI/JMRI/files/38868/Fresh_Test.zip.txt)

Ran JMRI build with debugging. Pressed OK to migration warning and Quit
[Fresh_Test 2.zip.txt](https://github.com/JMRI/JMRI/files/38874/Fresh_Test.2.zip.txt)

Ran JMRI build with debugging. Profile obviously screwed:
[locations.txt](https://github.com/JMRI/JMRI/files/38877/locations.txt)
Quit
[Fresh_Test 3.zip.txt](https://github.com/JMRI/JMRI/files/38878/Fresh_Test.3.zip.txt)

Ran JMRI build with debugging
Preferences->Roster->Roster location->Reset
Save->Later & Quit
[Fresh_Test 4.zip.txt](https://github.com/JMRI/JMRI/files/38887/Fresh_Test.4.zip.txt)

Ran JMRI build with debugging
(accidentally opened preferences instead of Help->Locations)
[locations 2.txt](https://github.com/JMRI/JMRI/files/38891/locations.2.txt)
All Good now.
Quit
[Fresh_Test 5.zip.txt](https://github.com/JMRI/JMRI/files/38895/Fresh_Test.5.zip.txt)

Log attached:

[messages.txt](https://github.com/JMRI/JMRI/files/38896/messages.txt)

dheap, 2015-11-19:
> Please use a real 4.1.3 or 4.0.1; the logs indicate you used a 4.1.3ish version, which is not reliable for testing.

4.1.3 is fine and has no issues. 4.1.3ish is how the current HEAD of Git reports. That is what we are testing, not 4.1.3!

dheap, 2015-11-19:
There was no problem with 4.1.3.

The problem only began after your https://github.com/JMRI/JMRI/pull/113 was merged last week.

dheap, 2015-11-19:
I would have thought the current JMRI master should report 4.1.4ishxxxxxx. But it doesn't. It reports 4.1.3ishxxxxxx.

rhwood, 2015-11-19:
I am unable to recreate this issue going from 4.0.1 to HEAD of master, which leads me to think that we really do have a problem with 4.1.3.

dheap, 2015-11-19:
Can you recreate it going from 4.1.3 to HEAD of master?

My test case was created in 4.1.3, but all my many existing profiles were created long before 4.1.3, most before 4.0.1, but most  have probably been touched by 4.1.3. Can you see anything odd in the profile before HEAD of master tried to migrate it.

dheap, 2015-11-19:
I was able to reproduce it going directly from 4.0.1 to my checkout (updated last night so at https://github.com/JMRI/JMRI/pull/223). Will update and tray again (I've just woken up).

Updated to HEAD of master and tried again. Same problem and profile never been touched by 4.1.3. So its not a 4.1.3 problem, Here

 is the profile created by 4.01:
[4.0.1_Created.zip.txt](https://github.com/JMRI/JMRI/files/39365/4.0.1_Created.zip.txt)

dheap, 2015-11-19:
Just in case I had some problem in my local checkout, i tried the latest Jenkins build:http://builds.jmri.org/jenkins/job/Development/job/Packages/lastSuccessfulBuild/artifact/dist/release/JMRI.4.1.3-R3f58385.dmg

Same problem occurred.

Dick Bronson also reported a similar problem on the developers list. I sent a test build to Marcus Amman (testing QSI stuff) and h had the same problem.

dheap, 2015-11-19:
The default behaviour up to and including 4.1.3 was that Roster Location was unset. When unset, it should follow User Files Location (preference:) and that is how the code was working. Refer:
http://jmri.sourceforge.net/help/en/package/apps/TabbedPreferences.shtml#FileLocations
Also the procedure in:
http://jmri.sourceforge.net/help/en/html/Dropbox.shtml

Why then is a profile that has no reference to roster location in ProfileConfig.xml explicitly setting roster location to profile:\ in profile.properties during migration? If it needs to be set to anything explicit, it should be set to preference: 

Thinking corner cases. I can't test at present, but my preference: path has embedded spaces. Is that being handled correctly (seems to be)? Did your test use embedded spaces?

dheap, 2015-11-20:
I have now been able to completely reliably reproduce this problem on two Macs and a Windows 7 (32bit) machine with the following procedure:

* Use DecoderPro V4.0.1 (or V4.1.3) to create a new profile and when it restarts into the new profile make the following settings:
 * NCE Simulator
 * Preferences->Location->User Files Location: set to any directory **outside your normal JMRI settings folder**.
 * Restart and check Help->Locations. Both "User Files Location" and "Roster Location" should be the same and pointing to the directory you selected. Create a new loco if you selected a new empty folder.

* Using either of these:
 *  [Mac JMRI.4.1.3-R02524cd.dmg](http://builds.jmri.org/jenkins/job/Development/job/Packages/lastSuccessfulBuild/artifact/dist/release/JMRI.4.1.3-R02524cd.dmg)
 * [Windows JMRI.4.1.3-R02524cd.exe](http://builds.jmri.org/jenkins/job/Development/job/Packages/lastSuccessfulBuild/artifact/dist/release/JMRI.4.1.3-R02524cd.exe)
* Start DecoderPro. You will get a migration message.
 * Restart and check Help->Locations. Every time I try this:
 * "User Files Location" is unchanged from above but "Roster Location" has changed and is now the same as "Settings Location".
 * Your roster will be empty due to change of ""Roster Location"

Can **anyone** replicate my results?

dheap, 2015-11-20:
Merge https://github.com/JMRI/JMRI/pull/230 has fixed both Mac and Windows Roster Location issue, thanks.

closed 2015-11-20 by dheap

rhwood, 2015-11-20:
I know what #230 fixed, but still have no idea how the problem it fixed could be generated.

dheap, 2015-11-20:
I was able to generate the problem by simply creating a new profile under 4.0.1 or 4.1.3 and making the exact Wizard settings I documented, plus a simple change of User Files Location using the File Chooser tool in Preferences->Locations->User Files Location nothing more, nothing less. No manual typing of path or anything like that. Didn't touch Roster Location or even open Roster tab. Happened every time on Mac or Windows.

rhwood, 2015-11-20:
I followed your steps as documented and could not recreate it. When I say I don't understand, I mean I could not find a code path that would cause it.

dheap, 2015-11-20:
> I followed your steps as documented and could not recreate it.
That is odd.

> When I say I don't understand, I mean I could not find a code path that would cause it.
I realised that after commenting (it was late at night).