JMRI/JMRI #217: GUI tests fail in CI

rhwood, 2015-11-18:
The following tests are causing #213 to fail:

- [ ] testOptionFrameWrite(jmri.jmrit.operations.setup.OperationsSetupGuiTest)
- [ ] testCarsTableFrame(jmri.jmrit.operations.rollingstock.cars.CarsTableFrameTest)
- [ ] testTrainSwitchListEditFrame(jmri.jmrit.operations.trains.OperationsTrainsGuiTest)
- [ ] testCreateHorizontalNone(jmri.jmrit.progsupport.ProgServiceModePaneTest)
- [ ] testCreateVerticalNone(jmri.jmrit.progsupport.ProgServiceModePaneTest)

The logs are:
```
    [junit] Testcase: testOptionFrameWrite(jmri.jmrit.operations.setup.OperationsSetupGuiTest):	FAILED
    [junit] rfid
    [junit] junit.framework.AssertionFailedError: rfid
    [junit] 	at jmri.jmrit.operations.setup.OperationsSetupGuiTest.testOptionFrameWrite(OperationsSetupGuiTest.java:157)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$5.run(JFCTestCase.java:580)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$3.run(JFCTestCase.java:478)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
    [junit] 
    [junit] 
    [junit] Testcase: testCarsTableFrame(jmri.jmrit.operations.rollingstock.cars.CarsTableFrameTest):	FAILED
    [junit] 1st car in sort by rfid list expected:<NH 5> but was:<SP 2>
    [junit] junit.framework.AssertionFailedError: 1st car in sort by rfid list expected:<NH 5> but was:<SP 2>
    [junit] 	at jmri.jmrit.operations.rollingstock.cars.CarsTableFrameTest.testCarsTableFrame(CarsTableFrameTest.java:182)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$5.run(JFCTestCase.java:580)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$3.run(JFCTestCase.java:478)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
    [junit] 
    [junit] 
    [junit] Testcase: testTrainSwitchListEditFrame(jmri.jmrit.operations.trains.OperationsTrainsGuiTest):	FAILED
    [junit] Real Time
    [junit] junit.framework.AssertionFailedError: Real Time
    [junit] 	at jmri.jmrit.operations.trains.OperationsTrainsGuiTest.testTrainSwitchListEditFrame(OperationsTrainsGuiTest.java:583)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$5.run(JFCTestCase.java:580)
    [junit] 	at junit.extensions.jfcunit.JFCTestCase$3.run(JFCTestCase.java:478)
    [junit] 	at java.lang.Thread.run(Thread.java:745)
    [junit] 
    [junit] 
    [junit] Testcase: testCreateHorizontalNone(jmri.jmrit.progsupport.ProgServiceModePaneTest):	Caused an ERROR
    [junit] null
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrit.progsupport.ProgServiceModePane.<init>(ProgServiceModePane.java:102)
    [junit] 	at jmri.jmrit.progsupport.ProgServiceModePane.<init>(ProgServiceModePane.java:83)
    [junit] 	at jmri.jmrit.progsupport.ProgServiceModePaneTest.testCreateHorizontalNone(ProgServiceModePaneTest.java:27)
    [junit] 
    [junit] 
    [junit] Testcase: testCreateVerticalNone(jmri.jmrit.progsupport.ProgServiceModePaneTest):	Caused an ERROR
    [junit] null
    [junit] java.lang.NullPointerException
    [junit] 	at jmri.jmrit.progsupport.ProgServiceModePane.<init>(ProgServiceModePane.java:102)
    [junit] 	at jmri.jmrit.progsupport.ProgServiceModePaneTest.testCreateVerticalNone(ProgServiceModePaneTest.java:51)
    [junit] 
    [junit] 
    [junit] Test apps.tests.AllTest FAILED
```

DanielBoudreau, 2015-11-18:
I believe the failure is due to the panel not completely being shown on the screen. The first failure is due to a checkbox not being selected. It has to be showing in order to be selected by the operation:

 

getHelper().enterClickAndLeave(new MouseEventData(this, p.rfidCheckBox));

 

The panel in question uses a scrollbar, but on my machine the entire panel is shown. A smaller font size might fix this problem.

 

Dan

bobjacobsen, 2015-11-18:
There's info on changing the testing screen size in the "Using xvfb to Run Tests That Require a GUI #" section here: http://docs.travis-ci.com/user/gui-and-headless-browsers/

DanielBoudreau, 2015-11-18:
Are we now running all of the GUI tests?  I don’t see a failure? I tried to fix one of the operation tests.

 

Dan

rhwood, 2015-11-18:
Pull request 213 proposes it, and shows how.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

DanielBoudreau, 2015-11-18:
I understand that we’re now running all of the GUI tests. But three of the operations tests were failing. Now passing?

 

So did someone change the screen resolution for the tests?

 

Dan

 

randall.h.wood@alexandriasoftware.com <mailto:randall.h.wood@alexandriasoftware.com> 
http://alexandriasoftware.com

rhwood, 2015-11-18:
We are *not* running the GUI tests except in PR 213 and my (and apparently Bob Jacobsen's) forks. Look at the CI test logs, you will see that headless tests are the only tests being run against JMRI/master. (Note that neither Jenkins server runs GUI tests.)

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2015-11-18:
On Wed, Nov 18, 2015 at 12:49 PM, Daniel Boudreau <notifications@github.com>
wrote:

> I understand that we’re now running all of the GUI tests. But three of the
> operations tests were failing. Now passing?
>
> So did someone change the screen resolution for the tests?
>
>
Are you sure this is just because of the screen resolution?

If it is, then that test is fragile, and it be rebuilt to be more robust.

Paul

rhwood, 2015-11-18:
The screen resolution for the tests is 1280x1024x8 (SXGA, the 4th most common monitor size on the Steam gaming network per Wikipedia), so it's better to make tests work in this size than to bypass them at this size.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

DanielBoudreau, 2015-11-18:
Thanks Randall for the screen resolution. That was very helpful. I agree that the tests should all work properly at that resolution.

 

The three operation windows that fail all fit without scrollbars using that resolution on my PC.  Do you know the default font size used during the CI tests? On my machine the font looks like a 11 or 12 during the testing. The tests force the panels into the upper left hand corner of the screen, so I don’t quite understand why some of the checkboxes aren’t being displayed.

 

The other two windows have their most right hand checkboxes hidden, so most of the window is being displayed.

 

Dan

 

http://alexandriasoftware.com

pabender, 2015-11-18:
On Nov 18, 2015 3:49 PM, "Daniel Boudreau" <notifications@github.com> wrote:
>
> The issue is that if a checkbox isn’t visible, the ability to change the
checkbox selection fails. The windows all have scroll bars, and in this
case the window is too tall to fit the CI screen size. This of course
happens when you use a large font, or a low screen resolution.
>
>

Also ask yourself what are you testing.

If what you care about is what happens when the checkbox changes state,
then you don't necessarily need to "click" on the checkbox to change the
state, you can do that programmatically.

( actually clicking the box  is why you're using the GUI helper for the
test, and the helper  is what can't handle the window size.)

Paul

DanielBoudreau, 2015-11-18:
Good question. Less testing, less problems to fix! In many of the operation windows, a radio button or checkbox changes what is displayed, (features are grayed out, the table display order is changed, etc.) so I do want to test the action performed by changing the button state. Plus these tests are there to work the GIU and make sure everything works correctly. I’ve seen them catch issues in the past, so I think the testing is worthwhile.

 

Dan

DanielBoudreau, 2015-11-18:
How do I see which test fail in PR 213? Can I push changes to this branch?

 

Dan

 

http://alexandriasoftware.com

pabender, 2015-11-19:
On 11/18/2015 04:19 PM, Daniel Boudreau wrote:
> Good question. Less testing, less problems to fix! 
That wasn't what the question was asking though.

> In many of the operation windows, a radio button or checkbox changes 
> what is displayed, (features are grayed out, the table display order 
> is changed, etc.) so I do want to test the action performed by 
> changing the button state. 
Exactly, but what I was implying is that there is more than one way to 
test that.

You're using a "GUI Helper" that does some work to actually click the 
button.  This gives you code that looks like this:
         getHelper().enterClickAndLeave(new MouseEventData(this, 
p.rfidCheckBox));
         Assert.assertTrue("rfid", p.rfidCheckBox.isSelected());

(which is one of the cases that was failing...)

I'm pretty sure you can get the same result with:

        p.rfidCheckBox.doClick();
        Assert.assertTrue("rfid", p.rfidCheckBox.isSelected());
or
         p.rfidCheckBox.processMouseEvent(new MouseEvent(p,
java.util.Calendar.getInstance().getTimeInMillis()
,0,p.rfidCheckBox.getX(),p.rfidCheckBox.getY(),1,MouseEvent.BUTTON1);
          )));
         Assert.assertTrue("rfid", p.rfidCheckBox.isSelected());

The first version just performs the actions of a click.  The second form 
actually sends the mouse event (I may have that MouseEvent constructor 
off slightly. I didn't actually test that code).

I know the "doClick" version doesn't care if the JCheckbox is on the 
screen or not.

> Plus these tests are there to work the GIU and make sure everything 
> works correctly. I’ve seen them catch issues in the past, so I think 
> the testing is worthwhile.

And that's exactly the point of unit level regression testing, so that's 
a good thing.

Paul

DanielBoudreau, 2015-11-19:
There was a problem using doClick() but I don’t remember what the issue was. It was many years ago.  Bob J. helped me convert to the getHelper() method, maybe he remembers why we changed, it might be that the doClick() doesn’t cause a property change event?

 

I’ve been trying to fix the test code today. I tried scrolling the window so the checkbox shows, but the getHelper().enterClickAndLeave() causes the window to scroll back and I end up with the component not showing.  I even tried disabling the scrollbar after getting the component to show, but the getHelper().enterClickAndLeave() still causes a change in the view.

 

Dan

bobjacobsen, 2015-11-19:
>There was a problem using doClick() but I don’t remember what the issue was. It was many years ago.  Bob J. helped me convert to the getHelper() method, maybe he remembers why we changed, it might be that the doClick() doesn’t cause a property change event?

doClick() does (or at least should) do an action event. Not sure about property change event.  And it doesn't pay attention to e.g. whether the widget is enabled/disabled, etc.

DanielBoudreau, 2015-11-19:
I think I was able to fix at least two of the three failures.  I don’t understand why the third test is failing.

 

Could you try merging in #228 to your test PR and let me know if there’s still an issue?

 

Dan

rhwood, 2015-11-21:
Since #213 has been merged into master, any further failures should be dealt with as they occur.

closed 2015-11-21 by rhwood