JMRI/JMRI #229: Preferences window and help opens on startup when using NCE simulator as connection

DanielBoudreau, 2015-11-19:
Recently introduced bug.

rhwood, 2015-11-20:
This needs more information, since it cannot be recreated by:
 1. Create new profile with only an NCE Simulator connection in DecoderPro 4.0.1
 2. Quit DecoderPro
 3. Open DecoderPro from HEAD of JMRI/master with the new profile so it can migrate
 4. Quit DecoderPro
 5. Open DecoderPro from HEAD of JMRI/master with the new profile

DanielBoudreau, 2015-11-20:
I’m using PanelPro.  I still have two issues with the latest commits using PanelPro, the GUI font size doesn’t work, it is saved properly, but the display doesn’t change. Second, when using the NCE with simulator, the help and preferences window opens on start up.  The NCE simulator seems to work just fine.

 

Dan

DanielBoudreau, 2015-11-20:
I did the same sequence with the latest commits, and the preference window still pops up using PanelPro or DecoderPro.

 

I’ve attached the files for the new profile that I created.

 

Dan

DanielBoudreau, 2015-11-21:
Seems to be fixed.  Still have an issue with the GUI font size not working properly.

closed 2015-11-21 by DanielBoudreau