JMRI/JMRI #232: Unable to create NCE connection using initial configuration wizard

rhwood, 2015-11-20:
In a new Profile, creating an NCE simulator connection using the initial configuration wizard in DecoderPro (HEAD from master) throws a NullPointerException that blocks further activity. Creating an NCE simulator connection in the Preferences window does not throw a NullPointerException. Subsequent launches do not cause the Preferences window to open (#229).
Creating a LocoNet simulator connection in either the wizard or the Preferences window does not have this effect.

dheap, 2015-11-20:
That is odd because it was exactly the procedure I was using to create my tests for https://github.com/JMRI/JMRI/issues/193/ and I didn't get blocked (being a new profile I didn't have System Console open so would not have noticed an NPE).

dheap, 2015-11-20:
There have been some ongoing odd issues when adding a new NCE connection to an existing profile using the Preferences->Connections->**+** tab.

Things like the connection type failing to show, but not necessarily repeatable.

bobjacobsen, 2015-11-20:
This is sounding more like there's a problem in the NCE infrastructure code rather than the preferences code.  Can take a look late today.

bobjacobsen, 2015-11-22:
Would be interested to hear if #252 fixes this

bobjacobsen, 2015-11-23:
I think this specific issue (problem with wizard) is fixed by #252

bobjacobsen, 2015-11-30:
This seems to be OK with current HEAD of JMRI/JMRI master. I tried a new profile on both DecoderPro (with the wizard) and PanelPro (manual configure); all OK.

mattharris, 2015-12-04:
Closing this as part of release process

closed 2015-12-04 by mattharris