JMRI/JMRI #236: NPE when trying to change connection preferences

DanielBoudreau, 2015-11-21:
If I change the connection from NCE simulator to serial or any other connection and press "Save" it causes an NPE.  Afterwards I'm not able to restore the connection back to simulator, as that also causes an NPE.  So I'm not able to close the preferences window.

Here's the trace:

2015-11-21 09:55:34,793 jmrix.AbstractSerialPortController    ERROR - No usable ports returned [AWT-EventQueue-0]
2015-11-21 09:55:34,793 ptionhandler.UncaughtExceptionHandler ERROR - Unhandled Exception: java.lang.NullPointerException [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrix.configurexml.AbstractConnectionConfigXml.storeCommon(AbstractConnectionConfigXml.java:37)
	at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.store(AbstractSerialConnectionConfigXml.java:44)
	at jmri.jmrix.configurexml.AbstractConnectionConfigXml.store(AbstractConnectionConfigXml.java:33)
	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:515)
	at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:101)
	at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:90)
	at jmri.jmrix.swing.ConnectionsPreferencesPanel.savePreferences(ConnectionsPreferencesPanel.java:283)
	at apps.gui3.TabbedPreferences.invokeSaveOptions(TabbedPreferences.java:230)
	at apps.gui3.TabbedPreferences.lambda$0(TabbedPreferences.java:147)
	at apps.gui3.TabbedPreferences$$Lambda$29/1092348790.actionPerformed(Unknown Source)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3321)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2739)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:746)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:719)
	at java.awt.EventQueue$4.run(EventQueue.java:717)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

DanielBoudreau, 2015-11-21:
Additional testing has revealed that I can't set other connections like a Digitrax simulaor without getting an NPE.  So the issue is there's an NPE if there isn't a port specified in the GUI.

rhwood, 2015-11-21:
Something is setting the SerialPortAdapter to null in the AbstractSerialConnectionConfig, but I haven't figured out what is yet.

pabender, 2015-11-21:
On Nov 21, 2015 11:25 AM, "Randall Wood" <notifications@github.com> wrote:
>
> Something is setting the SerialPortAdapter to null in the
AbstractSerialConnectionConfig, but I haven't figured out what is yet.
>
>

Since this is the NCE simulator we are talking about, my guess is it is the
simulator itself doing this.  I seem to recall that adapter ( and the CMRI
simulator ) requiring a port to be selected at one point in time.

Paul

DanielBoudreau, 2015-11-21:
Stilling seeing NPE if no connection is specified.  But NCE simulator is now working, preference window now closes correctly.

Trace:

2015-11-21 18:35:27,522 ptionhandler.UncaughtExceptionHandler ERROR - Unhandled Exception: java.lang.NullPointerException [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrix.AbstractSimulatorConnectionConfig.getConnectionName(AbstractSimulatorConnectionConfig.java:247)
	at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:100)
	at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:90)
	at jmri.jmrix.swing.ConnectionsPreferencesPanel.savePreferences(ConnectionsPreferencesPanel.java:283)
	at apps.gui3.TabbedPreferences.invokeSaveOptions(TabbedPreferences.java:230)
	at apps.gui3.TabbedPreferences.lambda$0(TabbedPreferences.java:147)
	at apps.gui3.TabbedPreferences$$Lambda$28/60884742.actionPerformed(Unknown Source)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3321)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2739)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:746)
	at java.awt.EventQueue.access$400(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:697)
	at java.awt.EventQueue$3.run(EventQueue.java:691)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:719)
	at java.awt.EventQueue$4.run(EventQueue.java:717)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

bobjacobsen, 2015-11-22:
There seems to be several things going on that are causing various NPEs.  

One is that, for some as-yet unknown reason, calls to AbstractSimulatorConnectionConfig.setInstance(..) that used to happen no longer are.  That's the underlying cause of NPEs when AbstractSimulatorConnectionConfig.getConnectionName() is called.  It seems to be hitting all simulator connections I've checked.

We can bypass the symptom by protecting against the NPE, and I'll PR that shortly.  But there's something more basic going on.

Second, there seems to be something NCE-specific happening too.  Working on that now.

DanielBoudreau, 2015-11-22:
I’ve only been testing with the NCE simulator, which has been working fine with the current set of commits.  So I don’t see any NCE-specific issues.

 

Dan

bobjacobsen, 2015-11-22:
Would be interested to hear if #252 fixes this

DanielBoudreau, 2015-11-22:
Changing the manufacture from NCE to “None” produces the following NPE when the “Save” button is pressed:

 

2015-11-22 15:49:02,037 ptionhandler.UncaughtExceptionHandler ERROR - Unhandled Exception: java.lang.NullPointerException [AWT-EventQueue-0]

java.lang.NullPointerException

       at jmri.jmrix.configurexml.AbstractConnectionConfigXml.storeCommon(AbstractConnectionConfigXml.java:37)

       at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.store(AbstractSerialConnectionConfigXml.java:44)

       at jmri.jmrix.configurexml.AbstractConnectionConfigXml.store(AbstractConnectionConfigXml.java:33)

       at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:515)

       at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:101)

       at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:90)

       at jmri.jmrix.swing.ConnectionsPreferencesPanel.savePreferences(ConnectionsPreferencesPanel.java:283)

       at apps.gui3.TabbedPreferences.invokeSaveOptions(TabbedPreferences.java:230)

       at apps.gui3.TabbedPreferences.lambda$0(TabbedPreferences.java:147)

       at apps.gui3.TabbedPreferences$$Lambda$25/1179654193.actionPerformed(Unknown Source)

       at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)

       at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)

       at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)

       at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)

       at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)

       at java.awt.Component.processMouseEvent(Component.java:6525)

       at javax.swing.JComponent.processMouseEvent(JComponent.java:3321)

       at java.awt.Component.processEvent(Component.java:6290)

       at java.awt.Container.processEvent(Container.java:2234)

       at java.awt.Component.dispatchEventImpl(Component.java:4881)

       at java.awt.Container.dispatchEventImpl(Container.java:2292)

       at java.awt.Component.dispatchEvent(Component.java:4703)

       at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)

       at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)

       at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)

       at java.awt.Container.dispatchEventImpl(Container.java:2278)

       at java.awt.Window.dispatchEventImpl(Window.java:2739)

       at java.awt.Component.dispatchEvent(Component.java:4703)

       at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:746)

       at java.awt.EventQueue.access$400(EventQueue.java:97)

       at java.awt.EventQueue$3.run(EventQueue.java:697)

       at java.awt.EventQueue$3.run(EventQueue.java:691)

       at java.security.AccessController.doPrivileged(Native Method)

       at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)

       at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)

       at java.awt.EventQueue$4.run(EventQueue.java:719)

       at java.awt.EventQueue$4.run(EventQueue.java:717)

       at java.security.AccessController.doPrivileged(Native Method)

       at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)

       at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)

       at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)

       at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)

       at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)

       at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)

       at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)

       at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

pabender, 2015-11-22:
Bob,

On 11/22/2015 02:23 PM, Bob Jacobsen wrote:
> One is that, for some as-yet unknown reason, calls to
> AbstractSimulatorConnectionConfig.setInstance(..) that used to happen no
> longer are. That's the underlying cause of NPEs when [...]

I can confirm both that the problem was happening on the XPressNet 
Simulator before you caught the NPE AND that the problem doesn't happen 
now that the NPE is caught.

Now let's figure out what changed that caused this.

Paul
-- 
______________________________________________________________________________

	"Quality is a Characteristic of thought and statement that
	is recognized by a nonthinking process. Because
	definitions are a product of rigid formal thinking,
	quality cannot be defined."
		Robert M. Pirsig
			Zen and The Art of Motorcycle Maintenance

DanielBoudreau, 2015-11-22:
Paul,

 

Can you now change the manufacture from XPressNet to “None” without an NPE?

 

Dan

rhwood, 2015-11-22:
This behavior was most likely introduced in one of the commits listed at https://github.com/JMRI/JMRI/pull/113/commits

pabender, 2015-11-22:
Dan,

On 11/22/2015 04:53 PM, Daniel Boudreau wrote:
> Can you now change the manufacture from XPressNet to “None” without an NPE?

Yes, I can change the manufacturer to none without an NPE.

Paul
-- 
______________________________________________________________________________

	"Quality is a Characteristic of thought and statement that
	is recognized by a nonthinking process. Because
	definitions are a product of rigid formal thinking,
	quality cannot be defined."
		Robert M. Pirsig
			Zen and The Art of Motorcycle Maintenance

DanielBoudreau, 2015-11-22:
I can do anything using the latest commits, I’m stuck with NCE simulator.  Any change causes an NPE. Can you try NCE simulator and see if you run into the same issues?

 

Dan

DanielBoudreau, 2015-11-22:
Just tried using the LocoNet simulator, having the same issue, NPEs after changing the manufacture to “None” or any other manufacture and pressing Save.

 

Dan

DanielBoudreau, 2015-11-22:
One thing that might be different on my system is that I don’t have any serial ports available. Just a standalone PC.

 

Dan

bobjacobsen, 2015-11-23:
I am completely confused by "shared connections".  

Sorry to be obtuse about this, but I just don't get what's supposed to be shared and what's not, and how the program knows.  Could somebody point me to a writeup on this, please?

I ask because I keep running across aspects of it in the load/init/store code, and I don't want to break anything.

Would be happy to add whatever info is available either to the [XML Persistence](http://jmri.org/help/en/html/doc/Technical/XmlPersistance.shtml), the [profiles user page](http://jmri.org/help/en/html/profiles.shtml) or where-ever thought useful, just so long as Google can find it.

rhwood, 2015-11-23:
The changes to the configuration are step one of solving issue #69. "Shared" is configuration data that all computers should get (currently everything) and "perNode" is per-computer configuration that overrides "Shared"

The connections should be writing everything to shared and perNode, and reading both, ignoring any perNode configuration that does not match a shared connection by system prefix.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

DanielBoudreau, 2015-11-23:
Still seeing NPE with latest commits. All I have to do is change the manufacture to “None”.  Pretty sure the “adapter” in line 37 is null.

 

Dan

 

2015-11-22 20:24:33,339 ptionhandler.UncaughtExceptionHandler ERROR - Unhandled Exception: java.lang.NullPointerException [AWT-EventQueue-0]

java.lang.NullPointerException

       at jmri.jmrix.configurexml.AbstractConnectionConfigXml.storeCommon(AbstractConnectionConfigXml.java:37)

       at jmri.jmrix.configurexml.AbstractSerialConnectionConfigXml.store(AbstractSerialConnectionConfigXml.java:44)

       at jmri.jmrix.configurexml.AbstractConnectionConfigXml.store(AbstractConnectionConfigXml.java:33)

       at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:515)

       at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:101)

       at jmri.jmrix.ConnectionConfigManager.savePreferences(ConnectionConfigManager.java:90)

       at jmri.jmrix.swing.ConnectionsPreferencesPanel.savePreferences(ConnectionsPreferencesPanel.java:283)

       at apps.gui3.TabbedPreferences.invokeSaveOptions(TabbedPreferences.java:230)

       at apps.gui3.TabbedPreferences.lambda$0(TabbedPreferences.java:147)

       at apps.gui3.TabbedPreferences$$Lambda$25/336361095.actionPerformed(Unknown Source)

       at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)

       at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)

       at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)

       at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)

       at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)

       at java.awt.Component.processMouseEvent(Component.java:6525)

       at javax.swing.JComponent.processMouseEvent(JComponent.java:3321)

       at java.awt.Component.processEvent(Component.java:6290)

       at java.awt.Container.processEvent(Container.java:2234)

       at java.awt.Component.dispatchEventImpl(Component.java:4881)

       at java.awt.Container.dispatchEventImpl(Container.java:2292)

       at java.awt.Component.dispatchEvent(Component.java:4703)

       at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)

       at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)

       at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)

       at java.awt.Container.dispatchEventImpl(Container.java:2278)

       at java.awt.Window.dispatchEventImpl(Window.java:2739)

       at java.awt.Component.dispatchEvent(Component.java:4703)

       at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:746)

       at java.awt.EventQueue.access$400(EventQueue.java:97)

       at java.awt.EventQueue$3.run(EventQueue.java:697)

       at java.awt.EventQueue$3.run(EventQueue.java:691)

       at java.security.AccessController.doPrivileged(Native Method)

       at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)

       at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)

       at java.awt.EventQueue$4.run(EventQueue.java:719)

       at java.awt.EventQueue$4.run(EventQueue.java:717)

       at java.security.AccessController.doPrivileged(Native Method)

       at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)

       at java.awt.EventQueue.dispatchEvent(EventQueue.java:716)

       at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)

       at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)

       at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)

       at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)

       at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)

       at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

 

randall.h.wood@alexandriasoftware.com <mailto:randall.h.wood@alexandriasoftware.com> 
http://alexandriasoftware.com

pabender, 2015-11-23:
On 11/22/2015 06:29 PM, Bob Jacobsen wrote:
> I am completely confused by "shared connections".
>
> Sorry to be obtuse about this, but I just don't get what's supposed to
> be shared and what's not, and how the program knows. Could somebody
> point me to a writeup on this, please?

I think Randall is the only one who really understands this.

The intention (ultimately) is that a user will be able to load a profile 
to dropbox (or some other file sharing service), and run that profile 
from multiple machines, but allowing individual machines to define what 
connections map to each system prefix in the profile.

I still argue that is the wrong approach, unless the machines are 
homogeneous.  In the homogeneous case, one set of configuration files 
suffices for all machines involved.

Paul
-- 
______________________________________________________________________________

	"Quality is a Characteristic of thought and statement that
	is recognized by a nonthinking process. Because
	definitions are a product of rigid formal thinking,
	quality cannot be defined."
		Robert M. Pirsig
			Zen and The Art of Motorcycle Maintenance

bobjacobsen, 2015-11-23:
#257 is all I can do on this for the next couple days. See the PR comment for a description of what I did.

It is very clear that this code doesn't have a clear concept when a missing object (null reference) is an error, and when it's routine (e.g. due to migration from an earlier form).  That PR fixes a couple cases I encountered while testing, but without a clear idea of what's supposed to be around when, I have no confidence that anything is really fixed.

In any case, I'm done.  If somebody else can't make this reliable, I suggest we back it all out.

Bob

dheap, 2015-11-23:
Has anyone checked to make sure the problem was not already in 4.1.3? I know there were issues around NCE connections preferences in earlier versions.

Sent from my iPad
-- 
Dave 

> On 23 Nov 2015, at 1:01 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> It is very clear that this code doesn't have a clear concept when a missing object (null reference) is an error, and when it's routine (e.g. due to migration from an earlier form). That PR fixes a couple cases I encountered while testing, but without a clear idea of what's supposed to be around when, I have no confidence that anything is really fixed.
> 
> In any case, I'm done. If somebody else can't make this reliable, I suggest we back it all out.

DanielBoudreau, 2015-11-23:
Dave,

 

Works fine in 4.1.3.  And I’m also having problems with other manufactures.

 

Dan

 

Dave

DanielBoudreau, 2015-11-23:
Still have issues. Can’t change the connection, no NPE.  I’m going away for a few days, need to let someone else test this.  Done for now. 

 

Dan

rhwood, 2015-11-23:
> On 22 Nov 2015, at 20:30, Paul Bender <notifications@github.com> wrote:
> 
> The intention (ultimately) is that a user will be able to load a profile 
> to dropbox (or some other file sharing service), and run that profile 
> from multiple machines, but allowing individual machines to define what 
> connections map to each system prefix in the profile.
> 
> I still argue that is the wrong approach, unless the machines are 
> homogeneous. In the homogeneous case, one set of configuration files 
> suffices for all machines involved.

It is impossible to create a single Profile that can support swapping two LocoBuffer-USB devices (and most likely other FTDI-based USB-serial devices) in OS X, or using two different MacBooks with two different LocoBuffer-USB devices. This is because the LocoBuffer-USB devices include a serial number that is exposed to the FTDI drivers, so OS X generates devices with the in form tty.usbserial-<SERIAL#>, which change for each device. Some versions of udev on Linux do similar things to ensure that (for example) /dev/ttyAMA0 is always tied to only a single FTDI device, and other devices cannot be /dev/ttyAMA0. Note that this is considered a feature, in that, no matter how an FTDI device is connected, it always gets the same name.

This in contrast to the Prolific PL2303-based devices I have, which lack serial numbers, and are given the identity of the USB bus they are connected to on OS X, or are enumerated as /dev/ttyUSB# based on how the USB bus is traversed.

Basically, there are serial devices we support, the existence of which, can guarentee non-homogeneous machines.

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

pabender, 2015-11-23:
On 11/23/2015 08:56 AM, Randall Wood wrote:
> It is impossible to create a single Profile that can support swapping
> two LocoBuffer-USB devices (and most likely other FTDI-based USB-serial
> devices) in OS X, or using two different MacBooks with [...]

That may be true, but it is also possible to create a symbolic link to 
the device, giving it a common name.  We can then use the symbolic links 
for communication.

> Some versions of udev on Linux do similar things to
> ensure that (for example) /dev/ttyAMA0 is always tied to only a single
> FTDI device, and other devices cannot be /dev/ttyAMA0. Note that this is
> considered a feature, in that, no matter how an FTDI device is
> connected, it always gets the same name.

For serial devices with serial numbers, It's actually possible to have 
the udev rules create a device with a specific name.  I know many people 
have recommended making a /dev/locobuffer device.

> This in contrast to the Prolific PL2303-based devices I have, which lack
> serial numbers, and are given the identity of the USB bus they are
> connected to on OS X, or are enumerated as /dev/ttyUSB# based on how the
> USB bus is traversed.

Yes, but that is an issue for all USB implementations at some point. 
Enumeration order can change the underlying device names.  At least one 
MacOS and Linux, once the order of enumeration at start up is known, you 
can just build symbolic links to get the device names you want.

Windows is worse.  Many versions will give a different port name to the 
same device, even if it has a serial number, based on where it is 
plugged in.

Paul

rhwood, 2015-12-01:
Fixed by #280

closed 2015-12-01 by rhwood