JMRI/JMRI #259: Alternative to ArduinoCMRI?

k4kfh, 2015-11-23:
I am trying to get a railroad crossing working on my layout, and I'd like to use Arduino to do it so the occupancy detection system will function within JMRI. I know about the Arduino CMRI library, but it only works with certain Arduino models with an FTDI chip due to some incompatibility in JMRI.

I tried to set up the connection type "Others" today, but it was very buggy and crashed pretty frequently. How should I go about setting this up? Could JMRI maybe make a guide to do this?

pabender, 2015-11-24:
Which arduino are you trying to use?

Paul

k4kfh, 2015-11-24:
I was hoping to use the Uno, since it is so widely available and dirt cheap. It doesn't have an FTDI chip though, it has some replacement from Atmel that doesn't quite behave the same. I'm no expert on this, I just know that according to [ArduinoCMRI's README.md](http://github.com/madleech/ArduinoCMRI) the Uno, Mega 2560, and some others won't work due to this replacement chip not playing nice with JMRI.

bobjacobsen, 2015-11-24:
You might want to check whether the Arduino statement is actually correct.  JMRI has been setting RTS and DTR active on C/MRI serial connections since 2003:

https://github.com/JMRI/JMRI/blame/master/java/src/jmri/jmrix/cmri/serial/serialdriver/SerialDriverAdapter.java

pabender, 2015-11-24:
On 11/23/2015 06:46 PM, Hampton wrote:
> I was hoping to use the Uno, since it is so widely available and dirt
> cheap. It doesn't have an FTDI chip though, it has some replacement from
> Atmel that doesn't quite behave the same. I'm no expert [...]

I'm not sure how valid that statement is.

You may want to join the Arduini list on yahoogroups:

https://groups.yahoo.com/neo/groups/Arduini/info

The Arduini group talks about various uses of Arduino devices in model 
railroading, including using them for C/MRI.

Paul
-- 
______________________________________________________________________________

	"Quality is a Characteristic of thought and statement that
	is recognized by a nonthinking process. Because
	definitions are a product of rigid formal thinking,
	quality cannot be defined."
		Robert M. Pirsig
			Zen and The Art of Motorcycle Maintenance

bobjacobsen, 2015-11-24:
Might want to look at some other CMRI implementations for the Arduino, too:

[davidzuhn/cpNode](https://github.com/davidzuhn/cpNode)
[davidzuhn/CMRI](https://github.com/davidzuhn/CMRI)

(David has lots of good stuff on GitHub)

k4kfh, 2015-11-25:
## Update - You're right!

I got curious this evening and tested the [madleech/arduinoCMRI](https://github.com/madleech/ArduinoCMRI) library with an Uno I had laying around. Much to my surprise, it worked just fine. I set up a pin connected to an LED on the Arduino as output bit 0, then set that up in JMRI as a light, and I was able to toggle the LED on and off without any issues. I didn't test inputs, but I assume those work as well.

Thanks guys!

closed 2015-11-25 by k4kfh