JMRI/JMRI #261: Making custom roster attribute using WebSockets?

k4kfh, 2015-11-24:
I am developing a software to simulate realistic acceleration/deceleration and controls for models, similar to what Bruce Kingsley has done with his "Ultimate Throttle" project. I'm building this based around JMRI's WebSockets server, which has worked excellent so far.

Obviously to simulate scale speed correctly, information that is specific to a particular model is needed. There are also a few other things that it would be very nice to store in the JMRI roster.

I am wondering if there is a way to create a custom roster entry via WebSockets. Can someone shed some light on this? It does not seem to be documented on the WebSockets page on the JMRI.org site.

rhwood, 2015-11-24:
There is no way to write anything to a roster entry via the WebSockets or JSON web server in the current code.

I'm going to interpret this as an enhancement request to make roster entries readable and writable via JSON. I am certain this will not make it into JMRI until sometime next year at the earliest.

KenC57, 2015-11-24:
Randall,

 

Did JSON/Websockets have a way to call a script?? And pass arguments?? If so
a workaround could be done that way to manage the parameters he needs to
cache in the roster entry.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

rhwood, 2015-11-24:
There are no mechanisms for any writes other than throttle + consist changes, and known sensor, route, turnout, and memory states changes.

k4kfh, 2015-11-24:
@rhwood Thank you! It's no huge hurry, I've found another way to do what I want for now, but this feature will be excellent when done.

rhwood, 2016-03-10:
JMRI 4.3.4 has a modularized JSON server that implements a Service Provider Interface (SPI) that allows third-party code packaged in a JAR to extend the JSON API with new tokens or actions. This is partial in that I have not yet reimplemented all the old tokens to use the new SPI. Once the roster handler is reimplemented, it can be extended to handle additional properties.

rhwood, 2016-05-02:
JMRI 4.3.5 has a modular Roster handler, so modular/multiple handlers of a Roster object can now be created and used.

rhwood, 2018-03-16:
Assignee is no longer involved.

closed 2021-04-14 by stale[bot]