JMRI/JMRI #272: Connections in Preferences Window is empty on first run.

rhwood, 2015-11-25:
The first time a Profile is used in DecoderPro, after the First Time Wizard is used to configure the first connection, that connection panel is empty, but present, in the Preferences window until after DecoderPro is restarted.

From the 02f2236 commit message:
> Note that these changes do not stop the preferences window from displaying an empty pane for the first connection until after JMRI is restarted. I suspect we need a repaint or similar call when the
TabbedPreferences window is opened to fix that.

This is confirmed to be present in 3.8 (that's as far back as I tested).

bobjacobsen, 2016-05-02:
Thanks for finding this.

I don't think a repaint() call is the issue. Resizing and overhauling the window doesn't fix it; the other panes are OK.  Looks more like some notification hasn't made it from the underlying setup (the effective Model part) to the GUI objects (the View).  But I don't see how that's supposed to work.

rhwood, 2016-11-27:
It appears the underly issues are:
1. ConnectionsPreferencesPanel does not listen to changes in ConnectionConfigManager indexed property "connections", so the ConnectionsPreferencesPanel instance in the TabbedPreferences never sees the new connection in the wizard.
2. The JmrixConfigPane maintains a single shared list of JmrixConfigPanes (the per-connection configuration panels), and, IIRC, a single Swing component can not be reused. (If I make the wizard dispose the JmrixConfigPane it uses, the TabbedPreferences shows an empty, unconfigured JmrixConfigPane.)

PR #2604 (2016-11-27): Remove static list of connection configuration panels
Fixes #272 by removing the list of connection configuration panels from JmrixConfigPane and having consumers of the configuration panels (TabbedPreferences, FirstTimeStartupWizard, and AppsConfigBase) either maintain their own sets of configuration panels or consume connection configuration information directly from ConnectionConfigManager.
closed 2016-11-27 by rhwood