JMRI/JMRI #276: Potential WebSockets bug: connection drops when requesting throttle?

k4kfh, 2015-11-25:
I have run into a snag while trying to develop a WebSockets-based JMRI app. Here is an explanation of what causes it:

1. I request a throttle, let's say for "CBQ2" on address 2. This works fine, and I can control it just fine.

2. I close out of my app and open it again, thus disconnecting and reconnecting to JMRI.

3. I request a throttle again, for the same locomotive.

4. JMRI throws a number of Jetty-related errors to the console (I'd say more but I don't know Java) and terminates the WebSockets connection.

5. After this, I cannot access this locomotive's throttle via WebSockets at all, not even through JMRI's built-in webThrottle app. When I try, the page just doesn't do anything. I can control it fine from my JMRI system, or directly through LocoNet.

This could be due to an issue in my own code, but I'm not sure. The only thing I'm not sure about is "letting go" of the throttle. On disconnecting, is there a certain JSON command I have to send to "drop" my throttle(s), or should JMRI just do that automatically?

Interestingly, following the exact same procedure with the exact same roster entry, I am unable to reproduce the bug in JMRI 3.10.1.

bobjacobsen, 2015-11-26:
> JMRI throws a number of Jetty-related errors to the console

Could you a attach a copy of the log, please?  Those messages might contain a clue.

k4kfh, 2015-11-26:
```
2015-11-25 18:48:17,694 json.JsonServlet                      ERROR - Timeout on Read [WebServer-55]
2015-11-25 18:48:17,712 json.JsonServlet                      ERROR - Timeout on Read [WebServer-100]
2015-11-25 18:48:17,728 json.JsonServlet                      ERROR - Timeout on Read [WebServer-68]
2015-11-25 18:48:21,661 json.JsonServlet                      ERROR - Timeout on Read [WebServer-53]
2015-11-25 18:49:04,317 json.JsonServlet$JsonWebSocket        WARN  - Unhandled Error (closing connection) [WebServer-65]
org.eclipse.jetty.websocket.api.WebSocketException: RemoteEndpoint unavailable, current state [CLOSED], expecting [OPEN or CONNECTED]
	at org.eclipse.jetty.websocket.common.WebSocketSession.getRemote(WebSocketSession.java:252)
	at jmri.jmris.JmriConnection.sendMessage(JmriConnection.java:103)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:55)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:60)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:280)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:274)
	at jmri.jmris.json.JsonThrottle.getThrottle(JsonThrottle.java:102)
	at jmri.jmris.json.JsonThrottleServer.parseRequest(JsonThrottleServer.java:81)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:299)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:151)
	at jmri.web.servlet.json.JsonServlet$JsonWebSocket.onMessage(JsonServlet.java:749)
	at sun.reflect.GeneratedMethodAccessor11.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:70)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:225)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:64)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:217)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:160)
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:309)
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:214)
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:220)
	at org.eclipse.jetty.websocket.common.Parser.parse(Parser.java:258)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:613)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:468)
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Thread.java:745)
2015-11-25 18:49:04,331 json.JsonServlet                      ERROR - RemoteEndpoint unavailable, current state [CLOSED], expecting [OPEN or CONNECTED] [WebServer-65]
org.eclipse.jetty.websocket.api.WebSocketException: RemoteEndpoint unavailable, current state [CLOSED], expecting [OPEN or CONNECTED]
	at org.eclipse.jetty.websocket.common.WebSocketSession.getRemote(WebSocketSession.java:252)
	at jmri.jmris.JmriConnection.sendMessage(JmriConnection.java:103)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:55)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:60)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:280)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:274)
	at jmri.jmris.json.JsonThrottle.getThrottle(JsonThrottle.java:102)
	at jmri.jmris.json.JsonThrottleServer.parseRequest(JsonThrottleServer.java:81)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:299)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:151)
	at jmri.web.servlet.json.JsonServlet$JsonWebSocket.onMessage(JsonServlet.java:749)
	at sun.reflect.GeneratedMethodAccessor11.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:70)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:225)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:64)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:217)
[... cut ...]
2015-11-25 18:48:17,694 json.JsonServlet                      ERROR - Timeout on Read [WebServer-55]
```

I have wiThrottle Server set to run on startup, and the "Timeout on Read" error shows up immediately after:

```
INFO  - Creating new WiThrottle DeviceServer(socket) on port 25566, waiting for incoming connection... [WiThrottleGUIServer]
```

If you need me to post any additional info/logs, just let me know. I can also post code from the app I'm getting these errors with if you'd like, but since it doesn't occur in the last production release I'm assuming it's not my code.

rhwood, 2015-11-26:
Can you try editing the default.lcf file in the JMRI distribution to include the line:

log4j.category.jmri.web.servlet.json.JsonServlet=DEBUG

This will give us a more verbose logging.

The first message you should see from the JSON servlet is the (unfortunately undocumented) “hello” message:

{
  "type": "hello",
  "data": {
    "JMRI": "4.1.5ish-201511260216-rhwood-R463049d",
    "json": "3.0",
    "heartbeat": 13500,
    "railroad": "Remote Config",
    "node": "jmri-001E52C9A2E3-3f4c010e",
    "activeProfile": "Remote Config"
  }
}

You will need to send a “ping” or other message note less frequently than the heartbeat timeout listed in the hello message (in milliseconds). Are you? (A ping looks like {"type":"ping”} if observed between JMRI and your client).

Between the two production versions of JMRI, we switched from using Jetty 8 to using Jetty 9 as the underlying WebSocket server. Perhaps the Jetty documentation might shed some light on what changed.

Randall, sorry about the brevity of response, I haven’t the time right now to dive deeper.

> 2015-11-25 18:48:17,694 json.JsonServlet                      ERROR - Timeout on Read [WebServer-55]
> 2015-11-25 18:48:17,712 json.JsonServlet                      ERROR - Timeout on Read [...]

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

k4kfh, 2015-11-26:
> The first message you should see from the JSON servlet is the (unfortunately undocumented) “hello” message

Yes, I actually use data from that (the railroad name) in my software, so I know I'm getting that message.

> You will need to send a “ping” or other message note less frequently than the heartbeat timeout listed in the hello message (in milliseconds). Are you?

Yes, I am sending pings at a regular interval. I checked just now to be sure, and I am also getting the "pong" replies back from the server. This is the case with both versions of JMRI.

When I edited the ``default.lcf`` file per your instructions, this is the console output.

```
2015-11-25 20:35:54,494 json.JsonServlet                      DEBUG - Opening connection [WebServer-54]
2015-11-25 20:35:55,008 json.JsonServlet                      DEBUG - Sending hello [WebServer-54]
2015-11-25 20:36:30,048 json.JsonServlet                      DEBUG - Opening connection [WebServer-56]
2015-11-25 20:36:30,067 json.JsonServlet                      DEBUG - Sending hello [WebServer-56]
2015-11-25 20:36:58,081 json.JsonServlet                      DEBUG - Closing connection because null (1001) [WebServer-52]
2015-11-25 20:36:58,783 json.JsonServlet                      DEBUG - Opening connection [WebServer-51]
2015-11-25 20:36:58,792 json.JsonServlet                      DEBUG - Sending hello [WebServer-51]
2015-11-25 20:37:00,779 json.JsonServlet$JsonWebSocket        WARN  - Unhandled Error (closing connection) [WebServer-52]
org.eclipse.jetty.websocket.api.WebSocketException: RemoteEndpoint unavailable, current state [CLOSED], expecting [OPEN or CONNECTED]
	at org.eclipse.jetty.websocket.common.WebSocketSession.getRemote(WebSocketSession.java:252)
	at jmri.jmris.JmriConnection.sendMessage(JmriConnection.java:103)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:55)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:60)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:280)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:274)
	at jmri.jmris.json.JsonThrottle.getThrottle(JsonThrottle.java:102)
	at jmri.jmris.json.JsonThrottleServer.parseRequest(JsonThrottleServer.java:81)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:299)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:151)
	at jmri.web.servlet.json.JsonServlet$JsonWebSocket.onMessage(JsonServlet.java:749)
	at sun.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:70)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:225)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:64)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:217)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:160)
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:309)
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:214)
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:220)
	at org.eclipse.jetty.websocket.common.Parser.parse(Parser.java:258)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:613)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:468)
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Thread.java:745)
2015-11-25 20:37:00,803 json.JsonServlet                      ERROR - RemoteEndpoint unavailable, current state [CLOSED], expecting [OPEN or CONNECTED] [WebServer-52]
org.eclipse.jetty.websocket.api.WebSocketException: RemoteEndpoint unavailable, current state [CLOSED], expecting [OPEN or CONNECTED]
	at org.eclipse.jetty.websocket.common.WebSocketSession.getRemote(WebSocketSession.java:252)
	at jmri.jmris.JmriConnection.sendMessage(JmriConnection.java:103)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:55)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:60)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:280)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:274)
	at jmri.jmris.json.JsonThrottle.getThrottle(JsonThrottle.java:102)
	at jmri.jmris.json.JsonThrottleServer.parseRequest(JsonThrottleServer.java:81)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:299)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:151)
	at jmri.web.servlet.json.JsonServlet$JsonWebSocket.onMessage(JsonServlet.java:749)
[... cut ...]
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Thread.java:745)
2015-11-25 20:37:00,824 json.JsonServlet                      DEBUG - Closing connection because WebSocketException (1011) [WebServer-52]

```
Would you like me to document the communication between my software and the JMRI server? I can post that here if that would make things easier.

rhwood, 2015-11-26:
I can’t see anything obvious, so some more logging would help. Can you also add log4j.category.jmri.web.server.WebServer=DEBUG as well (it will be verbose).

Between the two versions of Jetty WebSockets handling was rewritten to adhere to the JSR356 standards (I don’t know the impact of that). Jetty 9 also drops support for pre-RFC 6455 versions of WebSockets protocols (again, I don’t know what impact that has).

> The first message you should see from the JSON servlet is the (unfortunately undocumented) “hello” message
>
> Yes, I actually use data from that (the railroad name) in my software, so I know I'm [...]

This log shows 4 WebSockets connections being opened, and one failing to open (each [WebServer-##] is a separate thread containing a separate connection). Is this number of connections expected (it’s generally preferred to open a single WebSocket per client, as each WebSocket connection takes up independent resources on the JMRI server).

> 2015-11-25 20:35:54,494 json.JsonServlet                      DEBUG - Opening connection [WebServer-54]
> 2015-11-25 20:35:55,008 json.JsonServlet                      DEBUG - Sending hello [...]
These following two errors are because the “hello” message can’t be sent.
> 2015-11-25 20:37:00,779 json.JsonServlet$JsonWebSocket        WARN  - Unhandled Error (closing connection) [WebServer-52]
> org.eclipse.jetty.websocket.api.WebSocketException: RemoteEndpoint [...]

Randall Wood
Alexandria Software
202.683.8604
randall.h.wood@alexandriasoftware.com
http://www.alexandriasoftware.com

k4kfh, 2015-11-26:
Here's the log generated by my software. I edited out several sections that were nothing but ping/pong messages, but aside from that this is untouched.

```json
LOG: Connection opened.
LOG: Beginning heartbeats.
SENT : {"type":"power","data":{}}
SENT : {"list":"roster"}
RECIEVED : {"type":"hello","data":{"JMRI":"4.0.1-r29956","json":"3.0","heartbeat":13500,"railroad":"Hampton's Railroad","node":"jmri-00238BEDC26D-3f0fb59a","activeProfile":"My JMRI Railroad"}}
RECIEVED : {"type":"power","data":{"state":2}}
Updated layout track power status to TRUE
RECIEVED : [{"type":"rosterEntry","data":{"name":"AnalogLoco","address":"0","isLongAddress":false,"road":"N/A","number":"N/A","mfg":"N/A","decoderModel":"1 function decoder (36-551)","decoderFamily":"E-Z Command decoders","model":"N/A","comment":"This is the generic address for Analog (DC) locomotives","maxSpeedPct":"100","image":null,"icon":null,"shuntingFunction":"","functionKeys":[{"name":"F0","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F1","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F2","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F3","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F4","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F5","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F6","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F7","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F8","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F9","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F10","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F11","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F12","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F13","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F14","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F15","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F16","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F17","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F18","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F19","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F20","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F21","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F22","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F23","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F24","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F25","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F26","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F27","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F28","label":null,"lockable":true,"icon":null,"selectedIcon":null}],"rosterGroups":[]}},{"type":"rosterEntry","data":{"name":"CBQ1","address":"1","isLongAddress":false,"road":"Chicago, Burlington, and Quincy","number":"1","mfg":"Bachmann Trains","decoderModel":"1 function decoder (36-551)","decoderFamily":"E-Z Command decoders","model":"","comment":"This locomotive does NOT currently have sound, it only has a single function EZCommand decoder","maxSpeedPct":"100","image":null,"icon":null,"shuntingFunction":"","functionKeys":[{"name":"F0","label":"Light","lockable":true,"icon":null,"selectedIcon":null},{"name":"F1","label":"Dim Light","lockable":true,"icon":null,"selectedIcon":null},{"name":"F2","label":"Horn","lockable":true,"icon":null,"selectedIcon":null},{"name":"F3","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F4","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F5","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F6","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F7","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F8","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F9","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F10","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F11","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F12","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F13","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F14","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F15","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F16","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F17","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F18","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F19","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F20","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F21","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F22","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F23","label":null,"lockable":true,"icon":null,"selectedIcon":null},{"name":"F24","label":null,"lockable":true,"icon":null,"selectedIcon":null}, [... cut ...]
RECIEVED : {"type":"pong"}
SENT : {"type":"ping"}
RECIEVED : {"type":"pong"}
SENT : {"type":"throttle","data":{"throttle":"1","address":2}}

* Here is where the connection closes *
```

I know received is spelled wrong, but I didn't notice until after I'd done this.

I'll post the log from JMRI with the new default.lcf settings in a moment. Let me know if you have any questions about anything in that log, or if there's anything else you want me to add to that logging system (timestamps, log something else, etc.)

k4kfh, 2015-11-26:
Here's the log from JMRI (with both your logging configuration edits)

```
2015-11-26 11:11:50,201 server.WebServer                      DEBUG - Setting up handler chain for /robots.txt [Thread-11]
2015-11-26 11:11:50,299 server.WebServer                      DEBUG - Setting up handler chain for /web [Thread-11]
2015-11-26 11:11:50,325 server.WebServer                      DEBUG - Setting up handler chain for /fonts [Thread-11]
2015-11-26 11:11:50,336 server.WebServer                      DEBUG - Setting up handler chain for /css [Thread-11]
2015-11-26 11:11:50,344 server.WebServer                      DEBUG - Setting up handler chain for /js [Thread-11]
2015-11-26 11:11:50,364 server.WebServer                      DEBUG - Setting up handler chain for /favicon.ico [Thread-11]
2015-11-26 11:11:50,371 server.WebServer                      DEBUG - Setting up handler chain for /images [Thread-11]
2015-11-26 11:11:50,395 server.WebServer                      INFO  - Starting Web Server on port 12080 [WebServer]
2015-11-26 11:11:50,495 igurexml.DefaultConditionalManagerXml WARN  - No state variables found for conditional IX:AUTO:0003C1 [AWT-EventQueue-0]
2015-11-26 11:11:50,705 server.WebServer                      INFO  - Starting ZeroConfService _http._tcp.local for Web Server with properties {path=/, json=3.0} [WebServer]
2015-11-26 11:12:00,369 server.WebServer                      DEBUG - Web Server finished starting [WebServer]
2015-11-26 11:12:00,499 withrottle.UserInterface              INFO  - Creating new WiThrottle DeviceServer(socket) on port 25566, waiting for incoming connection... [WiThrottleGUIServer]
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM1: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM1: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM1: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM0: File exists
RXTX fhs_lock() Error: creating lock file: /var/lock/LCK..ttyACM1: File exists
2015-11-26 11:13:10,842 json.JsonServlet                      DEBUG - Opening connection [WebServer-50]
2015-11-26 11:13:11,007 json.JsonServlet                      DEBUG - Sending hello [WebServer-50]
2015-11-26 11:13:39,070 json.JsonServlet                      DEBUG - Closing connection because null (1001) [WebServer-50]
2015-11-26 11:13:39,838 json.JsonServlet                      DEBUG - Opening connection [WebServer-51]
2015-11-26 11:13:39,858 json.JsonServlet                      DEBUG - Sending hello [WebServer-51]
2015-11-26 11:13:42,844 json.JsonServlet$JsonWebSocket        WARN  - Unhandled Error (closing connection) [WebServer-52]
org.eclipse.jetty.websocket.api.WebSocketException: RemoteEndpoint unavailable, current state [CLOSED], expecting [OPEN or CONNECTED]
	at org.eclipse.jetty.websocket.common.WebSocketSession.getRemote(WebSocketSession.java:252)
	at jmri.jmris.JmriConnection.sendMessage(JmriConnection.java:103)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:55)
	at jmri.jmris.json.JsonThrottleServer.sendMessage(JsonThrottleServer.java:60)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:280)
	at jmri.jmris.json.JsonThrottle.sendMessage(JsonThrottle.java:274)
	at jmri.jmris.json.JsonThrottle.getThrottle(JsonThrottle.java:102)
	at jmri.jmris.json.JsonThrottleServer.parseRequest(JsonThrottleServer.java:81)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:299)
	at jmri.jmris.json.JsonClientHandler.onMessage(JsonClientHandler.java:151)
	at jmri.web.servlet.json.JsonServlet$JsonWebSocket.onMessage(JsonServlet.java:749)
	at sun.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:70)
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:225)
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:64)
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:217)
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:160)
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:309)
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:214)
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:220)
	at org.eclipse.jetty.websocket.common.Parser.parse(Parser.java:258)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:613)
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:468)
[... cut ...]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
	at java.lang.Thread.run(Thread.java:745)
2015-11-26 11:13:42,889 json.JsonServlet                      DEBUG - Closing connection because WebSocketException (1011) [WebServer-52]
```

Also @rhwood to answer your question earlier about there being 4 connections, yes, that was intended. I accidentally refreshed the page, so I just restarted my procedure for reproducing the bug. I meant to tell you guys that but I guess I forgot.

rhwood, 2015-11-29:
@k4kfh It would be helpful if I could see some more activity.

Can add the logging controls:
log4j.category.jmri.jmris.json.JsonThrottle=DEBUG
log4j.category.jmri.jmris.json.JsonClientHandler=DEBUG

The JsonThrottle will tell us if anything is happening within the throttle code, and the JsonClientHandler will tell us if it's dropping anything.

Can you also add timestamps to the logs from your software, so I can try to correlate your software's logs with the JMRI servers logs?

It would be better to attach the logs as .txt files than to cut and paste contents from the logs (I think this can only be done when using the web site).

k4kfh, 2015-11-30:
I can add timestamps to my software logging easily. However, in the interest of
me not having to sort through it and remove the non-websockets log entries, I'm
not going to "edit" the log this time. All the WebSockets messages will be
prefixed with either "SENT" or "RECIEVED", so it should be pretty easy to read.
I'll be happy to answer any questions about the rest of the software as well.
I'll also add the additional debug logging settings to JMRI.

I have some schoolwork I need to finish up, so I may or may not be able to do
this tonight, but I'll try to have it posted by Tuesday evening at the latest.

Thanks,
Hampton

[https://github.com/JMRI/JMRI/issues/276#issuecomment-160426073] .[https://github.com/notifications/beacon/AKNEhxPbid2fgLxaU-RceQBrF_paPb9Pks5pKxgHgaJpZM4GpjPo.gif]

k4kfh, 2015-12-02:
Here are both the log files from my test this evening. It behaved just as it has been in the past.
[Client-side log file (from my software)](https://github.com/JMRI/JMRI/files/49343/1dec2015-1-client.txt)

NOTE: JMRI's log will be uploaded in just a moment, I have to upload them from their respective computers so I have to edit this message after initial posting to post the JMRI log from my layout PC. Give me about 5 minutes and it should be up.

The difference with this client-side log file is that I didn't have time to remove all the non-WebSockets-related log entries in my software. They are few and far between, but if you aren't sure what it's from I'm happy to answer any questions. The WebSockets log entries all begin with "[timestamp] SENT" or "[timestamp] RECIEVED". The timestamps are in CST, if that matters.

For the JMRI log, I have included the entire log (startup and all) this time, just in case there's some clue in there.

rhwood, 2015-12-10:
I don't see anything obvious, and can't reproduce this behavior using WebThrottle on an iPhone, so I am inclined to think it's not in JMRI.

k4kfh, 2015-12-12:
@rhwood Is there any way I can test this with Jetty 9.3? I'm curious if that fixed it or not, so if there's somewhere I can download a build of JMRI with Jetty 9.3 in it, that would be absolutely fantastic.

bobjacobsen, 2015-12-12:
You can download a development installer from Jenkins:

http://builds.jmri.org/jenkins/job/Development/job/Packages/

k4kfh, 2015-12-21:
My apologies for not replying for a while, it was exam week and I didn't get a minute's rest until this weekend. But I have good news! The development build I downloaded would not reproduce the bug, no matter how hard I tried. I repeated the test 3 times and had no problems at all; everything worked as it should. Thank you so much for working with me to fix this!

Also, this isn't related to this bug, but when I ran ``./PanelPro`` on the dev build, it told me that my profile had been updated to a new storage format. I don't know exactly what sort of changes this involved, but it would be nice for it to automatically back up the old profile. (unless it did and I just didn't notice?) I happened to have mine backed up already, so if I need to revert to an older version I can restore it, but in the final version you might consider having it dump the old one in a subfolder of the new one somewhere. Of course, if the new one is backwards-compatible, none of that matters.

rhwood, 2015-12-21:
Good to know.

The old profile storage format (in ProfileConfig.xml; the new format is multiple files under .profile) is currently maintained even after the migration, although this will stop once the first feature that relies on the new format is introduced.

closed 2015-12-21 by rhwood