JMRI/JMRI #283: Roster location set to jython directory for some old profiles

bobjacobsen, 2015-11-28:
(This has been discussed in jmri-developers, but I wanted to make sure the files were available here)

With the current HEAD of JMRI/JMRI master, certain old profiles convert to new ones that have the roster directory reference pointing to the jython (!) subdirectory of the program directory.

Here's a [.zip version of the profile before conversion](https://www.dropbox.com/s/vovw4vabtohmb4d/Klingon_check_before.zip?dl=0) and a [.zip version of the results of the conversion](https://www.dropbox.com/s/8cwxc1ctcxgh105/Klingon_check_after.zip?dl=0).  The easiest way to check is to set the "before" in place, run the program, let it convert (it gives a message about that), and check the locations via the Help menu.

rhwood, 2015-11-28:
There is some code, and I haven't fully found it, that strip the trailing separator (```/``` or ```\``` depending on OS) off the roster directory location so that when checking that the roster is within the user files location (which *always* end with a trailing separator), that test fails.

Note that tests that something is in ```scripts:``` are always after tests that something is in ```preferences:```, so if the preferences test fails, the item would be dropped into ```scripts:```.

There also has been some strange bug that would cause the ```scripts:``` location to be set to the root of the filesystem and then cause any other path that does not match ```preferences:``` or ```settings:``` to be set to ```scripts:/ab/so/lute/path```

I think I have killed the problem with the ```scripts:``` location getting set incorrectly in #196.

rhwood, 2015-11-28:
In this case, the problem was that the Roster would listen for *any* change in a settable directory location, instead of listening for just the User Files location.

PR #285 (2015-11-28): Listen only for changes in the User Files location
The Roster was listening for any settable portable JMRI path to change, not just the User Files path to change.

Fixes #283.
commit 67c91345 (2015-11-28): Merge pull request #285 from JMRI/rhwood-patch-4

Listen only for changes in the User Files location.

Fixes #283.
commit 6df23055 (2015-11-28): Listen only for changes in the User Files location

The Roster was listening for any settable portable JMRI path to change, not just the User Files path to change.

Fixes #283.
closed 2015-11-28 by rhwood