JMRI/JMRI #291: Some tests can no longer run standalone due to jmri.spi.PreferencesProvider exception

bobjacobsen, 2015-11-30:
Some tests that used to run standalone (e.g. by using runtest.csh to invoke the main() in the test class) no longer do.  A typical traceback is below.  Is there some initialization step that's needed to prevent this error?  

The test does pass when run via e.g. alltest, which makes me think that some global initialization is happening earlier.  But I haven't been able to track it down.

Thanks

````
junit.framework.AssertionFailedError: Exception in constructor: testShow (java.util.ServiceConfigurationError: jmri.spi.PreferencesProvider: Provider apps.StartupActionsManager not a subtype
	at java.util.ServiceLoader.fail(ServiceLoader.java:239)
...
	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
	at jmri.implementation.JmriConfigurationManager.<init>(JmriConfigurationManager.java:40)
	at jmri.configurexml.LoadStoreBaseAction.<init>(LoadStoreBaseAction.java:33)
...
	at jmri.jmrit.display.panelEditor.PanelEditor.<init>(PanelEditor.java:120)
	at jmri.jmrit.display.LinkingLabelTest.<init>(LinkingLabelTest.java:20)
````

rhwood, 2015-11-30:
What tests are failing?

I am unable to get the LinkingLabelTest to fail running:
```
ant clean
ant tests
runtest.csh jmri.jmrit.display.LinkingLabelTest
```

DanielBoudreau, 2015-11-30:
Bob.

 

I had similar problems running some of the operation test that referenced files.  I added the “noloading” option and it seemed to fix the problems for me.

 

Dan

bobjacobsen, 2015-11-30:
````
./runtest.csh jmri.jmrit.display.PackageTest
````
seemed to fail this way pretty consistently.  I added the “noloading” option, and now it seems to work.  (I thought I'd gotten all of those a while back, but apparently not).  Thanks!

I'll make a pass through for missing noloading options and fix them.  

Does anybody understand how that drives this fault?  Are we doing more class loader stuff that's going to be a problem down the line?

closed 2015-11-30 by bobjacobsen