JMRI/JMRI #305: Create release-4.1.5

mattharris, 2015-12-03:
Starting process

mattharris, 2015-12-03:
On my Linux box, I'm seeing 6 errors via alltest right now.

They are:

- testShowBlank(jmri.jmrit.display.MemoryIconTest): Expect gray pixels
- testBackgroundColors(jmri.jmrit.display.PositionableLabelTest): F Bkg none, label Bkg none expected:<0xff[ffffff]> but was:<0xff[eeeeee]>
- testLayoutEditor(jmri.jmrit.display.SensorIconWindowTest): state after one click expected:<4> but was:<1>
- testLayoutEditor(jmri.jmrit.display.TurnoutIconWindowTest): state after one click expected:<2> but was:<1>
- testOptionFrameWrite(jmri.jmrit.operations.setup.OperationsSetupGuiTest): build normal
- testStagingEditFrame(jmri.jmrit.operations.locations.StagingEditFrameTest): new staging track

Now, the first two I was, sort of, expecting, given that PR #301 is linked to that.

However, the others look a bit more suspicious.

mattharris, 2015-12-03:
Running same on Windows gives just 4 errors:

- testShowBlank(jmri.jmrit.display.MemoryIconTest): Expect gray pixels
- testBackgroundColors(jmri.jmrit.display.PositionableLabelTest): F Bkg none, label Bkg none expected:<0xff[ffffff]> but was:<0xff[eeeeee]>
- testDemoWithBuildLoad(jmri.jmrit.operations.XmlLoadTests): Number of Locations expected:<12> but was:<0>
- testDemoLoad(jmri.jmrit.operations.XmlLoadTests): Number of Locations expected:<12> but was:<0>

So, first two as expected (given #301) but the fact that the four others do not occur, plus that two different ones now arise, is odd.

Now these were run on the same physical machine (it's natively Windows 7, with Ubuntu Mate 15.10 running in a full-screen VM).

So, screen resolution is the same in both cases (1600x900).

In both cases, ant alltest is invoked from command line

DanielBoudreau, 2015-12-03:
I’m only seeing the first two test failures on my IDE, Windows Eclipse.

 

Dan

bobjacobsen, 2015-12-03:
I'm trying to merge #301 now. Pete says that will fix the MemoryIconTest and PositionableLabel test errors. But I might run out of time this afternoon.

mattharris, 2015-12-03:
OK.

I've been messing with UI settings on the Linux VM (hiding the various toolbars and reducing standard font sizes) and this seems to have resolved (usually) the erroneous UI tests.

Occasionally, one or two of these fail but, on a subsequent run, they seem OK.

Still cannot understand why jmri.jmrit.operations.XmlLoadTests fails on my Windows machine yet passes fine on the Linux VM.

bobjacobsen, 2015-12-03:
on Mac OS X, I get no failures on headlesstest, two on alltest:

	at jmri.jmrit.display.MemoryIconTest.testShowBlank(MemoryIconTest.java:96) (expect white pixels)

	at jmri.jmrit.display.PositionableLabelTest.testBackgroundColors(PositionableLabelTest.java:89) (F Bkg none, label Bkg none color expected:<0xff[eeeeee]> but was:<0xff[ffffff]>)

The first is almost certainly Mac specific, and the second appears to be. I think we should press on.  I've merged #301 to master.  Could you please merge it into this branch?  I can create a specific PR to the release branch if you'd prefer.

mattharris, 2015-12-03:
I don't see changes from #301 in HEAD

mattharris, 2015-12-03:
Ignore that, it's there now.

mattharris, 2015-12-03:
The release-4.1.5 branch has been created. 

From now on, please document your changes in the [jmri4.1.6.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.1.6.shtml) release note file.

Maintainers, please set the 4.1.6 milestone on pulls from now on.

mattharris, 2015-12-04:
First files for JMRI 4.1.5 are available in the usual place:

http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.1.5/

bobjacobsen, 2015-12-04:
Looks pretty good on Mac OS X.  There’s still a tiny artifact in empty-String Memory variables, but I don’t think that’s a reason to rebuild. Can include in 4.1.6 easily.

In addition to mentioning the files in this GitHub Item, should we post something in jmri-developers?  Not sure how many people are following GitHub JMRI/JMRI yet.

Bob
--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

mattharris, 2015-12-04:
Thought that I'd also sent the message to jmri-developers

bobjacobsen, 2015-12-04:
Oops. Sorry. You did send it to jmri-developers, but for some reason I didn't receive it until late that evening (after Dan's reply). Several other posts delayed too. Not sure why.

bobjacobsen, 2015-12-04:
I'm trying to think through the best way to handle two types of requests with our GitHub tools: "If you rebuild the files, here's a change I'd like to get in for the current release" (in this case 4.1.5) and the more specific "In the next patches only-release, please include this" (in this case, 4.1.6)

I wrote a [section on this in the release HOWTO](https://github.com/JMRI/JMRI/blob/master/scripts/HOWTO-distribution.md#further-changes) but I don't think it really captures it.  That suggests making two pull-requests:  One to master and another to the release branch.

The problem is that, if you've created that branch from master _after_ the release branch was created, it'll have all the intervening changes in it.  When I created the fix in #310 from master, it picked up some changes (e.g. #308) that had already been merged.  We can deal with that via e.g. cherry-picking, but it seems a pain.

On the other hand, we could recommend that people start "fixes after deadline" via branching from the release branch.  I think it would be easy to eventually merge that onto master (now that we don't back-change the release number in release.properties).  But that requires a degree of sophistication in development that I'm not sure we have: Somebody might have been working on that fix for a while, for example, and is keeping it merged close to the HEAD of master.

One more problem: Timing. Right now, I've got a local branch  (bobjacobsen/JMRI bobj-4.1.6-add-icon-test-fix) that's meant to be in 4.1.5-rebuild or 4.1.6, whichever makes sense to Matt as he builds (he's doing it, he decides).  It's already on JMRI/JMRI master. If I PR it to the 4.1.5 branch, it's easy for him to accept it or not, and it'll probably stay consistent because there won't be a lot of changes.  it's also cheap for him to close it. Maybe I should go ahead and and do that PR to the 4.1.5 branch.

But I can't do it on the 4.1.6 branch, because that doesn't exist yet. The PR to master does exist (it's even closed) so I could make that with 4.1.6 milestone. Would that be enough?  Not really, as there's no way to retarget that PR from the 4.1.5 release branch to the 4.1.6 release branch when it exists.

Maybe we need a slightly different structure for the "series of test releases with limited content that's aimed at a production release".  E.g. create a "4.1-leading-to-4.2" branch that starts with the 4.1.5 build, but still gets PRs that can be accepted or not to create 4.1.6, 4.1.7 etc as needed.

As you can see, I don't have a clear picture yet, and am not wedded to any of this. I don't want to build something that's impractical, or too big a burden on either the (many) individual developers, or the (few) release builders. I certainly don't want to scare people away.  But what we have now is already a bit complex and confusing...

mattharris, 2015-12-04:
What I've done with #310 is to git cherry-pick the individual commits from the PR into the branch and then directly commit to JMRI/JMRI rather than over my own repo.

Not ideal, but manageable...

Let's see what Jenkins makes of it before doing similar for others...

mattharris, 2015-12-04:
Given that 4.3.1 will be the next release from master, I'm going to update 'release.properties' accordingly.

mattharris, 2015-12-04:
OK, Jenkins seems to have built that OK.

Could some people please check the latest build:

http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.1.5/

DanielBoudreau, 2015-12-04:
Downloaded and installed okay on Win 7. Loaded a panel, tested routes and turnouts using NCE simulator, all good.

 

Dan

 

http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.1.5/

mattharris, 2015-12-04:
That's great - thanks Dan.

Just got Mark's updates for DCC++ to merge in and then will do one final rebuild.

Then it will be onto the public release itself.

bobjacobsen, 2015-12-04:
JMRI.4.1.5-Rbab5a95.dmg from Jenkins looks good!

mattharris, 2015-12-04:
Thanks Bob.

The final one is the next from Jenkins job 3 JMRI.4.1.5-R25e54ec which incorporates Mark's DCC++ updates.

Now doing the next stages.

mattharris, 2015-12-04:
Closing this one now as released on 04 Dec.

closed 2015-12-04 by mattharris

rhwood, 2015-12-06:
If I have a PR against the 4.1.5 branch or against master that is targeted for 4.1.6, should Matt be the only person who merges the PR?

mattharris, 2015-12-06:
Bringing commits into the release branches should only be done by the pumpkin (currently me).

Best way to communicate this is to add a comment in the PR for it to be included in next release (and also add it to the 4.1.6 milestone.)

mattharris, 2015-12-06:
In fact, better still, add the comment to the issue just created, #322, for the build of the next dev release