JMRI/JMRI #316: Develop release-4.2

mattharris, 2015-12-05:
First pass at release note for 4.2 has been submitted.

Please review this (it's just a concatenation of pretty much everything from the 4.1.x series).

Thanks.

mattharris, 2015-12-20:
Production Release 4.2 will be built based upon 4.1.6.

mattharris, 2015-12-20:
The release-4.2 branch now exists.

mattharris, 2015-12-20:
First candidate files for JMRI 4.2 are available in the usual place:

http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.2/

mattharris, 2015-12-20:
Pausing release at the moment due to issue noticed by Dan (copied from developers list):

>Using Panel Editor, I found that if I select a blinking icon for a sensor icon, that it doesn't blink.  I checked the other test versions 4.1.1 through 4.1.6 and they all have the same problem. Version 4.0.1 works correctly.  If I just add a blinking icon to the panel, it does blink, and it seems to work ok if I assign blinking icons to turnouts, so the problem seems to be with sensors.
> Testing using NCE simulator.
>
> Here's an example of one of the icons that I'm having problems with:
>
>program:resources/icons/USS/sensor/blink/red-b.gif
>
>Dan

mattharris, 2015-12-20:
Do we feel that this is serious enough to stop the release?

Only, we really should try and avoid pushing out a 4.2.1. Yes, we have
done this in the past, but it can get a bit messy.

Thoughts?

(sent also to jmri-developers)

KenC57, 2015-12-21:
I know we will hear of this from the users. A number of them (ok at least 5
or 10) use blinking icons for either signals or control panels and expect
them to flash. Non-flashing means something else. I recall helping (at
least) twice on how to substitute the icons to make that happen.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

DanielBoudreau, 2015-12-21:
I went back and retested, and I was wrong to report that versions 4.1.1 through 4.1.4 had the blinking sensor icon problem.  The first release that shows the problem is version 4.1.5.  So I think this problem was introduced in the last 4 weeks.  History shows that Pete C. was making changes to the sensor code to fix an issue reported by Walt.

Dan

mattharris, 2015-12-24:
Pete's updates have been brought in via #401 for the release-4.2 branch

mattharris, 2015-12-24:
Closing as part of release process

closed 2015-12-24 by mattharris