JMRI/JMRI #322: Develop release-4.1.6

mattharris, 2015-12-06:
Start this issue for any communications related to 4.1.6.

Note: 4.1.6 is based upon the 4.1.5 branch and NOT master.

Only explicit commits will be brought in by request

mattharris, 2015-12-08:
OK, the pending requests for release-4.1.6 have been merged.

First Jenkins build from this branch has been triggered.

mattharris, 2015-12-08:
Should #323 be included?

bobjacobsen, 2015-12-08:
If you get the chance, it would be nice to get #336 included. It's one decoder file committed to the HEAD of master by 8feb7ef58add2a7b15670d498d9678aa6885ab39 - no update to the decoder index needed.

mattharris, 2015-12-09:
8feb7ef is now in.

Next Jenkins build (#3) should pick this up.

bobjacobsen, 2015-12-09:
#341 is additional decoder files. If there's a rebuild, it would be good to get in; OK for 4.3.1 if not.

Note the note in #341 about the contents, please.

mattharris, 2015-12-10:
#341 is now included - see note about the inclusion strategy used

mattharris, 2015-12-11:
First candidate files for JMRI 4.1.6 are available in the usual place:

http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.1.6/

(Build 5 or later)

mattharris, 2015-12-12:
Brought in part of #349 as per Pete C email message on jmri-developers

Build 6 or later on Jenkins will incorporate this change.

mattharris, 2015-12-12:
Closing as part of release process

closed 2015-12-12 by mattharris