JMRI/JMRI #338: Test of access to GitHub Issues

bobjacobsen, 2015-12-09:
This is a test of access to the GitHub Issue system. It is only a test.

Whhhhheeeeeeeeeeeeeee

We now return you to your normally scheduled issues.

Seriously, the idea of this is to start an issue, and see whether people with regular GitHub accounts can:
- [ ] Add or remove labels (I put "question" on so somebody could try to remove it)
- [ ] Change the milestone
- [ ] Close the item

Could somebody _without_ Git write access please try these?  Thanks.

dpharris, 2015-12-09:
I can leave a comment - DPHarris

mmosher5501, 2015-12-09:
The only things I can do is view it and add this comment.

PUlvestad, 2015-12-09:
Just signed up, can only add this comment.
Peter

rhwood, 2015-12-10:
Non-contributors:
 - [ ] can *not* assign labels
 - [ ] can *not* assign milestones
 - [ ] can only close *their own* issues

I tested this by filing octocat/Spoon-Knife#8242 to verify that behavior on a repo I cannot contribute to.

closed 2015-12-10 by bobjacobsen