JMRI/JMRI #342: AppVeyor broken, not our fault

bobjacobsen, 2015-12-09:
The AppVeyor CI tests are failing, and it's not our fault. 

It loads and I talks ant and the Java SDK on each run. Sometime yesterday, that process updated from Java 1.8 65 to 66, which is not currently working. 

Don't worry about AppVeyor CI test fails until further notice.

mattharris, 2015-12-09:
I think that you meant to say 'It loads and installs ant and the Java SDK on each run.'

:smiley: 

Bit of a pain that JDK 1.8_66 is causing a problem though.

bobjacobsen, 2015-12-09:
Thanks. Fixed. Stupid autocorrect....

mattharris, 2015-12-09:
Taking a quick look at the log seems to suggest that JAVA_HOME is being set incorrectly.

According to the log, the JDK is being installed in:

C:\Program Files\Java\jdk1.8.0_66

but we seem to be explicitly setting it in the yml file:

SET JAVA_HOME=C:\Program Files\Java\jdk1.8.0

mattharris, 2015-12-09:
Damn. That doesn't seem to have worked.

But, looking at the [installed software page](http://www.appveyor.com/docs/installed-software#java), suitable JDK's should already be installed by default - it's just 'ant' that's missing.

closed 2015-12-10 by rhwood

bobjacobsen, 2015-12-10:
It had been setting to C:\Program Files\Java\jdk1.8.0 when the problem happened. I added the _66 because that's where the new one was being installed, but there was some kind of Windows-fu that was keeping that form working.  Randall's fix of not downloading a SDK is a better approach.

commit f1881ab2 (2015-12-10): Merge pull request #348 from rhwood/master

Do not reinstall the JDK in AppVeyor.

AppVeyor tests run, but failed tests do not upload test results.
Fixes #342