JMRI/JMRI #359: Make MS Windows installer a bit more tolerant to changes in provided files

mattharris, 2015-12-11:
Currently, the installer for MS Windows performs a merge operation with any existing JMRI installation.

Whilst this gives slight advantages with respect to speed of installation, it does mean that we need to remember to include explicit deletions of any files that are no longer necessary.

An example of this would be with the effect of PR #358 that, unless we change behaviour, will require that we add explicit deletions of the superseded library files.

There are a number of approaches that could be taken:

1. Deletion of entire existing JMRI installation
2. Deletion of sub-sets of JMRI installation i.e. remove entire `lib` tree
3. Do nothing and continue with manually adding explicit deletions to the [`InstallJMRI.nsi`](https://github.com/JMRI/JMRI/blob/master/scripts/WinInstallFiles/InstallJMRI.nsi) file when encounted

Thoughts?

rhwood, 2015-12-11:
I think we should remove the entire previous installation. This would make the Windows installer behavior consistent with the other platforms.

That said, I am aware that we previously accepted, if not required, users to add personal files to the JMRI installation, so if we are upgrading an existing installation of JMRI version 4.2 or older, we may want to copy that installation to another directory, so a user can recover anything he placed there.

mattharris, 2015-12-11:
As you mention:

> we previously accepted, if not required, users to add personal files to the JMRI installation, so if we are upgrading an existing installation of JMRI version 4.2 or older, we may want to copy that installation to another directory, so a user can recover anything he placed there.

We already are aware of which previous version that we are going from if we upgrade (it's one of the _very_ few things we write into the Windows Registry - it's actually part of the pretty much mandated entry for JMRI to appear in the `Add/Remove Programs` list.

So, we can conditionally perform a backup and state that, from version 4.x onwards, an existing installation is completely removed.

TL&DR:

I agree with you.

bobjacobsen, 2015-12-11:
I also agree.

Backing up the whole thing seems like the best approach.  It's a little bit of disk space, but not that much, and people can delete the backup.

dheap, 2015-12-12:
But we would need to zip it up, otherwise users will manage to run the executables or scripts in the backup directory by mistake, or base a modified decoder definition or script on an obsolete copy.

Sent from my iPad
-- 
Dave in Australia

> On 12 Dec 2015, at 1:28 AM, Bob Jacobsen <notifications@github.com> wrote:
> 
> I also agree.
> 
> Backing up the whole thing seems like the best approach. It's a little bit of disk space, but not that much, and people can delete the backup.

bobjacobsen, 2015-12-12:
I don't understand the expectations of Windows users: what they tend to do or not do.  Are .zip files considered friendly or fiendish?

AFAIK, [the installer](http://jmri.sourceforge.net/install/WindowsNew.shtml) is already backing up the roster and settings to c:\Documents and Settinsg\(user)\JMRI_backup without .zipping them, though of course that's all user-created content.

To prevent running from there, we could omit the jmri.jar file, and perhaps the startup scripts (is there a reason for users to make valuable changes to those)?  But I think it's OK to leave the decoder definitions there, even unzipped.

dheap, 2015-12-14:
> I don't understand the expectations of Windows users: what they tend to do or not do. Are .zip files considered friendly or fiendish?
> 
Modern Windows versions show .zip files somewhat like folders, but still need unpacking.
> AFAIK, the installer is already backing up the roster and settings to c:\Documents and Settinsg(user)\JMRI_backup without .zipping them, though of course that's all user-created content.
> 
Quite different from backing up the program: location.
> To prevent running from there, we could omit the jmri.jar file, and perhaps the startup scripts (is there a reason for users to make valuable changes to those)? But I think it's OK to leave the decoder definitions there, even unzipped.
> 
If we do any, I think it should be only XML, resources and scripts.

rhwood, 2015-12-14:
Windows XP does not (or did not early in its release cycle) include a native ability to zip a directory. The NSIS installer definitely has no native ability to handle zipping folders.

Those issues aside, since moving all existing content to another directory will break any existing shortcuts which point to something not included in the newly installed items replacing them, I don't think we need to worry about people accidentally launching the wrong thing.

We do need to worry about inadvertently deleting anything the user added or modified anywhere in the existing directory, so the entire directory must be backed up. We know, from the JMRI Users list, that JMRI users replace libraries we ship, add libraries to JMRI, and can be generally comfortable with these actions. We only need to protect their changes (by shunting them aside), sufficient to allow them to copy them out of the directory.

Note that I am proposing this copy as a one time action from < 4.3.1 to 4.3.1+; subsequent upgrades from 4.3.1+ to 4.3.1++ should wipe out everything, including the archived < 4.3.1 content.

dheap, 2015-12-15:
There is a current thread over on the users list complaining about the current size of the JMRI package and suggesting cut-down versions. We may not win too many friends backing up ~100MB to user space.

I haven't played with NSIS installer for a while, but could we force it to run an uninstall prior to the backup, thereby leaving only user-added (and possibly also user-changed - I'm not sure whether we can govern this behaviour) content to back up? Then delete all and run the install pass.

rhwood, 2015-12-15:
The complaints are about the download size, and not about the installed size. It's really difficult to find a computer that JMRI and Windows can function on that does not have sufficient size for that, if only so the user can determine he does not need it, and can delete it on his own.

dheap, 2015-12-15:
> The complaints are about the download size, and not about the installed size. It's really difficult to find a computer that JMRI and Windows can function on that does not have sufficient size for that

Agreed, but user perception is not always in line with ours.

bobjacobsen, 2015-12-15:
Not a big fan of options, I think they usually just add complexity, but in this case would it make sense for the installer to ask the user whether to make a backup?

The whole backup mechanism, including prompts etc, can get tricky. This is a backup and remove: what if somebody runs the installer twice?  Has more than one install? Etc.

dheap, 2015-12-15:
> Not a big fan of options, I think they usually just add complexity, but in this case would it make sense for the installer to ask the user whether to make a backup?
> 
We already offer an option to backup User Files on Windows.

rhwood, 2018-01-23:
Can this be addressed before we release 4.12?

devel-bobm, 2018-01-24:
One problem I see (and have seen!) is that users on Windows, especially long-time JMRI users, are prone to installing XML and scripts into the program directory, instead of the JMRI user directory.  

To me, it would appear that anything short of a "back-up-the-whole-program-directory" is going to leave something (and someone) "stranded" account "lost custom content".

We need to make sure that every customization a user could want to do could be done from a "user directory"; once this is firmly in-place, we could guiltlessly enforce a "nuke the program directory" process, instead of a "back-it-up, just in case" process.

If I recall correctly, it can be difficult to get custom "sound clips" (.wav files) to be playable from the JMRI user directory, but it has been more than a year since I've tried to do so; code may have changed since then.

bobjacobsen, 2018-01-25:
> One problem I see (and have seen!) is that users on Windows, especially long-time JMRI users, are prone to installing XML and scripts into the program directory, instead of the JMRI user directory. [...]

The installer that JMRI uses on Windows (but not Linux and macOS) is pretty powerful; it can remove and replace individual files, use some (limited) logic about that, etc.  So it could be set to remove specific old libraries and jar files. But it’s a lot of work to get all the details of that right.  

C.f. lines 478-500 of:

https://github.com/JMRI/JMRI/blob/master/scripts/WinInstallFiles/InstallJMRI.nsi

You’re right that Windows people are (particularly) likely to put things in the program directory, which makes it difficult to remove that entire directory.  But perhaps we could remove the entire contents of the lib directory? Parts of the XML hierarchy, like xml/schema?

Bob

rhwood, 2018-01-25:
I think that being selective about replacement is no better (and maybe even worse from a user’s perspective) than what we do today. As initially hostile as it may appear, only installing in a clean folder (and being clear about that) is the best way forward for both developers (we aren’t having to touch the installer or maintain exceptions in it) and users (nothing from an old installation can pollute a new installation).

DanielBoudreau, 2018-02-24:
We got bit again.  Randall recently updated jackson-core.2.8.5.jar to jackson-core.2.8.11.jar and cleaned up JsonManifest.java to use the new methods. The windows installer wasn't updated to delete the old version.

I've been getting a bunch of emails from operation users complaining about exceptions, here's what they are seeing:

java.lang.NoSuchMethodError: com.fasterxml.jackson.databind.node.ObjectNode.set(Ljava/lang/String;Lcom/fasterxml/jackson/databind/JsonNode;)Lcom/fasterxml 
at jmri.jmrit.operations.trains.JsonManifest.getLocations(JsonManifest.java:101) 
at jmri.jmrit.operations.trains.JsonManifest.build(JsonManifest.java:66) 
at jmri.jmrit.operations.trains.TrainBuilder.build(TrainBuilder.java:329) 
at jmri.jmrit.operations.trains.TrainBuilder.build(Trainbuilder.java:110) 
at jmri.jmrit.operations.trains.Train.build(Train.java:2914) 
at jmri.jmrit.operations.trains.TrainsTableModel$3.run(TrainsTableModel.java:402) 
at java.lang.Thread.run(Unknown Source) 

I don't know how to update the installer, so I need some help.

Dan

dsand47, 2018-02-24:
If the Windows installer was changed to **Replace** instead of **Merge**, a lot of problems would be eliminated.  Or have the installer present a choice with **Replace** as the default.

Dave Sand

bobjacobsen, 2018-02-24:
I think it's up to the Windows community to figure out what the JMRI Windows installer should do about backups, old content, terminology, etc. I don't have a good feel for what Windows users expect. So I'll stay out of that.

But I think the current issue is just a bug.  The "jackson" lines in the scripts/InstallJMRI.nsi file that defines the installer say:
```

    ; -- Delete old jackson jar files as of JMRI 4.7.1
    Delete "$OUTDIR\jackson-annotations-2.0.6.jar"
    Delete "$OUTDIR\jackson-core-2.0.6.jar"
    Delete "$OUTDIR\jackson-databind-2.0.6.jar"

    ; -- Delete insecure jackson jar files as of JMRI 4.11.3
    Delete "$OUTDIR\jackson-annotations-2.8.5.jar"
    Delete "$OUTDIR\jackson-core-2.8.5.jar"
    Delete "$OUTDIR\jackson-databind-2.8.5.jar"
```

but lots of other clean up sections look like (note the "lib" reference):
```
    ; -- Delete older outmoded jetty .jar and outmoded servlet files, as of
    ;    JMRI 3.11.3 (added for version 0.1.21.5 of Windows installer)
    Delete "$OUTDIR\lib\jetty-continuation-8.1.0.v20120127.jar"
    Delete "$OUTDIR\lib\jetty-http-8.1.0.v20120127.jar"
    Delete "$OUTDIR\lib\jetty-io-8.1.0.v20120127.jar"
```

I don't know enough about NSIS to be sure, but I think $OUTDIR is the installation directory.  In that case, the jackson files should be removed from$OUTDIR\lib\ instead.

Could somebody please confirm that the problem currently seen is due to the continued, undeleted presence of one or more of the jackson 2.8.5 files?

bobjacobsen, 2018-02-24:
#4947 is a PR to remove the jackson files from \lib\ too.  CI doesn't do anything with NSIS, but perhaps some Windows-literate person can pull the bobjacobsen/bobj-windows-installer-drops-lib branch, build an installer and check this. Otherwise, we can merge and wait for Jenkins to make Packages, and then have some Windows user check.

DanielBoudreau, 2018-02-24:
We've also confirmed that deleting the old jackson files fixes the exception.  Thanks for fixing the installer delete commands.

closed 2021-04-14 by stale[bot]