JMRI/JMRI #369: Travis tests crashing VM

bobjacobsen, 2015-12-13:
The last couple of Travis tests have failed. There's a typical log section below.

This is happening even when rerunning a previously-passed test, so I don't think it's in our code. 

Each of the failures have been in a "Test schema valid" step. Because it's only showing 1 test run, it appears that the failure is during the pass through the tests to decide which will be run and which will not, because the tests themselves.  But it's a little mysterious why it's failing in different places. Could this be a new timeout in Travis?

````
ci-test:
    [junit] Running apps.tests.AllTest
    [junit] Testsuite: apps.tests.AllTest
    [junit] Testsuite: apps.tests.AllTest
    [junit] Tests run: 1, Failures: 0, Errors: 1, Time elapsed: 0 sec
    [junit] 
    [junit] Testcase: apps.tests.AllTest:Test schema valid: xml/signals/BNSF-1996/appearance-SE-2S.xml:	Caused an ERROR
    [junit] Forked Java VM exited abnormally. Please note the time in the report does not reflect the time until the VM exit.
    [junit] junit.framework.AssertionFailedError: Forked Java VM exited abnormally. Please note the time in the report does not reflect the time until the VM exit.
    [junit] 
    [junit] 
    [junit] Running apps.tests.AllTest
    [junit] Tests run: 1, Failures: 0, Errors: 1, Time elapsed: 0 sec
    [junit] Test apps.tests.AllTest FAILED (crashed)
BUILD FAILED
/home/travis/build/JMRI/JMRI/build.xml:1020: condition satisfied
Total time: 3 minutes 19 seconds
/home/travis/build.sh: line 112:  1956 Terminated              travis_jigger $! $timeout $cmd
````

rhwood, 2015-12-14:
It's interesting that this is only happening in JMRI/master and PRs against it. I'm not seeing this in the CI for [my fork](https://travis-ci.org/rhwood/JMRI).

mattharris, 2015-12-15:
Could it have been related to [this issue](https://www.traviscistatus.com/incidents/z5xjzpb8fwzg) reported by Travis?

Especially seen as the [latest build](https://travis-ci.org/JMRI/JMRI/builds/96908877) has passed.

mattharris, 2015-12-17:
It seems to be rather inconsistent.

Take the latest PR that I've just merged #374.

On the initial Travis build, it failed somewhere in the schema validations with the Forked VM exit.

However, on subsequently restarting that build via the Travis UI, it passed. No code changes.

bobjacobsen, 2015-12-20:
Seems fixed now, at least hasn't happened recently.

closed 2015-12-20 by bobjacobsen