JMRI/JMRI #383: Develop release 4.3.1

mattharris, 2015-12-20:


mattharris, 2015-12-20:
It is expected to release 4.3.1 built from the content of the HEAD of master branch around the same time as Production release 4.2.

bobjacobsen, 2015-12-30:
Planning to build 4.3.1 from HEAD of master late (California time) on December 30

DanielBoudreau, 2015-12-30:
Good for me.

 

Dan

DanielBoudreau, 2015-12-30:
Bob,

 

We still have a problem with blinking sensor icons.  Pete hasn’t been able to commit his changes to the head.

 

I got knocked off sourceforge email last night, so I might be out of the loop.

 

Dan

bobjacobsen, 2015-12-31:
Starting build.

We probably do have a problem in the jmrit.display *Icon classes, but I don't think the solution is further mucking around in muddy code.  I've started a top-down re-implementation, with the base PositionableLabel in master for this release.  Hasn't fixed all the problems (yet), but it has a better chance of converging in the long run.

bobjacobsen, 2015-12-31:
The release-4.3.1 branch has been created. 

From now on, please document your changes in the [jmri4.3.2.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.3.2.shtml) release note file.

Maintainers, please set the 4.3.2 milestone on pulls from now on.

Jenkins will be creating files shortly at [the usual place](http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.3.1/)

closed 2015-12-31 by bobjacobsen