JMRI/JMRI #387: jmrit.display icon graphics test cases

bobjacobsen, 2015-12-21:
The run up to JMRI 4.2 has found another problem with panel icon graphics.  (See #316 for more info)

This particular problem is that SensorIcons don't display animated gifs properly.  The attached [SensorIconTest.xml.txt](https://github.com/JMRI/JMRI/files/68011/SensorIconTest.xml.txt) file demonstrates the problem.  The lower icon flashes in JMRI 4.1.3 and doesn't in the candidate 4.2 build.

There's work in #316 about what to do for the immediate issue, which is deciding what code to include in JMRI 4.2. Please discuss that over there, not here.

This issue is to create a set of test cases that we can define and implement, and then start to refactor the jmrit.display code into a maintainable state.  Please add things that I've not thought of.  I'm going to do some test coding (for the 4.3 series) over the holiday break.

- Need to individually test all the *Icon classes (_without_ a lot of cut&paste code!): Sensor, Turnout, Light, Memory, etc.

- Text and icon basic functions work
  - Text
    - Font size
    - Font color
    - Font style, including bold, etc
    - What shows for empty string
  - Icons
    - What shows when no icon present
- Icons are in front of panel background, but alpha-transparency allows panel background to show through
  - You can set the background on the icon separately from the panel background, and this works with transparency
- Icons can rotate; they don't have to be square to rotate
  - Transparency still works for rotated icons
  - Animation still works for rotated icons
  - (Can text be rotated? When?)
- Icons can scale
  - Transparency still works for scaled icons
  - Animation still works for scaled icons
  - (Can text be scaled? When?)

Please add a comment if I've missed anything.

Bob

rhwood, 2015-12-21:
Does the refactoring consider the periodically repeated requests to allow Railroad Panels to be opened as Swing Panels without requiring an encapsulating Frame, or is that a separate concern?

bobjacobsen, 2015-12-21:
> Does the refactoring consider the periodically repeated requests to allow Railroad Panels to be opened as Swing Panels without requiring an encapsulating Frame?

I think getting clean implementations of the *Icon classes is the first step toward that.  So I think the refactoring here will definitely help.  But it's not likely to be sufficient by itself.

petecressman, 2016-01-14:
Text can be decorated and constrained.
1. Margins can be specified in pixels to extend the bounding box.  The margin are may be transparent or have the background color of the text.
2. Borders can be specified in pixels to surround the the margin area and have color.  A transparent background can still be bordered.
3. Text width and/or height be specified to be a fixed size.

Any text, including its decorations, can be rotated to an arbitrary degree.

Loco Icons and the "Overlay mode" of Sensor Icons are both text and icon.  Any decorations or properties associated with the text can be rotated.

Sensor Icons have a "Text mode"  Any decorations or properties associated with the text can be rotated.

As far as I know, animated icons have never be able to rotate to an arbitrary degree.

petecressman, 2016-01-23:
I have a near "proof of concept" for rotating, scaling and mirroring animated icons with or without text and all their decorator properties.  Getting the boundary rectangle properly specified for the transformed figure is the remaining problem.  I would like to refactor the panel figures to use this code but cannot push anything to GitHub for review.

bobjacobsen, 2016-01-23:
>  I would like to refactor the panel figures to use this code but cannot push anything to GitHub for review.

What happens when you try to do that?

Which method (command line, GitHub Desktop program, etc) are you using?

petecressman, 2016-01-24:
Bob,
For some time Eclipse has marked positionableLable.java with an error symbol, but there is no indication of the conflict in the file or otherwise in a message.
Frankly, I would just as soon remove everything locally and remotely and start fresh.  However here is what the my current state is:

First I will describe what git commands I have tried from Eclipse and then what commands I have done from the git shell.

Eclipse has marked both the top help and the jmri directories with 
"[jmri|Conflicts master |8|235"
I used the Merge Tool of Eclipse from the top jmri directory but still got no information about conflicts in positionableLable.java.  However a text compare is shown between 2 log files where one has an additional line.
Log "Fix Blinking Sensor icons display - 25d19a6..." andLog "Merge pull request #412 from DanielBoudreau/master - fb65a76..."
the line
/.idea/workspace.xmlis in the first file and not the second

# build artifacts
jmri.jar
/dist
java/docs

# test artifacts
junit-results.xml
junit*.properties

# runtime artifacts
messages.log*
session.log

# transient artifacts
.ant-targets-build.xml
/java/classes
/java/tmp
/java/doc
*.class
/temp
.pydevproject
*.swp
/lib/cachedir
/xml/XSLT/DTD
/xml/XSLT/properties
/xml/XSLT/xml
/xml/XSLT/pages

# private / per-user configuration files
/nbproject/private
/.idea/workspace.xml
local.properties

# CATS source, sometimes present for test
/java/src/cats

# OS junk files
[Tt]humbs.db
*.DS_Store
/jmri-fb.html
I used the Merge Tool of Eclipse from the display directory and this time configurexml\positionablrLabelXml was shown in a compare, also titled
"Fix Blinking Sensor icons display - 25d19a6..." and"Merge pull request #412 from DanielBoudreau/master - fb65a76..."
The discrepancy is the movement of line 389 to 394        l.rotate(deg);

From Eclipse a commit command shows the message: Cannot commit now "Repository Conflicts"

From Eclipse a pull command shows the message: Cannot pull into a repository with state MERGING "WrongRepositoryStateException"
From Eclipse a push appears to work but nothing is shown at my GitHub fork
****************************************I tried to duplicate the same commands from the git shell as follows:

C:\git [master +4 ~0 -0 !]> git merge origin/master
merge: origin/master - not something we can merge
C:\git [master +4 ~0 -0 !]> git add jmri\java\src\jmri\jmrit\display\positionableLable.java
fatal: pathspec 'jmri\java\src\jmri\jmrit\display\positionableLable.java' did not match any files
C:\git [master +4 ~0 -0 !]> git merge origin/master
merge: origin/master - not something we can merge
C:\git [master +4 ~0 -0 !]> git commit -m "Fix issue 567 - icon positioning bug"
On branch master
Untracked files:
        .metadata/
        .recommenders/
        jmri/
        oldGit/

nothing added to commit but untracked files present
C:\git [master +4 ~0 -0 !]> git push all
fatal: The current branch master has no upstream branch.
To push the current branch and set the remote as upstream, use

    git push --set-upstream all master

C:\git [master +4 ~0 -0 !]> git push --set-upstream all master
fatal: 'all' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
C:\git [master +4 ~0 -0 !]>
*************************As you see, i lack understanding on how to use git.
Most likely it will be some time before I can become a reliable user, so in the meantime, issue bug 567 can be fixed with a one line change:
In the overridden call in PositionableLable
    public void setText(String text) 
change 
        _text = (text !=null);to        _text = (text !=null && text.length()>0);
(when "" is entered for text, and a font has been specified, the descender distance moves the position.)
Thanks,Pete

 

 

What happens when you try to do that?

Which method (command line, GitHub Desktop program, etc) are you using?

bobjacobsen, 2016-01-26:
@petecressman:

> In the overridden call in PositionableLable
>     public void setText(String text) 
> change 
>         _text = (text !=null);to        _text = (text !=null && text.length()>0);
> (when "" is entered for text, and a font has been specified, the descender distance moves the position.)

Sorry you’re having trouble with Git.  

I put that change in #712, will try to get a closer look at it this evening.

Bob

bobjacobsen, 2018-01-23:
The original issue (with icon base code) seems resolved now, and there are tests in place.

closed 2018-01-23 by bobjacobsen