JMRI/JMRI #426: Line Ends in Git?

bobjacobsen, 2015-12-28:
(See #67 and #66 for some earlier context)

We might still have a problem with line-ends in the repository.  

For an example, see:
https://github.com/JMRI/JMRI/blame/master/java/src/jmri/jmrix/rfid/serialdriver/SerialDriverAdapter.java

We've lost easy access to the history for that file because a line-end mismatch makes Git think that the entire file was replaced.  But note the date on the commit:  Sept 9.  That's slightly _before_ we moved to GitHub.

Another example
https://github.com/JMRI/JMRI/blame/master/java/src/jmri/jmrix/loconet/LnTurnout.java

Here, it's not obvious that the whole file is replaced, so one might not suspect that a lot of history has been lost. You'd be misled by the history you're looking at. But again the date is _before_ the move.

Do we have a problem now? Or is this just an artifact of the conversion process?

Here's the list of files that currently have CRLF line ends (I've removed a few that just got committed that way, see commit #425):

	modified:   java/src/jmri/jmrit/display/controlPanelEditor/shape/LocoLabel.java
	modified:   java/src/jmri/jmrit/display/controlPanelEditor/shape/configurexml/LocoLabelXml.java
	modified:   java/src/jmri/jmrix/can/cbus/CbusConfigurationManager.java
	modified:   java/src/jmri/jmrix/can/cbus/CbusReporter.java
	modified:   java/src/jmri/jmrix/can/cbus/CbusReporterManager.java
	modified:   java/src/jmri/jmrix/can/cbus/configurexml/CbusReporterManagerXml.java
	modified:   java/src/jmri/jmrix/loconet/Intellibox/IBLnPacketizer.java
	modified:   java/src/jmri/jmrix/loconet/LnPacketizer.java
	modified:   java/src/jmri/jmrix/loconet/LnTurnout.java
	modified:   java/src/jmri/jmrix/loconet/LnTurnoutManager.java
	modified:   java/src/jmri/jmrix/loconet/lnsvf2/LnSv2MessageContents.java
	modified:   java/src/jmri/jmrix/loconet/uhlenbrock/UhlenbrockPacketizer.java
	modified:   java/src/jmri/jmrix/loconet/uhlenbrock/UhlenbrockSystemConnectionMemo.java
	modified:   java/src/jmri/jmrix/rfid/protocol/em18/COPYING
	modified:   java/src/jmri/jmrix/rfid/protocol/em18/Em18RfidProtocol.java
	modified:   java/src/jmri/jmrix/rfid/protocol/em18/package.html
	modified:   java/src/jmri/jmrix/rfid/serialdriver/SerialDriverAdapter.java

I'm unhappy that the "blame" history in those jmrix/loconet files have been lost, and really don't want to lose any more.  If this is just an unfortunate artifact of the SVN-to-Git conversion process, then I guess it is what it is.  But I really don't want any more of this to be happening.

Should I recommit those files with LF ends?  Or is there some other way forward?

Bob

rhwood, 2016-01-04:
This needs to be fixed with a ```.gitattributes``` file in the repository itself.

bobjacobsen, 2016-01-06:
See discussion of possible Travis tests in #498

bobjacobsen, 2016-01-07:
Reminder that there's conversation about this in PR #498 in case people are not watching PRs.

DanielBoudreau, 2016-01-07:
BobI'm away. Are any of my commits causing problems? Using Eclipse and Win7.DanSent from my Verizon Wireless 4G LTE Droid

bobjacobsen, 2016-01-07:
@DanielBoudreau - No, your commits are not a problem. For example, a recent one (763b1e318a9523) changed just a a few lines in a  LF-terminated file. That's fine.

That might mean that Eclipse is handling these files properly. Did you install any plugins or do any configuration for this?

DanielBoudreau, 2016-01-07:
Nothing that I haven't documented. So pretty basic.DanSent from my Verizon Wireless 4G LTE Droid

KenC57, 2016-01-07:
I thought there was a setting about line endings in Eclipse. But at least
the current version I have (Mars.1) I couldn't find anything.

 

-ken c

bobjacobsen, 2016-01-08:
(Ongoing discussion in #498, but this is the right place for larger discussion) 

I've done some testing on this.

We have to have consistent line ends _before_ installing the .gitattributes file, or people will have confusing problems.  So the steps above have to be reordered:

- After 4.3.2, probably mid-morning Saturday Pacific time, I'll commit/PR/merge the necessary line-end changes outside outside PR #498 (which contains the proposed .gitattributes file).
- Then I'll fix what's needed to merge PR #498, and merge it; that installs the .gitattributes file

From there, we can go back to updating instructions, etc, outside this PR.

bobjacobsen, 2016-01-10:
See #565 for a PR that's supposed to install a .gitattributes file and get this ready to go.

rhwood, 2016-01-14:
New files in Eclipse on Windows get the wrong line endings, but existing files do not get changed. See #604.

bobjacobsen, 2016-01-20:
#679 has a comment about line ends and the NetBeans plugin.

SteveRawlinson, 2016-01-22:
I've recreated the JMRI project in netbeans and once again it defaults to CRLF. When I change the setting to LF this line gets added to nbproject/project.properties:

auxiliary.com-junichi11-netbeans-changelf.lf-kind=LF

Maybe that should be in there be default?

rhwood, 2016-01-22:
@SteveRawlinson Please include the changed nbproject/project.properties in a PR. It appears that when I added the plugin's other settings I missed that because I use OS X where LF is the native line endings.

SteveRawlinson, 2016-01-22:
@rhwood Sure, see PR #691

bobjacobsen, 2016-01-25:
I think this is all set now (though I've been wrong on this topic before)

closed 2016-01-25 by bobjacobsen