JMRI/JMRI #429: Serial adaptor settings in UNIX systems

rhwood, 2015-12-28:
UNIX (Linux and OS X) systems exhibit the following errors when listing serial adaptors:

- OS X (tested on El Capitan 10.11.2):
  - [ ] In use adaptor is highlighted in red (it should not be)
  - [ ] Available adaptors are not listed if a serial adaptor is in use
  - Problems cannot be reproduced if gnu.io.rxtx.SerialPorts system property is *not set* (see PR #407 which ensures this property is not set in launch script for OS X-style adaptors)

- Linux (tested on Fedora 23):
  - [ ] In use adaptor is highlighted in red (it should not be)
  - [ ] Available adaptors are not listed if gnu.io.rxtx.SerialPorts system property is *not set* (this is the exact opposite behavior of OS X)

bobjacobsen, 2015-12-28:
To get some user feedback, I merged PR #407 (Quick fix for issues selecting ports on OS X) and there's now a set of development-build installers for this at http://builds.jmri.org/jenkins/job/Development/job/Packages/2832/

When I use the Mac version of that download, everything appears normal.  PanelPro gives me a normal hardware connection with the configured device, the preferences show other ports, and everything seems fast as it typically has been.  DecoderPro also OK: status line is green, configured device shows, preferences list populated.  'print java.lang.System.getProperty("gnu.io.rxtx.SerialPorts")' shows 'None'

Building with ant from the current HEAD of JMRI/JMRI master does the same things - all good. 'print java.lang.System.getProperty("gnu.io.rxtx.SerialPorts")' shows 'None'

Mac OS 10.10.5, with LocoBuffer-USB, PR3.

Should we ask some jmriusers people having trouble to try this?

rhwood, 2015-12-28:
Does 4.2 as shipped exhibit the problem behaviors on OS X 10.10.5 or is this only being observed on 10.11.2?

bobjacobsen, 2015-12-28:
Using a new download of JMRI.4.2-R6f3c1be.dmg on MacOS X 10.10.5 I see:
- PanelPro
  - Starts up OK with defined port, shows OK on splash screen
  - short (3-4 seconds) but noticeable delay when opening preferences.
  - port is red in list in preferences
  - port list displays as a blank white block when trying to select another port
- DecoderPro
  - Starts up OK with defined port, items show green at bottom
  - Similar delay to preferences, shows port red, similar white block

So I'd say it's showing the reported behavior

bobjacobsen, 2016-01-02:
#407 is in both JMR 4.2.1 and 4.3.1. It seems to have fixed the problem. 

I updated the 4.2 release note to defer to the 4.2.1 download for MacOS X. Should I do that for Linux too?

rhwood, 2016-01-02:
There's no need to change the release notes for the Linux download, only OS X was broken (Linux breaks if the property *is not* set, while OS X breaks if the property *is* set).

bobjacobsen, 2016-01-02:
Thanks. In that case, I'm closing this.

rhwood, 2016-01-02:
Reopening, since 4.3.1/4.2.1 only contain partial fixes for this. Will assign milestone once completely fixed.

rhwood, 2016-02-17:
This was fixed with #502 in 4.3.2.

closed 2016-02-17 by bobjacobsen