JMRI/JMRI #434: Use of InvokeAndWait by AbstractMRTrafficController

bobjacobsen, 2015-12-28:
See initial discussion in #422 

AbstractMRTrafficController uses invokeAndWait instead of InvokeLater to put events into the system via the Swing event queue.

Easy to change, but what are its consequences?

The children of AbstractMRTrafficController are:

- [ ] jmri.jmrix.can.AbstractCanTrafficController
- [ ] jmri.jmrix.AbstractMRNodeTrafficController (which has seven systems as children of its own):
  - [ ] jmri.jmrix.acela.AcelaTrafficController
  - [ ] jmri.jmrix.ieee802154.IEEE802154TrafficController
  - [ ] jmri.jmrix.cmri.serial.SerialTrafficController
  - [ ] jmri.jmrix.grapevine.SerialTrafficController
  - [ ] jmri.jmrix.maple.SerialTrafficController
  - [ ] jmri.jmrix.oaktree.SerialTrafficController
  - [ ] jmri.jmrix.secsi.SerialTrafficController
- [ ] jmri.jmrix.dcc4pc.Dcc4PcTrafficController
- [ ] jmri.jmrix.dccpp.DCCppTrafficController
- [ ] jmri.jmrix.easydcc.EasyDccTrafficController
- [ ] jmri.jmrix.ecos.EcosTrafficController
- [ ] jmri.jmrix.jmriclient.JMRIClientTrafficController
- [ ] jmri.jmrix.jmriclient.JMRIClientTrafficController
- [ ] jmri.jmrix.marklin.MarklinTrafficController
- [ ] jmri.jmrix.nce.NceTrafficController
- [ ] jmri.jmrix.rfid.RfidTrafficController
- [ ] jmri.jmrix.powerline.SerialTrafficController
- [ ] jmri.jmrix.tmcc.SerialTrafficController
- [ ] jmri.jmrix.srcp.SRCPTrafficController
- [ ] jmri.jmrix.tams.TamsTrafficController
- [ ] jmri.jmrix.lenz.XNetTrafficController
- [ ] jmri.jmrix.roco.z21.z21TrafficController

and some of those have connection-specific subclasses within their package (c.g. the powerline package).  Many of them get  lot of use (C/MRI, NCE, Lenz) but others have just a couple of users (maple, SECSI) and very little developer support. 

#422 made the synchronization a run-time variable (which someday might be a sub-classed method).  Should we roll it out to more systems, and if so, how?

rhwood, 2015-12-29:
If I understand the problem correctly, invokeAndWait gives all listeners a chance to change the system state before the rest of the processing in handleOneIncomingReply() by invoking them on the UI thread in a Runnable.

However, I don't see what state can be changed by a listener that changes how handleOneIncomingReply completes its processing. I am also under the impression that the Runnable passed in invokeLater(Runnable) is guaranteed to be run in the order in which Runnables were queued by invokeLater(Runnable). So if listeners notified by the Runnable can't change how handleOneIncomingReply processes its data, and are guaranteed to be notified in packet order, what does the invokeAndWait buy us that invokeLater does not?

I think it buys us the ability to ensure all state is changed before the next reply is read, but don't know if that's a good thing or not.

bobjacobsen, 2015-12-29:
I'll use C/MRI as an example.  There are three threads running:  line transmit (xmt for short), line receive (rcv) and Swing/AWT (AWT).

When the end of an incoming message is reached, the rcv thread drops through loadChars at AbstractTrafficController (ATC) line 852, notifies everybody of the response which processes inputs which can in turn generate output requests (see below).  It then looks at the state, and in normal operation does a notify() of the xmt thread.

For C/MRI, the notify()ed xmt thread then runs the pollMessage() routine that goes through checking if nodes need to be initialized, if nodes have output that should be sent, and finally if any nodes need to be polled. It sends the 1st thing that scan finds.

invokeAndWait ensures that any output that's been generated by a poll response will go out right away.  InvokeLater ensures that it won't.  (The AWT/Swing events happen in the same order, because invokeLater and invokeAndWait invoke at the same point in the queue; they only differ in whether the posting thread waits)

Is that a problem?  I don't know.  I'm pretty sure that it wasn't thought through when that code was originally written.  I don't think the "synchronized" attributes were checked with that in mind.  Maybe it's fine.  But I don't know that it is.

rhwood, 2015-12-29:
@pabender @bobjacobsen Thanks for the explanations here and in #422.

I'm thinking this problem can be safely ignored unless someone thinks they get a performance boost out of testing if InvokeLater() works for a specific system. I'm thinking that without physical access to a real system, changes will be hard to test.

closed 2015-12-31 by bobjacobsen