JMRI/JMRI #445: default Decoder Read setting is Paged

billchown, 2015-12-29:
As most decoders now work in Direct Bit mode, and all modern sound decoders do so, please can the default mode out-of-the-box be changed to Direct Bit.
Ideally, there will be a user defined Preference, (or the users selected value be saved, for instance by saving Window, or some such,) but keeping the default to the antique Paged mode is no longer useful for most users.

bobjacobsen, 2016-01-04:
Could you be a bit more specific?

DecoderPro, for example, checks the modes in the decoder definition and in the DCC hardware, and picks in the order you can see at line 628-660 of [PaneProgFrame].  (https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/symbolicprog/tabbedframe/PaneProgFrame.java). That should be choosing the direct modes before paged, if they're available. 

If that's not working, could you provide enough detail to reproduce?  DCC system type, decoder type, how you're opening the programmer, etc.

Thanks

billchown, 2016-01-04:
Bob,thanks for the response, and I fully understand the way in which the decoder read process is supposed to operate once a decoder is identified. And in my experience it usually works.
It is the out-of-the-box default that is my concern.Start DecoderPro, and the Programming Mode is Paged.Click Identify and it can take over a minute to find a loco, or Click New Loco and then Read Type from decoder, and an impatient user will see it as "it  doesn't read my decoder", and that is my big issue.
I am the distributor of the SPROG devices, and get questions from users about this; why does it take so long, etc., and most of my customers, it seems, are not especially computer-savvy, and don't know about this part of how a decoder-programmer interaction works.As it seemingly would be simple to start in Direct Bit mode, I am requesting that.I realize that an ancient decoder might not read if it does not support Bit mode, but I also believe that most users of those older decoders are also early-adopter types, and will deal with it more easily that the new purchaser of a sound loco making their first foray into DCC.
If I knew where in the code that is set, I would build my own version, but I haven't found it so far, and so I'm raising this issue.
ThanksBill
 Bill Chownorrrbrit2004@yahoo.comhttp://bchown.com

mattharris, 2016-01-04:
Bill,

The default programming mode is (initially) set at line 128 of  [AbstractProgrammer](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/AbstractProgrammer.java)

Now, I would __NOT__ recommend changing it here as this _will_ likely mess up other programming interfaces.

However, it _should_ be possible to override the default in the [SprogProgrammer](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/sprog/SprogProgrammer.java) implementation.

Specifically, if you change the constructor from:

    public SprogProgrammer() {
    }

to:

    public SprogProgrammer() {
        setMode(DefaultProgrammerManager.DIRECTBITMODE);
    }

this _might_ do the trick.

billchown, 2016-01-04:
Thank you, I'll give it a try there.Bill
 Bill Chownorrrbrit2004@yahoo.comhttp://bchown.com

bobjacobsen, 2016-01-05:
We can certainly change the overall (all systems) default is that's the right thing for the general user. But different hardware systems do different things, so maybe just changing the default for SPROG is a good 1st step.  Happy to do that if you want.

We can also persist the mode choice so that, once set, it comes back when the program starts up again. That's a bit more work than just changing the default (mostly because of having to handle the case of more than one programmer), but it's not that hard.

dheap, 2016-01-05:
@bobjacobsen
> We can certainly change the overall (all systems) default is that's the right thing for the general user.
> 
I think it's time to revisit that one:
- Most systems and decoders now support a Direct mode.
- Some decoders need to read quite a few CVs (at least six for ESU decoders) and that takes quite a long time in paged mode.
- There was a problem that no retries were performed on a read failure in Identify and Read Type. Direct mode is a bit more prone to failure than paged mode. But I fixed that some time ago and retries are now performed in Identify and Read Type.

My suggestion is that the default be set to a Direct Mode (if the programmer supports it) and if an Identify or Read Type fails the retry count, falls back to Paged mode (probably just for the duration of that Identify or Read Type).

Once a roster entry is open, the code and XML already cooperate to pick the best mode.

billchown, 2016-01-05:
Dave,I agree, and the additional factors that you have addressed for retries will help make it even more robust.ThanksBill
 Bill Chownorrrbrit2004@yahoo.comhttp://bchown.com/

bobjacobsen, 2016-01-05:
I think the addition of "Direct" (as opposed to DirectByte and DirectBit) makes it possible for use to have a global Direct default now.  Before, since neither DirectByte nor DirectBit was really widely available, picking one or the other would have caused problems.  But Direct is much smarter about working with the command stations capabilities.

bobjacobsen, 2016-01-06:
"Direct" as default seems to work on the systems I can test from here.  @billchown, if you can confirm that it works for SPROG, I'll go ahead and change it befor 4.3.2 this weekend. Thanks.

billchown, 2016-01-06:
Bob,yes, Direct works for SPROG, and was the default until not too long ago!Please make the change.Bill
 Bill Chownorrrbrit2004@yahoo.comhttp://bchown.com/

bobjacobsen, 2016-01-07:
Done: See PR #517

closed 2016-01-07 by bobjacobsen