JMRI/JMRI #448: DCC++ Serial Flow control

Dwayne75, 2015-12-29:
When I enable hardware flow control with DCC++ it breaks in master branch. After receiving the initial message from the Arduino it hangs. Here is a snip from the log

2015-12-29 15:12:30,585 dccpp.DCCppReply                      DEBUG - DCCppReply.toString(): char i i msg iDCC++ BASE STATION FOR ARDUINO MEGA / ARDUINO MOTOR SHIELD: BUILD Dec 25 2015 12:40:17 [Receive]
2015-12-29 15:12:30,585 jmrix.AbstractMRTrafficController     DEBUG - dispatch reply of length 87 contains iDCC++ BASE STATION FOR ARDUINO MEGA / ARDUINO MOTOR SHIELD: BUILD Dec 25 2015 12:40:17 state 25 [Receive]
2015-12-29 15:12:50,139 dccpp.DCCppSerialPortController       DEBUG - CTS: false [Transmit]
2015-12-29 15:12:50,139 dccpp.DCCppPacketizer                 DEBUG - DCC++ port not ready to receive [Transmit]
2015-12-29 15:12:50,139 jmrix.AbstractMRTrafficController     DEBUG - Retry message: s attempts remaining: 5 [Transmit]

In dccpp.DCCppSerialPortController 
    public boolean okToSend

With flow control enabled it returns activeSerialPort.isCTS() which always stays at false.
With flow control disabled things work.

In Realease-4.2 even with flow control enabled. activeSerialPort.getFlowControlMode() is false. so it never returns activeSerialPort.isCTS().

I looked at the code on the Arduino and I couldn't see any mention of hardware flow control, so perhaps its off by default or simply not supported?

bobjacobsen, 2015-12-29:
Sounds like the command station is not implementing its part. 

Is there any need for flow control to the command station?

If the command station doesn't support it, perhaps the option should be removed.

bobjacobsen, 2016-01-02:
@Dwayne75 @msunderwd Could somebody please confirm whether a DCC++ command station should be able to do hardware (CTS) flow control?  That's something that has to be built into the command station software, with some kind of controlled input queue.  

If it's not there, we should remove it from JMRI: Change file java/src/jmri/jmrix/dccpp//serial/DCCppAdapter.java to remove the "option1" references at lines 41-42, 296-302, 319-320.

msunderwd, 2016-01-02:
Yes, it should. Or at least hardware flow control does work on my Mac (and
does not work on the PC). I will test whether no flow control works on the
Mac this evening and if so I will remove hardware flow control as an option.

Aside from possible user confusion, what is the issue with leaving it in as
an option?

- Mark

bobjacobsen, 2016-01-02:
>Aside from possible user confusion, what is the issue with leaving it in as an option?

This is an open Issue, so we should try to resolve & close it....

But I have no strong opinion as to whether the option should stay in the code or not.

Hardware flow control is generally a wiring issue:  The control leads either have to be connected to an active voltage, or run into some pin and controlled by the software. 

Most simply, the DCC++ command station hardware could statically set the RS232 control leads to active, so the option would work either way.

msunderwd, 2016-01-02:
I've confirmed that both "no flow control" and "hardware flow control" work on my Mac...

Since it's probably easier than fixing the hang, I will remove "hardware flow control" as an option.

If at some later time hardware flow control is required, then we can fix the hang...

msunderwd, 2016-01-02:
Fixed in PR#481.

- Mark

closed 2016-01-02 by bobjacobsen