JMRI/JMRI #501: SoundPro not liking WAV metadata

mattharris, 2016-01-05:
Based on issue reported in [jmriusers](https://groups.yahoo.com/neo/groups/jmriusers/conversations/messages/123377) it would appear that the JMRI Audio tools do not like the metadata occasionally added to WAV files.

Need to investigate how to deal with this.

mattharris, 2016-01-05:
Attached file gives example of problem

[level crossing low level LOOP.zip](https://github.com/JMRI/JMRI/files/78182/level.crossing.low.level.LOOP.zip)

klk32003, 2016-01-05:
Matt,

This issue appeared firstly with JMRI Version 3.11.7.  I was working with "Virtual Sound Decoder" when I noticed an unwanted noise at the end of some WAV files. My first thought was that the Jogamp version of the JOAL libraries is not as "tolerant" as the SUN version concerning meta data.

Might be OS dependend. My OS is Windows (Windows 7 Professional and Windows XP). I have installed JMRI Release 4.1.2 and Java 8u66.

Thank you for taking care of this issue.

Klaus Killinger

mattharris, 2016-01-05:
Thanks Klaus - that is a likely explanation.

Will help narrow down the cause.

mattharris, 2016-01-05:
A quick look at the Jogamp WAVLoader would suggest that they are not correctly determining the size of the data chunk of the wav file.

It would seem as though they are using the size of the 'RIFF' chunk as opposed to the size of the 'data' chunk when determining how much data to load. Given that the example file has it's metadata at the end of the 'data' chunk explains why this extra 'noise' is added to the file.

mattharris, 2016-01-06:
OK, I think that I've tracked down the issue.

As suspected, it's in the Jogamp code and have a suitable fix for it.

Now, we've got three options:

1. Provide patch upstream to Jogamp and wait for them to incorporate and provide an updated release which we can then incorporate into JMRI
2. Provide patch upstream to Jogamp but include our own version of joal with the fix applied
3. Copy the classes needing modification from Jogamp into jmri and use 'our' version

In any case, we should provide the patches upstream - that is the whole point of Open Source after all. The question remains as to how deal with our immediate need.

Note: I've provided the proposal upstream at https://github.com/sgothel/joal/pull/15

rhwood, 2016-01-06:
How about using our own version of Jogamp until Jogamp produces a release that addresses the issue, at which point we revert to the Jogamp produced release?

mattharris, 2016-01-06:
That's pretty much what I was thinking of doing.

KenC57, 2016-01-06:
Given how many of those projects vary all over the calendar on when
something might be done, making our own might be better. And handling a
whole package of our own is likely cleaner than trying to mix the classes
that we need. It also should make it clean when we switch back, if we only
bring in the jar, we only have one thing to watch.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

mattharris, 2016-01-06:
Just done this - see PR #511

klk32003, 2016-01-06:
Great! Thank you, Matt. Thank you, all.

mattharris, 2016-01-06:
Closing as now incorporated for 4.3.2

closed 2016-01-06 by mattharris