JMRI/JMRI #503: JMRI JavaScript libary/documentation?

k4kfh, 2016-01-05:
I have found the new WebSockets server to be extremely useful, and it is very well put together. However, the documentation was a little lacking, and it took a little more blind trial/error than I would've liked to get things working. Additionally, some of the simple communication tasks were a bit of a hassle since I had to start "from scratch" instead of having a premade JavaScript library or something.

I was wondering if people would be interested in a JMRI JavaScript library (or maybe  jQuery plugin or something)? I have already written something of that sort for my own project, so I could tweak it just a bit and we could start with that. I also would love to see some better documentation for the WebSockets server (with or without the proposed JS library), so I would be happy to help write that. I don't know Java, but I would love to help provide some better examples and reference for people who want to use this.

Is any of this do-able?

rhwood, 2016-01-05:
It's in your JMRI installation under web/js/jquery.jmri.js but is incorrectly referred to in http://jmri.org/help/en/html/web/JsonServlet.shtml as jquery-json.js

Randall Wood

rhwood, 2016-01-06:
Documentation fixed in #506

closed 2016-01-06 by rhwood