JMRI/JMRI #509: Accept JVM arguments to Windows launcher

rhwood, 2016-01-06:
I've added the ability to [pass JVM arguments to the Linux / OS X launchers](#502) by prepending JVM arguments with ```-J```. Can this be added to the Windows launcher as well?

The pattern is taken from the NetBeans platform of allowing ```-JOPTION``` as a parameter, stripping the ```-J``` and passing ```OPTION``` to the JVM (for example, the argument ```-J-Dsun.java2d.hdiaware=true``` would be passed as ```-Dsun.java2d.hdiaware=true```).

mattharris, 2016-01-06:
I'll take a look. Although it will probably be a '/J' option as that is the usual way of doing that sort of thing in Windows-land.

rhwood, 2016-01-06:
I specifically want to use ```-J``` since it appears that Java uses ```-```s instead of ```/```s for arguments, even in Windows.

rhwood, 2016-01-06:
Actually, can we do both /s and -s for all arguments? This would allow existing scripts to not break, but make it possible to document a single set of arguments for all platforms using the - variant in the documentation.

rhwood, 2016-01-06:
Is ```/``` standard in Windows? PowerShell parameters all start with ```-``` ([reference](https://technet.microsoft.com/en-us/magazine/jj554301.aspx))

KenC57, 2016-01-06:
Long time ago, in a land far away..

 

CPM used / for options. Unix used - for options. Windows followed CPM since
CPM was PC at the time. As Unix grew and the whole concept of multiple
platforms came to be, Windows adopted the either or on a program by program
basis.

 

Meaning those programs that came from a Unix like development environment
got - while those that came from original Windows followed the / tag.

 

As far as I know that is still the case, neither is universal in the Windows
world.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

mattharris, 2016-01-06:
PowerShell != cmd.exe shell (usual Windows command line)

PowerShell brings a more POSIX style of operations to commands and the syntax is rather different in many cases.

mattharris, 2016-01-28:
This requirement should be covered by #728. Additionally, it is now possible to specify options using the '-' style as well as existing '/' style. So, to pass additional JVM options, use the form:

`-J-Doption1=value`
or
`/J-Doption1=value`

It is also possible to pass multiple options using the form:

`-J-Doption1=value -J-Doption2=value`

mattharris, 2016-01-28:
#728 is now merged

mattharris, 2016-02-03:
Closing as now merged.

closed 2016-02-03 by mattharris