JMRI/JMRI #522: Features around Programing Mode selection

bobjacobsen, 2016-01-08:
(See earlier discussion in Issue #445 and PR #517 and #521)

How to handle selection of the programming mode, particularly at program start-up and when there are multiple programmers?

It seems to me that programmers should (in order of decreasing importance):
- Always be in a mode that's valid for the use (this when a mode is automatically picked, it's a valid one; DP does this now)
- Work independently when there's more than one Programmer e.g. for multiple connections
- Remember how they were set, as a user convenience
- If all else fails, have a good default. 

The last one started the prior discussion: our always-pick-PAGED default probably isn't optimal. So we should make that system specific. We can do that. 

But I'm not sure how to do the "remember" bullet. Remember _what_?  The setting in e.g. DP changes automatically. Do we remember the last of those? Or the last single-CV programmer setting? Or something else?

dheap, 2016-01-08:
(moving this from #517)
> I think what we really need, given the realities of the hardware, is for systems to "pick default in some system-specific preferred order". This will let Digitrax's older systems stay with PAGEDMODE, which is the fastest one they have (yes, it is faster than those old system's implementation of Direct via DIRECTBYTE; that hardware isn't changing, because it's already in the hands of users) while letting SPROG and others pick their own best mode.
> 
If we set up a default  pick order algorithm of DIRECTMODE, DIRECTBITMODE, PAGEDMODE, DIRECTBYTEMODE that would satisfy almost every system (DIRECTBYTEMODE is slower on most if not all systems) that would satisfy almost everyone.

If we make that a default method in AbstractProgrammer as per below we can override that method in individual system Programmers. But we will need to invoke it before we first use a programmer (as in AbstractIdentify or when we first display a selector).

Maybe the best long-term way is to remove the assignment to PAGEMODE in line 128 of AbstractProgrammer and ensure all programmers invoke it in the constructor

    if (getMode == null ) setBestMode();
> From a software point of view, the best way to do that is to have an abstract method that the various systems have to (or, if there's a default, can) override to make the selection, instead of having it in the constructor. I can do that this weekend, but not in time for 4.3.2 this evening.

dheap, 2016-01-08:
> Maybe the best long-term way is to remove the assignment to PAGEMODE in line 128 of AbstractProgrammer and ensure all programmers invoke it in the constructor
> 
>     if (getMode() == null ) setBestMode();
Whoops. Guess that is a redundant if(). If we remove the assignment in AbstractProgrammer getMode() will always return null in the constructor.

mattharris, 2016-01-08:
Maybe I'm over simplifying but, if we use the order of the getSupportedModes list as the order in which we try things then, surely, the individual adapter implementations can influence this order.

We can then provide a UI that allows that order to be changed.

dheap, 2016-01-08:
> It seems to me that programmers should (in order of decreasing importance):
> 
> Always be in a mode that's valid for the use (this when a mode is automatically picked, it's a valid one; DP does this now)
This wasn't being checked. It was happening by default due to the assignment in AbstractProgrammer line 128:
    protected ProgrammingMode mode = DefaultProgrammerManager.PAGEMODE;
> Work independently when there's more than one Programmer e.g. for multiple connections
That seems to be happening fine already.
> Remember how they were set, as a user convenience
That isn't happening yet.
> If all else fails, have a good default.
My suggestion is a `setBestMode()` method that is called in the constructor for each extension of the AbstractProgrammer class. This simple overridable method in AbstractProgrammer would pick an available mode from `getSupportedModes()` in the priority order of DIRECTMODE, DIRECTBITMODE, PAGEMODE, DIRECTBYTEMODE. That would satisfy almost every system (DIRECTBYTEMODE is slower on most if not all systems).

Programmers without the `setBestMode()` method would continue to default to Paged Mode for now.

mattharris, 2016-01-08:
I've put an initial proposal together in the [prog-mode-order](https://github.com/mattharris/JMRI/tree/prog-mode-order) branch in my repo.

For a comparison, see:

https://github.com/JMRI/JMRI/compare/master...mattharris:prog-mode-order

Obviously no UI yet...

dheap, 2016-01-08:
> Maybe I'm over simplifying but, if we use the order of the getSupportedModes list as the order in which we try things then, surely, the individual adapter implementations can influence this order.

I've written code to add an optional `setBestMode()` method that can called in the constructor for each extension of the AbstractProgrammer class. I've added the call to PROG and NCE Power Pro systems and all seems to work well. I'm just running "alltest" to see if I've upset AbstractIdentify tests.

bobjacobsen, 2016-01-08:
> I've put an initial proposal together in the prog-mode-order branch in my repo.
> 
> For a comparison, see:
> 
> master...mattharris:prog-mode-order

I’ve got a slightly different mechanism for setting the Programmer’s default here:  https://github.com/bobjacobsen/JMRI/tree/default-progmode

By putting the configuration in a method, and making the variable private, the hope is to eventually be able to add read-from-preferences later.  It also lets Programmer implementations separate the function if (for some reason) they can’t fully implement getSupportedModes in their ctor.

Happy to add this to your branch as a PR if that’s OK.

mattharris, 2016-01-08:
A PR would be great.

mattharris, 2016-01-08:
In fact, I've generated the PR myself from your repo. Just need to sort out the merge conflict in AbstractProgrammer

dheap, 2016-01-08:
I've written code for an alternative proposal that adds a `setBestMode()` method that can called in the constructor for each extension of the AbstractProgrammer class. I've added the call to PROG and NCE Power Pro systems and all seems to work well.

I'll submit a PR against my repo (not a branch unfortunately) for inspection. But not for merging as although "alltest" passes on my Mac "ci-test" fails due to programmer facades that do not override an abstract method I have added to Programmer.

The advantage of mine is that we don't have to fiddle with order in specific programmers, just invoke `setBestMode()` in the constructor.

bobjacobsen, 2016-01-08:
> On Jan 8, 2016, at 1:57 PM, Dave Heap <notifications@github.com> wrote:
> 
> I've written code for an alternative proposal that adds a `setBestMode()` method that can called in the constructor for each extension of the AbstractProgrammer class. I've added the call to PROG and NCE Power Pro systems and all seems to work well.
> 
> I'll submit a PR against my repo (not a branch unfortunately) for inspection. But not for merging as although "alltest" passes on my Mac "ci-test" fails due to programmer facades that do not override an abstract method I have added to Programmer.
> 
> The advantage of mine is that we don't have to fiddle with order in specific programmers, just invoke `setBestMode()` in the constructor.

“setBestMode” as an operation might be a good thing to have in the Programming interface itself, so code can invoke that _after_ ctor time.

But I’d rather _not_ have that kind of function as an optional addition in ctors.  That’s what abstract and default methods are for:  They provide function which can be overridden as needed, but you don’t have a broken object if you forget to include them in your implementation.  It’s a separation of “needed & enforced by the compiler” vs “needed, and you just have to remember”. 

Hence having a method (which could be called setBestMode, that’s fine) in the actual AbstractProgrammer ctor which does something, always does something.  Subclasses can then either work with that (by setting order in a particular way; note that the order in the list is the order displayed, so this also has default-first-displayed, which either is or isn’t nice) or they can implement the method themselves.

mattharris, 2016-01-08:
@bobjacobsen your changes are now merged into my [prog-mode-order](https://github.com/mattharris/JMRI/tree/prog-mode-order) branch

dheap, 2016-01-08:
> Hence having a method (which could be called setBestMode, that’s fine) in the actual AbstractProgrammer ctor which does something, always does something. Subclasses can then either work with that [...]

The problem I had then was how to call setBestMode() at line 72 of AbstractIdentify. How could I do that?

mattharris, 2016-01-08:
Looking at [your proposed diff](https://github.com/JMRI/JMRI/pull/534/files#diff-21bec9ec6b9fcce973232638881bc59a) I'd suggest that placing the call to setBestMode() after recording the current mode would be the right approach.

So, first we save the current mode, then we setBestMode.

On exiting the routine, we revert the mode to that saved on entry.

mattharris, 2016-01-08:
(hit Comment too early...)

So, it would look like this:

        if (programmer != null) {
            savedMode = programmer.getMode(); // In case we need to change modes
            programmer.setBestMode();
        }

Look also at [what I've proposed](https://github.com/JMRI/JMRI/compare/master...mattharris:prog-mode-order#diff-21bec9ec6b9fcce973232638881bc59aL107) in AbstractIdentify.

dheap, 2016-01-08:
> 
> Looking at your proposed diff I'd suggest that placing the call to setBestMode() after recording the current mode would be the right approach.
> 
> So, first we save the current mode, then we setBestMode.
> 
> On exiting the routine, we revert the mode to that saved on entry.
> 
How to lose the wood for the trees. I don't actually need to call it. That means I can remove the abstract method from Programmer and undo all the consequent changes!!!

mattharris, 2016-01-08:
@bobjacobsen I had to laugh at your [AbstractProgrammerTest changes](https://github.com/mattharris/JMRI/pull/1/files#diff-cb15a2be5bf315b0c8b66f18c53cad12) as I'd just done the _exact_ same change and was about to commit when I saw the PR update.

dheap, 2016-01-08:
> Look also at what I've proposed in AbstractIdentify.

I've had a look at both proposals. I'm no doubt biased but I think mine passes the simplicity test as it only requires one line to be added to the constructor of each programmer brought into the fold, and it avoids tinkering around in the getSupportedModes code of individual programmers, some of which can be quite messy.

Different approaches to the same problem...

dheap, 2016-01-08:
With a cleanup my PR is down to only 3 files changed: 14 lines added to AbstractProgrammer and one line to each of NCE and SRPOG programmers.

bobjacobsen, 2016-01-08:
> I've had a look at both proposals. I'm no doubt biased but I think mine passes the simplicity test as it only requires one line to be added to the constructor of each programmer brought into the [...]

The other draft required only changing one subclass (the OpenLCB one, which wasn’t coded well in the first place) if the order didn’t need to be changed.  To change the order, either re-order in the list (for easy cases) or write a one-line @Override method that makes a clear statement of function.

I think it’s not a good practice to “require one line to be added”; that’s what interfaces and abstract classes are _for_.

Bob

dheap, 2016-01-09:
> I think it’s not a good practice to “require one line to be added”; that’s what interfaces and abstract classes are _for_.

It comes down to adding one line to the constructor of each programmer versus someone going through the sometimes quite messy code of each programmer to change the order, probably with no access to hardware to test the changes.

My proposal maintains status-quo for all existing programmers by default. Each can be brought into the fold by adding the method call to the constructor (because we can't add it to the abstract class). Not ideal, but neither is tinkering with method code in every programmer (we have never had a policy of ordering the return list in order of priority).

mattharris, 2016-01-09:
Naturally I'm biased also :smile:

However, the reasoning behind this proposal was that we've already got the list of suitable modes in-place and relying on the order of that list doesn't exactly break the existing API - [the JavaDoc notes that the order _might_ be significant](https://github.com/mattharris/JMRI/blob/28b89efe157c884f71014dce480bf7740688bf0b/java/src/jmri/Programmer.java#L145) - it rather enforces that the order _now becomes significant_

What will need to be reviewed is the order in each individual concrete implementation.

Additionally, this proposal is extensible if (when) shiny new programming modes become available.

Finally, from a UI perspective, sorting a list in the preferred order is not an unnatural thing to do, be it via drag-and-drop, up-down buttons etc.

bobjacobsen, 2016-01-09:
> It comes down to adding one line to the constructor of each programmer versus someone going through the sometimes quite messy code of each programmer to change the order, probably with no access to hardware to test the changes.

Apparently, I’m not being clear.

The other proposal _does_ _all_ _that_, with the exception that the one line is not in the constructor (which is an advantage, in my opinion).

If the person changing the default doesn’t want to reorder, they just write a one-line @Override method.

Bob

dheap, 2016-01-09:
> The other proposal _does_ _all_ _that_, with the exception that the one line is not in the constructor (which is an advantage, in my opinion).

I've just realised that Matthew's proposal adds his method to the constructor of AbstractProgrammer. I didn't realise you could do that. I could have done that with my proposal, removing the need to add code to each constructor. When I do, it works with SPROG but not with NCE (which has its own non-blank constructor).

Debugging shows my method is called but it possibly is too early to be able to get a list. I had to repeat setBestMode() in the NCE constructor (it actually being invoked twice). I see no reason why Matthews will not fail in the same way as mine with NCE programmer... (NCE has a selector in MASTER that needs removing first)

Food for thought.

Om going to be off-air for a while attending a daytime wedding of a relative.

dheap, 2016-01-09:
See my latest changes and comment on #534.

I think that makes my proposal more minimalist, as most programmers will require no changes.

rhwood, 2016-01-09:
For what it's worth, it's [not a good idea to include override-able methods in a constructor](http://stackoverflow.com/a/3404369).

bobjacobsen, 2016-01-09:
> For what it's worth, it's not a good idea to include override-able methods in a constructor.

Solving that issue is why the relevant methods are final and the data element is private. See also the comments.

bobjacobsen, 2016-01-09:
> For what it's worth, it's not a good idea to include override-able methods in a constructor.

Another approach to that would be for AbstractProgrammer to have no no-argument constructor, and always have to be constructed via a one-argument constructor that takes the default mode.

But this doesn’t really resolve the consistency issue, because it doesn’t compel the object to have consistency between the default mode and the list of available modes.  And you can’t pass in the list, because you can’t generate it (e.g. you'd have to invoke code) before invoking the super constructor.

Mostly, this kind of thing matters 

- when the mode logic is complicated; NCE is a good example of that. I pushed an update that moved DIRECT to the top of the list for NCE to Matt’s repository and it’s in my default-progmode branch.

and

- when the idea of a “default” mode is complicated, e.g. when it’s _sometimes_ set from a persistence mechanism.  

The combination of the two is particularly difficult.

bobjacobsen, 2016-01-09:
The most recent comments have gotten me thinking about combining Matt’s, David’s and Randall’s points.

Another way to implement this feature is something like this (comments elided for space):

``` java
    private ProgrammingMode mode = null;

    public final ProgrammingMode getMode() {
	if (mode == null {
		mode = getBestMode();
	}
        return mode;
    }

    public ProgrammingMode getBestMode() {
        return getSupportedModes().get(0);
    }

```
(and the ctor goes back to empty)

This allows reimplementing getBestMode() if teh programmer doesn’t want to have the modes in order with default at the top; gives a method that using code can use to get the best default; and defers setup until whenever the subclass is ready for it.

Should I code this up?

dheap, 2016-01-09:
> Should I code this up?

Looks a good idea. Not sure it will solve all the problems, but worth a try. I'll try similar and see how I go with a multi-setup

Just home and haven't looked at what you did to make NCE work, but no sooner had I left keyboard than I realised that the problem was that no modes were added until (tc != null) and that is why it didn't work in super().

dheap, 2016-01-09:
> getBestMode();

As it returns nothing but only sets mode, I think setBestMode() is a better name. I thought getters should return something and not make changes.

dheap, 2016-01-09:
Sorry, didn't read it correctly getBestMode() is no longer void.

> As it returns nothing but only sets mode, I think setBestMode() is a better name. I thought getters should return something and not make changes.

dheap, 2016-01-09:
I coded it up on my checkout and it seems to work fine with an NCE connection, a SPROG connection and also a dual SPROG and NCE connection.

I was thinking of coding up a version of AbstractIdentify that incorporates Matt's idea of iterating through the available modes, but I don't like the idea of the user having no control over the initial mode. I would have started with the current mode and then on a failure iterate through the list (skipping the one already tried).

But then I doubt whether there is any sense trying register mode given the limited number of CVs accessible in this mode compared to what we use in IdentifyLoco and IdentifyDecoder.

Given that and also that trying many modes will effectively tie up the programmer for a longer period (irritating the user who forgot to put the loco on the track) and run the risk of a "programmer in use error", I think the status quo of "try selected mode first and if it fails fall back to Paged mode if available and try that" should remain.

mattharris, 2016-01-09:
I like what @bobjacobsen has done in #541 as it clearly shows the behaviour of preferred programming mode selection and mitigates the potential issue @dheap highlighted in my proposed AbstractIdentify changes, namely:

> ...that trying many modes will effectively tie up the programmer for a longer period (irritating the user who forgot to put the loco on the track) and run the risk of a "programmer in use error"...

i.e. iterating through all available programming modes.

Now, when starting to look at this yesterday, I hadn't fully realised that the programming mode changes in AbstractIdentify were an extremely recent addition - #521 less than 24 hours ago.

dheap, 2016-01-09:
I'd like AbstractIdentify left as it was immediately post #521 for now. I can't see how Bob's changes mitigate the problems I pointed out, but I'll experiment once he merges his changes.

> Now, when starting to look at this yesterday, I hadn't fully realised that the programming mode changes in AbstractIdentify were an extremely recent addition - #521 less than 24 hours ago.

mattharris, 2016-01-09:
Bob's #541 doesn't touch `AbstractIdentify` - it would remains as it is in HEAD of master.

Going through the changes, what's been done is to set the default to `PAGEMODE` in `AbstractProgrammer` and to make that private - in other words, no concrete implementations can override this setting.

Now, `getBestMode` will now look to the first entry in the list returned by `getSupportedModes` for the preferred programming mode, unless a concrete implementation overrides that method.

`setMode` will, if `mode` is not yet set, use `getBestMode` for the programming mode.

So, we have pretty much everything in the `AbstractProgrammer` and we rely on either the order of the list returned by `getSupportedModes` or `getBestMode` being overridden in a concrete implementation.

There _could_ be a potential conflict with the initialisation of `mode` - I've noted that as a comment in #541.

bobjacobsen, 2016-01-09:
@mattharris, thanks for catching all that.

I messed up when copying the code from email, and forget to change the initialization to `= null`  Then, to make it worse, I picked a test sequence that didn't highlight the difference.

#541 has had fixes for those pushed.

`AbstractIdentify` remains the same, I haven't touched that. No strong opinion on it.

bobjacobsen, 2016-01-12:
#541 closes the original issue, so I'm going to close this.

At some point, it would be good to remember the current mode as a stored preference.  Given that the mode is automatically changed, not just manually, I'm not sure exactly what that means.  "Remember last manually chosen"?  "Remember last set, even by the program"? Something else?

But that's a topic for another day, and another Issue.

closed 2016-01-12 by bobjacobsen