JMRI/JMRI #542: null pointer exception saving layout xml file

SteveRawlinson, 2016-01-09:
If you delete a memory which was attached to a layout block the subsequent save of the xml file raises a null pointer exception.

The reason is that a layout block with no assocated memory has a blank string for memoryName, and this is what the save function checks for before dereferencing the pointer to the memory. However the means of removing a memory from the layout block can set the memoryName to null rather than "".

Patch attached.
[patch.txt](https://github.com/JMRI/JMRI/files/83601/patch.txt)

mattharris, 2016-01-09:
@SteveRawlinson: Given that this is a small patch, I'll add this manually.

However, for the future, could you commit changes to your checkout of the code and then push to your fork on GitHub.

From there, you can submit a Pull Request.

Makes it easier to handle.

SteveRawlinson, 2016-01-09:
Sure. I was doing my best to follow the guidelines at 

http://jmri.sourceforge.net/help/en/html/doc/Technical/index.shtml

"For your first small contributions, e.g. [...] small bug fix, please fill out a new "Patches" entry[...]. You can attach the entire file there." 

Maybe those instructions need to be updated.

mattharris, 2016-01-09:
As you might guess, we're still in a bit of a state of flux with regard the move to GitHub and how best to deal with contributions.

Anyhow, I've applied your patch but will make a small modification as there is the technical possibility of `name` being non-null at one of the points and we'd be ignoring that fact, so:

```java
    public void setMemory(Memory m, String name) {
        if (m == null) {
            namedMemory = null;
            memoryName = name == null ? "" : name;
            return;
        }
        namedMemory = jmri.InstanceManager.getDefault(jmri.NamedBeanHandleManager.class).getNamedBeanHandle(name, m);
    }
```

instead of:

```java
    public void setMemory(Memory m, String name) {
        if (m == null) {
            namedMemory = null;
            memoryName = "";
            return;
        }
        namedMemory = jmri.InstanceManager.getDefault(jmri.NamedBeanHandleManager.class).getNamedBeanHandle(name, m);
    }
```

Thanks again though for the fix.

mattharris, 2016-01-09:
Modifications included in #545

mattharris, 2016-01-09:
Now merged - thanks!

closed 2016-01-09 by mattharris

bobjacobsen, 2016-01-09:
> Sure. I was doing my best to follow the guidelines at
>
> http://jmri.sourceforge.net/help/en/html/doc/Technical/index.shtml
>
> "For your first small contributions, e.g. [...] small bug fix, please fill [...]

First, thanks for the fix. That’s the most important thing.

I updated that page (and put back #patch in NetBeans page) in #556.  That should be up on the web shortly. 

It’s hard to be certain about the best way to explain things to the wide range of people who might want to contribute to JMRI code.  We want to encourage people to contribute even tiny things, a spelling fix for example, _as_ _code_.  

Getting an email requesting a tiny change isn’t all that great, because it doesn’t really engage the person who wants the change (they might be come a real contribute in time), and it’s work for others who don’t really want to be “working” on their hobby.  So we want to lower the barrier as much as possible, even for people who aren’t in any way programmers.

But the tools are very powerful, and for people who know them, or would like to learn them, we need to describe those.  Without making the more-beginner folks think a huge amount of work is required to fix a detail.

Any suggestions (or doc changes! Even better!) in that area are appreciated.