JMRI/JMRI #562: Develop release 4.3.3

bobjacobsen, 2016-01-09:


mattharris, 2016-02-03:
Do we want to target bringing in the RXTX replacement for this? - see #573 for details

rhwood, 2016-02-03:
I would like to see the RXTX replacement in this.

bobjacobsen, 2016-02-14:
The release-4.3.3 branch has been created. 

From now on, please document your changes in the [jmri4.3.4.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.3.4.shtml) release note file.

Maintainers, please set the 4.3.4 milestone on pulls from now on.

Jenkins will be creating files shortly at [the usual place](http://builds.jmri.org/jenkins/job/Test%20Releases/job/4.3.3/).

bobjacobsen, 2016-02-15:
Merged in #805, will be in Jenkins build 2 shortly.

closed 2016-02-16 by bobjacobsen

SteveRawlinson, 2016-02-16:
There was a very small change in 4.3.3 which did not make it into the release notes. I'm not sure how important that is. This is the change:

A sensor attached to a layout block which is not in use can now be reassigned to a different layout block

This was in PR #679.

mattharris, 2016-02-16:
@SteveRawlinson Does [this comment](https://github.com/JMRI/website/commit/ed25e25df3c76ee874d0d33a0d97ed15201adb5d#diff-3019d410da42feb0ef44e7b57ef6cdadR367) not cover that change?

SteveRawlinson, 2016-02-16:
No, that's a different change. That one lets you remove a sensor from a block by selecting the blank line. This one lets you select a sensor that is allocated to another block, so long as that other block is about to be discarded.

mattharris, 2016-02-16:
OK, now problem. Will get this added to the release note

mattharris, 2016-02-16:
Now added - should turn up on website shortly

bobjacobsen, 2016-02-16:
> No, that's a different change. That one lets you remove a sensor from a block by selecting the blank line. This one lets you select a sensor that is allocated to another block, so long as that other block is about to be discarded.

Ah, sorry, my mistake. I got confused about the difference when copying things into the release note. Matt, thanks for fixing this.