JMRI/JMRI #567: Sensor icon positioning

bobjacobsen, 2016-01-10:
From@DanielBoudreau re  testing of 4.3.2, not sure if it appeared before:

Minor issue with sensor display, I use different size icons and they are no longer centered, slight offset to the bottom for the smaller icon.  See attached screen shot, the upper red senor icon is no longer centered around the gray icon.  Looks to me that the gray icon dropped down a bit.

[image]

bobjacobsen, 2016-01-15:
More info from @DanielBoudreau via jmri-developers:

```` xml
Same sensor icon, but they are different in size.  From the panel file:

  <sensoricon sensor="ISPA" x="323" y="17" level="10" forcecontroloff="false" hidden="no" positionable="false" showtooltip="false" editable="true" momentary="true" icon="yes" text="" size="12" style="0" red="0" green="0" blue="0" hasBackground="no" justification="left" class="jmri.jmrit.display.configurexml.SensorIconXml">
     <tooltip>ISPA</tooltip>
     <active url="program:resources/icons/smallschematics/LEDs/green.gif" degrees="0" scale="1.0">
       <rotation>0</rotation>
     </active>
     <inactive url="program:resources/icons/smallschematics/tracksegments/circuit-empty.gif" degrees="0" scale="1.0">
       <rotation>0</rotation>
     </inactive>
     <unknown url="program:resources/icons/smallschematics/tracksegments/circuit-empty.gif" degrees="0" scale="1.0">
       <rotation>0</rotation>
     </unknown>
     <inconsistent url="program:resources/icons/smallschematics/LEDs/yellow.gif" degrees="0" scale="1.0">
       <rotation>0</rotation>
     </inconsistent>
     <iconmaps />
     <activeText red="255" green="0" blue="0" />
     <inactiveText red="255" green="255" blue="0" />
     <unknownText red="0" green="0" blue="255" />
     <inconsistentText red="0" green="0" blue="0" />
   </sensoricon>
````

[image]

````
I've created a simple panel with a flashing sensor icon, and another icon on
top.  I created the panel using 4.2.  When opened using 4.2, the two icons
are centered.  Now using 4.3.2 the icon on top moves slightly lower.  I've
attached the panel xml file. Use NCE simulator.
````
[test.xml.txt](https://github.com/JMRI/JMRI/files/91337/test.xml.txt)

petecressman, 2016-01-23:
I have a fix for this bug.  However i am unable to push any commits to my fork in GitHub.

bobjacobsen, 2016-01-28:
@petecressman's fix was merged with #712

closed 2016-05-02 by bobjacobsen