JMRI/JMRI #573: Replacement of RXTX

mattharris, 2016-01-11:
Currently, we rely on [RXTX](http://rxtx.qbang.org/wiki/index.php/Main_Page) for serial communications.

This has a number of long-outstanding bugs and development seems to have stalled.

As a result, other projects that require to use Java-based serial communications have started to move to alternative libraries.

One such library is [PureJavaComm](http://www.sparetimelabs.com/purejavacomm/purejavacomm.php) which is designed as a drop-in replacement for RXTX.

Another option is [Java Simple Serial Connector](https://github.com/scream3r/java-simple-serial-connector) - for example the Arduino project has switched to using this project, but this is not such a simple drop-in replacement as the API is different and certain items that we utilise are not (immediately) obviously available.

PR #572 contains a proposal of RXTX replacement with PureJavaComm.

rhwood, 2016-01-11:
While this is ongoing, we probably also need to move the code for determining which serial ports to present (at least on Linux) into JMRI instead having the launcher script do it (since it appears that PureJavaComm does not process the ```gnu.io.rxtx.SerialPorts``` system property.

I have been looking into using JSSC, but this seems to be a much cleaner option.

mattharris, 2016-01-11:
I've just updated the PJC library to the recently released 1.0.0 version - PR is updated.

And I agree with @rhwood that we should consider how to process the equivalent of `gnu.io.rxtx.SerialPorts` for *nix systems.

mattharris, 2016-01-11:
Looks like [there is an equivalent](https://github.com/nyholku/purejavacomm/blob/master/src/jtermios/JTermios.java#L527)

For details, [see PureJavaComm wiki](https://github.com/nyholku/purejavacomm/wiki/System-properties-you-can-set)

mattharris, 2016-01-11:
The proposed changes are now in `rxtx-to-pjc` branch of JMRI/JMRI

rhwood, 2016-01-12:
Do we want to replace SerialIO with PureJavaComm as well?

mattharris, 2016-01-12:
SerialIO is there as a special-case fall-back for the 'odd' baud rate needed on the MS100 - see:

https://github.com/JMRI/JMRI/blob/rxtx-to-pjc/java/src/jmri/jmrix/loconet/ms100/MS100Adapter.java

We anyway would revert to PureJavaComm is SerialIO wouldn't work (i.e. when OS is !=Windows  || > Windows XP)

rhwood, 2016-01-12:
Is that special case needed with PureJavaComm, since it was a special case for RXTX?

rhwood, 2016-01-12:
PureJavaComm does not use the properties documented correctly. I'll develop a workaround and submit a PR against that project.

mattharris, 2016-01-12:
The workaround was also needed pre-RXTX

bobjacobsen, 2016-01-12:
SerialIO solved the problem what RXTX (at that point in time) could not set the 16600 baud rate the Digitrax MS100 prefers, nor the 16547 fallback rate, on certain Windows machines.  My notes are packed away, but IIRC the issue was that RXTX's native code used a family of interface calls that used an enum for speed, and the enum didn't have those values. SerialIO used a more-modern family that were introduced in Windows 98 and were "more flexible", so it could do the rates. 

Our our minimal Windows requirement has advanced a lot since then.  PJC may handle the MS100 rates without problem.  That seems to be an experimental question. 

IIRC, the SerialIO version we have licensed doesn't work with some Vista and all Windows 7 machines, so SerialIO may be of decreasing relevance in any case.

I think we should omit SerialIO, and just document what Windows versions can and can't (might be all!) be used the now-with-PJC versions of JMRI.

mattharris, 2016-01-12:
A quick scan through the PureJavaComm code seems to imply that it should be possible to set baud rates to any value - there does not seem to be any `enum` for this.

I'll work up a PR that removes SerialIO and simplifies the MS100 interface implementation.

mattharris, 2016-01-12:
And once we've got something together, either produce a specific test build from the branch, or roll it into the next dev release.

KenC57, 2016-01-12:
Matt,

 

I'd say putting out a call to find somebody using the old MS-100 would be a
good idea. My other memory of this was not all motherboards supported the
speed. So for the bulk of the world, supporting that odd speed would not
matter if the hardware under it couldn't run at that speed.

 

Otherwise it is a fine exercise but likelihood that we will ever know if it
works, I'd call that slim.

 

-Ken Cameron, Member JMRI Dev Team

www.jmri.org

www.fingerlakeslivesteamers.org

www.cnymod.com

www.syracusemodelrr.org

bobjacobsen, 2016-01-12:
> And once we've got something together, either produce a specific test build from the branch, or roll it into the next dev release.

We can create a Jenkins job to make installers available from the branch.

mattharris, 2016-01-13:
#592 contains an update for removal of SerialIO and associated updates to the MS100 adapter.

mattharris, 2016-01-13:
As will be evident, it seems that I missed the direct serial adapter - this has been changed also (PR #592 updated).

Also in there are updates for, I hope, all the various IDE control files that reference the various libraries we use.

rhwood, 2016-01-15:
Can we create the Jenkins job to make installers available from this branch?

mattharris, 2016-01-15:
@rhwood I'll sort that out.

mattharris, 2016-01-15:
@davidzuhn Looks like Jenkins might be down at the moment (or is it just me?)

bobjacobsen, 2016-01-15:
Jenkins is in a funny state.  It's still running, in that it's still generating results emails that look right.  But I can't reach it via the Web.  I've emailed david.

mattharris, 2016-01-15:
Yes, I saw that it's sending emails out - looks to be a routing problem as I get Gateway failures from here.

mattharris, 2016-01-15:
Ah, a bit more digging into DNS records - *.jmri.org is dealt with via vhost.sourceforge.net (216.34.181.97)

mattharris, 2016-01-15:
Probably nothing much david can do, aside from potentially offering an alternate URL

bobjacobsen, 2016-01-15:
Does anybody have an old email? You might be able to get an alternate host name or IP address from the headers. 

(Away from real connection this morning, sorry, so can't check myself)

mattharris, 2016-01-15:
Good call that man!

http://73.65.124.181/ gets us there

mattharris, 2016-01-15:
Right, a Jenkins job has been created and first build kicked-off.

Should be able to find it at:

http://73.65.124.181/jenkins/job/Test%20Releases/job/RXTX%20Replacement%20Alpha%20Test%20Build/

once it's ready.

mattharris, 2016-01-15:
First test builds are available.

OS | Download link
---|---
OS X | [JMRI.4.3.3ish-201601151630-zoo-R61b5a9f.dmg](http://73.65.124.181/jenkins/job/Test%20Releases/job/RXTX%20Replacement%20Alpha%20Test%20Build/lastSuccessfulBuild/artifact/dist/release/JMRI.4.3.3ish-201601151630-zoo-R61b5a9f.dmg)
MS Windows | [JMRI.4.3.3ish-201601151630-zoo-R61b5a9f.exe](http://73.65.124.181/jenkins/job/Test%20Releases/job/RXTX%20Replacement%20Alpha%20Test%20Build/lastSuccessfulBuild/artifact/dist/release/JMRI.4.3.3ish-201601151630-zoo-R61b5a9f.exe)
Linux | [JMRI.4.3.3ish-201601151630-zoo-R61b5a9f.tgz](http://73.65.124.181/jenkins/job/Test%20Releases/job/RXTX%20Replacement%20Alpha%20Test%20Build/lastSuccessfulBuild/artifact/dist/release/JMRI.4.3.3ish-201601151630-zoo-R61b5a9f.tgz)

bobjacobsen, 2016-01-16:
> First test builds are available.

Mac OS X (10.10.5) starts OK, enumerates ports OK, saves preferences and restarts OK.  Can’t try with actual DCC hardware for a couple days, but it does seem to talk and echo OK.

Bob

rhwood, 2016-01-16:
With OS X 10.11.2 + LocoBuffer USB + DCS 100 combination, DecoderPro from this branch:
 - [x] works as expected when starting with systems connected
 - [x] detects missing systems correctly at startup
 - [x] does not exhibit errors outlined in #429

mattharris, 2016-01-19:
I've run some tests on Windows 7 x64 with the following:

- [x] PR3 Standalone
- [x] SPROGvII Serial Standalone Programmer via FTDI USB-RS232 adapter)

In both cases, I read a decoder and ran a JMRI throttle successfully.

I also have the following that will be tested similarly (once I get them setup):

- MERG CBUS Command Station via CAN-USB
- Lenz LZV100 using LI101F via FTDI USB-RS232 adapter

I will also run the tests on Raspberry Pi.

silverailscolo, 2016-01-19:
@mattharris could you please check the Pi install steps in http://jmri.sourceforge.net/install/Raspbian.shtml <http://jmri.sourceforge.net/install/Raspbian.shtml> about RxTx if this is merged?

> Op 19 jan. 2016, om 20:43 heeft Matthew Harris <notifications@github.com> het volgende geschreven:
> I've run some tests on Windows 7 x64 with the following:
> 
> [snip]
> I will also run the tests on Raspberry Pi.

bobjacobsen, 2016-01-24:
Got a chance to try this with hardware.  Seems to work, even in MS100 mode.

mattharris, 2016-01-25:
Have we had any one able to replicate the problems seen by Walt - reported in ['jmri-developers' thread](http://sourceforge.net/p/jmri/mailman/jmri-developers/thread/CANp54oWpAFQ%2BU2RV0KDS913Jw5bQO%3D0iHRLDwnBjF5xfA_mmiw%40mail.gmail.com/#msg34771805)?

From the description, seems as though he was trying to just turn on and off track power using a LocoBuffer on a Digitrax system connected to a Windows box.

rhwood, 2016-01-25:
Ran a test on Fedora 23 (Intel x64) today:
- [x] LocoBuffer-USB to DCS 100 works (tested by setting track power on and off)
- [ ] connection appears red in Preferences
- [ ] wrong TTY devices are listed in the connections combo box
 
So although this works, it does display undesirable behaviors in #429.

SteveRawlinson, 2016-01-25:
I'm happy to test this on windows/loconet, should I pull from JMRI/JMRI:rxtx-to-pjc or MattHarris/JMRI:rxtx-to-pjc?

mattharris, 2016-01-25:
@SteveRawlinson Pull from `JMRI/JMRI:rxtx-to-pjc`

SteveRawlinson, 2016-01-25:
I'm seeing a problem, not sure if it's exactly the same as Walt. The power button on a throttle pane is yellow and the tooltip says "Layout Power state unknown". Pressing the power button has no effect and Loconet monitor shows no activity. The log says it's connecting to the locobuffer OK.

What would you like me to do?

rhwood, 2016-01-25:
What is the OS and Java version and are they both 64 bit?

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

mattharris, 2016-01-25:
Are you able to read a decoder on the programming track?

SteveRawlinson, 2016-01-25:
I'm running Windows 7 Home Premium 64 bit. Java is version 8 Update 71 64bit.

I'm actually access the PC using Teamview from work so I can't put anything on the programming track to test it. I'll do that when I get home in about 2 hours.

SteveRawlinson, 2016-01-25:
Actually, just to see what happened I loaded up a programmer and got it to read a sheet even though there's nothing on the programming track.  Error 306. Nothing in the Loconet monitor window. So I'm guessing the answer is no, I can't read a decoder. I'll confirm that later.

rhwood, 2016-01-25:
Can you test this with a 32-bit Java? That may give us what we need to engage the PJC developer on this (looking at the project, it appears to only have 32-bit Windows native code, but I may be reading it incorrectly).

SteveRawlinson, 2016-01-25:
OK, bear in mind I'm not actually a windows user and I've never user used Java.

I've tried installing Java again and it does not give me any options regarding 32/64 bit. I have the following installed:

Java 8 Update 71
Java 8 Update 71 (64 bit)
Java SE Development Kit 8 update 65 (64 bit)

So maybe the top option is 32 bit? I'm running this from within Netbeans, if there's a way to instruct Netbeans to use a different java, or something I can change in build.xml I can give it a go.

mattharris, 2016-01-25:
Steve, once running, bring up the Context (Help > Context) from within JMRI - that will provide us with the info.

Copy/Paste the details from that report here and we'll decode it for you.

SteveRawlinson, 2016-01-25:
Here you go:

[context.txt](https://github.com/JMRI/JMRI/files/103735/context.txt)

mattharris, 2016-01-25:
@SteveRawlinson So, the following lines give us what we need:

```
java.version: 1.8.0_65   
java.vendor: Oracle Corporation   
java.home: C:\Program Files\Java\jdk1.8.0_65\jre   
java.vm.version: 25.65-b01   
java.vm.vendor: Oracle Corporation   
java.vm.name: Java HotSpot(TM) 64-Bit Server VM   
```
and
```
os.name: Windows 7   
os.arch: amd64   
os.version: 6.1   
```
So, from this we can tell that you're running a 64-bit JVM on Windows.

(The `os.arch` setting and `java.vm.name` will differ if running in 32-bit mode)

SteveRawlinson, 2016-01-25:
Yes, there was no doubt I was running 64 bit but @rhwood asked me to run it under a 32 bit Java. The question is, how do I do that?

mattharris, 2016-01-25:
Ah, sorry - serves me right for reading and replying on a phone...

mattharris, 2016-01-25:
Easiest way to do this is to install JMRI proper (use one of the 4.3.3ish versions referenced earlier in this thread) and then use the `/32bit` on the shortcut to force the 32-bit JVM as detailed in the [startup scripts documentation](http://jmri.org/help/en/html/doc/Technical/StartUpScripts.shtml#windows)

SteveRawlinson, 2016-01-25:
OK, so context now says 

java.vm.name: Java HotSpot(TM) Client VM
os-arch: x86

But the results are the same: the power button is yellow, Layout Power state unknown, nothing in the Loconot monitor.

SteveRawlinson, 2016-01-26:
Now that I've actually sat at the layout I ca give you a bit of an update.

The power button is yellow, and clicking it does actually turn track power off. However, subsequent presses do not bring it back and it remains yellow. Nothing at all appears in the loconot monitor despite the fact that I can see my watchmans blinking so I know there's traffic.

I cannot make contact with the programming track at all. When I try to operate a turnout it is definitely getting some sort of message but never actually moves. The motors sounds like they getting a command to be in the state it's already in. 

Changing a signal works. My signals work on turnout commands so how that's happening I have no idea.

It does not seem to make any difference whether I run the 32 bit mode or 64 bit.

mattharris, 2016-01-26:
@SteveRawlinson Is there anything coming up in the LocoNet monitor at all?

Now, I _think_ that you need to open it first before it shows stuff - I don't think that it populates 'in the background'

bobjacobsen, 2016-01-26:
> On Jan 26, 2016, at 7:02 AM, Steve Rawlinson <notifications@github.com> wrote:
> 
> The power button is yellow, and clicking it does actually turn track power off. However, subsequent presses do not bring it back and it remains yellow. Nothing at all appears in the loconot monitor despite the fact that I can see my watchmans blinking so I know there's traffic.
> 
> I cannot make contact with the programming track at all. When I try to operate a turnout it is definitely getting some sort of message but never actually moves. The motors sounds like they getting a command to be in the state it's already in. 
> 
> Changing a signal works. My signals work on turnout commands so how that's happening I have no idea.

This sounds like the messages are going out to the LocoNet, but are not being echoed back.

1) Without the echo, there’s nothing for the monitor to show.  For LocoNet, the monitor displays what’s actually on the cable by listening for messages back from the interface.

2) Some commands (power, turnouts) don’t require any interaction with the command station.  They’ll work with uni-directional communication.  But others (programming, throttle) do require back and forth, so will fail without it.

Could you remind us what interface hardware you’re using (PR2, PR3, LocoBuffer (which one), etc?)

Also, open the JMRI log, and look for a message about opening the port.  IIRC, at least with the old code, that listed the control leads (DCD etc).  Perhaps there’s something wrong in those that’s preventing communication back.  That’s an idiosyncratic part of the “RS232” interface that the new library might be doing differently.

SteveRawlinson, 2016-01-26:
There is nothing at all on the loconet monitor at any point, no matter what I do.

I am using a LocoBuffer USB Rev-n bought in September, I suspect it's still the newest one.

The log line connecting to the com port is:

2016-01-26 15:45:09,635 locobuffer.LocoBufferAdapter          INFO  - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true  CD: false [main]

This line is absolutely identical to the line written by a fully working instance of JMRI.

In case it's important - The broken JMRI cannot set the state of control sensors - that is, sensors that are not connected to any feedback hardware but which I to click on to set routes. Scripts cannot set the state of those sensors either.

Clicking on a signal changes its state fine (and the relevant signal on the layout changes).

mattharris, 2016-01-26:
@SteveRawlinson could you turn on the debug level logging for the jmri.jmrix hierarchy?

In `default.lcf` change:

`# log4j.category.jmri.jmrix=DEBUG`

to

`log4j.category.jmri.jmrix=DEBUG`

(It's near the bottom of the file)

`default.lcf` is found in the root of the main JMRI program directory, or root of the checkout if running direct from NetBeans.

bobjacobsen, 2016-01-26:
> 2016-01-26 15:45:09,635 locobuffer.LocoBufferAdapter INFO - COM3 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: true CD: false [main]
> 
> This line is absolutely identical to the line written by a fully working instance of JMRI.

Right, but the software might be treating those differently now that there’s a different library.  It might be requiring the DSR (DataSet, i.e. device, Ready) CD (Carrier Detect) line to be true before listening to the data line.  

One can argue that the RS232 standard requires that, but in practice that requirement is usually unimplemented so that it’s possible to exchange characters with just the data lines.

> In case it's important - The broken JMRI cannot set the state of control sensors - that is, sensors that are not connected to any feedback hardware but which I to click on to set routes. Scripts cannot set the state of those sensors either.

That’s consistent with the hypothesis that the messages are going out, but not coming back. JMRI doesn’t know about the hardware, but does expect the sensor-change message to go out and come back.

SteveRawlinson, 2016-01-26:
Turning on debugging is is no problem. I assume you'd like to see the logs - what actions would you like me to perform so you can see the output?

mattharris, 2016-01-26:
Please just try the same interactions as before such as:

- connect
- power on/off
- turnout on/off
- attempt to read from programming track

Thanks for you help.

SteveRawlinson, 2016-01-26:
Here's the session log. I had to change the name to .txt to upload to github. 

I'm convinced that you are right - messages are getting out but nothing's getting back.

The reason turnouts were not working was because the layout power was off after I tried operating the power button which turns it off but not back on again (d'oh). Quite possibly it's trying to turn it off again. This time I turned layout power back on via the command station before doing the other tests. Turnouts move but only once and their state does not get updated. Clicking again does nothing and there's still nothing in the loconet monitor but I do see flickering on the watchmans so I think it's pretty obvious that it's setting the turnout the same way each time because it's not seeing it's state change.

Finally, reading from the decoder track looks like it's working - the numbers on the DCS50 flicker exactly like they normally do during CV reading. But there's a 306.

[session.txt](https://github.com/JMRI/JMRI/files/105274/session.txt)

mattharris, 2016-01-26:
Thanks Steve. Could you try again with debugging on for jmri.jmrix?

From the looks of it, updating `default.lcf` in `C:\Program Files (x86)\JMRI` should do it.

Thanks.

[43 more replies]

closed 2017-06-22 by rhwood