JMRI/JMRI #597: Broken Danish translation files

bobjacobsen, 2016-01-13:
@SonnysHansen mentioned on jmriusers:
````
When mr. jacobsen convert a lot off properties files in JMRI, the Danish special letters æ,ø and å has been corrupt. 
I think it's because some of them have not undergone a native2ascii converter. 
How can I fix this??? 
````

If there's an older version of the files that still have the right special characters, we can revert the _da.properties files to then, run them through native2ascii, change their line-ends, and recommit.

mattharris, 2016-01-13:
#600 should contain a fix for these

bobjacobsen, 2016-01-14:
See also #601 which is working toward adding a target to catch the next round.

mattharris, 2016-01-14:
#601 has now been merged. Just waiting for Travis to do it's thing - all being well, it should pass

mattharris, 2016-01-14:
Well, the Travis failure was _not_ due to the new `checknonascii` ant target, rather an error in `ci-tests`.

I've re-launched the Travis build to see if it goes through this time.

mattharris, 2016-01-14:
Yay! It passed the second time round.

mattharris, 2016-01-15:
One thing I've found through the various i18n issues is that using NetBeans as an editor of .properties files is _really_ good.

It's possible to just type directly in your chosen language and NetBeans itself deals with the native2ascii conversion _automatically_!!

SonnysHansen, 2016-01-15:
Sound like a good idea, I´ll try that later this week, but for now I 
just use the Native2ascii translator.

Den 15-01-2016 kl. 19:29 skrev Matthew Harris:

mattharris, 2016-01-15:
After several iterations, seem to now have the `checknonascii` ant target working for Travis - see #621 for an example

bobjacobsen, 2016-01-18:
The check is working, in that files are both passing and failing, so I'm closing this.

closed 2016-01-18 by bobjacobsen

mattharris, 2016-01-18:
@bobjacobsen I concur.