JMRI/JMRI #658: AppVeyor CI control issues

bobjacobsen, 2016-01-18:
Not all people with JMRI merge access can ask for an AppVeyor CI job to be restarted.

Since the AppVeyor CI jobs are (still) occasionally giving false negatives, and since passing that CI is an important quality check before something is merged, we want to be able to easily rerun them.

The discussion originally started in PR #637, see there for background.

mattharris, 2016-01-18:
[This AppVeyor help page](https://www.appveyor.com/docs/team-setup#github-teams) might be useful.

Seems to suggest that GitHub team integration, which I think is what we're after, is only possible when logged-in via GitHib credentials.

It also suggests that creating specific GitHub teams for AppVeyor authorisation checks might help.

mattharris, 2016-01-18:
Additionally, it talks about [setting up collaborators](https://www.appveyor.com/docs/team-setup#collaborator) - is that a step we're missing?

rhwood, 2016-01-18:
I don't think so, collaborators require per user roles, and GitHub teams should be providing per-group roles.

bobjacobsen, 2016-01-18:
Reading the AppVeyor [AppVeyor help page](https://www.appveyor.com/docs/team-setup#github-teams) is seems that that there needs to be a central AppVeyor account for our JMRI work, instead of it automatically picking up our GitHub accounts:
````
User is authenticating entity or login which is uniquely identified by email address. User is created under certain account and belongs to that account. User can have only one role assigned.

Once certain email is added as a user into AppVeyor account that user cannot have their own account. If you need to move a user to his/her own account delete this user from your account, allow them to register their own AppVeyor account and then add that email back as collaborator.
````

So when I created my own AppVeyor account, I probably disconnected from the JMRI GitHub-based one.

Who has access to "add that email back as a collaborator"? @rhwood, I think you set this up, do you have access to some kind of AppVeyor account screen? My individual AppVeyor account doesn't seem to have anything like that.

rhwood, 2016-01-19:
I think this is now fixed. There are two AppVeyor applications that needed to be associated with this Organization, but one was denied access.

mattharris, 2016-01-19:
I now appear to have the correct access in AppVeyor.

It was necessary to log-out of AppVeyor and then log-in using GitHub. Upon doing this, I was then prompted to choose the account to use as my email address seems to be associated with more than one account - choosing 'JMRI' then allows expected access.

bobjacobsen, 2016-01-19:
> @mattharris: It was necessary to log-out of AppVeyor and then log-in using GitHub. Upon doing this, I was then prompted to choose the account to use as my email address seems to be associated with more than one account - choosing 'JMRI' then allows expected access.

Thanks!  That was exactly the hint.  I was signed on, but not connected to the “JMRI” account, so I couldn’t access controls for JMRI builds.

And thanks to @rhwood  for setting this up. The combination of Travis, AppVeyor and Jenkins has helped me (hopefully) track down and kill a nasty threading bug in the scripting startup.

Bob

mattharris, 2016-01-19:
Happy to help Bob! Had me stumped for a while also.

And, I second the thanks to @rhwood.

The various CI checks we now have in-place bring a _huge_ benefit to the overall quality of the JMRI codebase (in my opinion anyway).

closed 2016-01-20 by bobjacobsen