JMRI/JMRI #727: CI failures due to threading in Warrant code

bobjacobsen, 2016-01-28:
There are occasional CI failures (usually Travis, but alsoAppVeyor) that look like:
````
Testcase: testLearnWarrant(jmri.jmrit.logix.LearnWarrantTest):	FAILED
    [junit] Level mismatch when looking for WARN message: "RosterSpeedProfile not found. Using default ThrottleFactor 0.75" found "Uncaught Exception: java.lang.ArrayIndexOutOfBoundsException: No such child: 24
    [junit] 	at java.awt.Container.getComponent(Container.java:332)
````
or sometimes "Comparator contract not being honored" at a similar point.

These appear to be due to the Warrant code manipulating the Swing GUI from a non-Swing/AWT thread.  

I'm not 100% sure of how that happens.  The Warrant code starts its own threads to do its work. The problem might be true Swing operations from those Warrant thread(s). But it might also be due to the Warrant thread(s) doing throttle and bean operations that on their own drive other parts of the GUI in ways inconsistent with what Warrant is doing.

In any case, I'd like to keep running those tests during CI and accumulating the results. If you happen to see a different failure message in this area of the tests, please cut and past the error message and top couple lines of the traceback as a comment.  (And if you see _any_ JMRI code in the traceback, please include it too)

Thanks

rhwood, 2016-01-30:
https://travis-ci.org/JMRI/JMRI/builds/105896715#L200 contains stack trace that appears to be cut off; here are the first few lines:
```
[junit] Level mismatch when looking for WARN message: "RosterSpeedProfile not found. Using default ThrottleFactor 0.75" found "Uncaught Exception: java.lang.IllegalArgumentException: Comparison method violates its general contract!
    [junit] 	at java.util.TimSort.mergeHi(TimSort.java:895)
    [junit] 	at java.util.TimSort.mergeAt(TimSort.java:512)
    [junit] 	at java.util.TimSort.mergeForceCollapse(TimSort.java:453)
    [junit] 	at java.util.TimSort.sort(TimSort.java:250)
    [junit] 	at java.util.Arrays.sort(Arrays.java:1512)
    [junit] 	at java.util.ArrayList.sort(ArrayList.java:1454)
    [junit] 	at java.util.Collections.sort(Collections.java:175)
```

rhwood, 2016-03-25:
https://ci.appveyor.com/project/JMRI/jmri/build/1770#L3048 is an error message thrown by the Warrant tests.

bobjacobsen, 2016-03-26:
PR #1035 should be a fix for this. I would appreciate reports if you see any more of these exceptions after that PR is merged.

rhwood, 2016-03-26:
[Build failure](https://travis-ci.org/JMRI/JMRI/builds/118651129)
```
ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [testNXWarrant] jmri.util.exceptionhandler.UncaughtExceptionHandler.?()
```
after merging #1039.

rhwood, 2016-03-26:
https://ci.appveyor.com/project/JMRI/jmri/build/1802#L3048 contains an unanticipated warrant error:
```
ERROR - checkCurrentBlock, block "OB2" not occupied! warrant NX(666(L)) [testNXWarrant] jmri.jmrit.logix.Warrant.?()
```

bobjacobsen, 2016-03-27:
Thanks. I think #1035 has a fix for the 'block "OB2" not occupied!' error. That one was (probably) more of a problem in the test threading than the underlying code, so I'll look around for more of the same pattern.

bobjacobsen, 2016-03-28:
Not exactly in the Warrant code, but close enough:
````
    [junit] junit.framework.AssertionFailedError: "mast IF$vsm:UP-2008:SL-3A($0170) aspect "Advance Approach" expected, currently showing "Stop"" did not occur in time
    [junit] 	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:129)
    [junit] 	at jmri.jmrit.display.SignalSystemTest.checkAspect(SignalSystemTest.java:148)
    [junit] 	at jmri.jmrit.display.SignalSystemTest.testLoadAA1UPtest(SignalSystemTest.java:103)
    [junit] 
    [junit] 
    [junit] Test apps.tests.AllTest FAILED
````

bobjacobsen, 2016-05-02:
#1035 did fix one problem, but that exposed a lower-rate one:
````
  [junit] WARN  - Engineer waiting at Block OB3 [testNXWarrant] jmri.jmrit.logix.Warrant.?()
    [junit] ERROR - checkCurrentBlock, block "OB2" not occupied! warrant NX(666(L)) [testNXWarrant] jmri.jmrit.logix.Warrant.?()
````

bobjacobsen, 2016-05-02:
Another error, perhaps related to Warrant code, but likely not:
````
    [junit] Testcase: testLoadAA1UPtest(jmri.jmrit.display.SignalSystemTest):	FAILED
    [junit] "mast IF$vsm:UP-2008:SL-3A($0170) aspect "Advance Approach" expected, currently showing "Stop"" did not occur in time
    [junit] junit.framework.AssertionFailedError: "mast IF$vsm:UP-2008:SL-3A($0170) aspect "Advance Approach" expected, currently showing "Stop"" did not occur in time
    [junit] 	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:126)
    [junit] 	at jmri.jmrit.display.SignalSystemTest.checkAspect(SignalSystemTest.java:148)
    [junit] 	at jmri.jmrit.display.SignalSystemTest.testLoadAA1UPtest(SignalSystemTest.java:103)
````

bobjacobsen, 2016-05-05:
Another example:
````
WARN  - Engineer waiting at Block OB3 [testNXWarrant] jmri.jmrit.logix.Warrant.?()
    [junit] ------------- ---------------- ---------------
    [junit] Testcase: testLoadAA1UPtest(jmri.jmrit.display.SignalSystemTest):	FAILED
    [junit] "mast IF$vsm:UP-2008:SL-3A($0170) aspect "Advance Approach" expected, currently showing "Stop"" did not occur in time
    [junit] junit.framework.AssertionFailedError: "mast IF$vsm:UP-2008:SL-3A($0170) aspect "Advance Approach" expected, currently showing "Stop"" did not occur in time
    [junit] 	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:127)
    [junit] 	at jmri.jmrit.display.SignalSystemTest.checkAspect(SignalSystemTest.java:148)
    [junit] 	at jmri.jmrit.display.SignalSystemTest.testLoadAA1UPtest(SignalSystemTest.java:103)
    [junit] 
````

pabender, 2016-05-06:
Findbugs points to several synchronization related errors in this code.

I'm going to start running some of those down to see if they help with these intermittent errors.

Paul

bobjacobsen, 2016-05-07:
#1287 has a possible fix for the ````testLoadAA1UPtest(jmri.jmrit.display.SignalSystemTest)```` failure; it may be due to some initialization operations taking place on the wrong thread.

bobjacobsen, 2016-05-30:
Looks like these are fixed.  If any new ones arise, please start a new issue.

closed 2016-05-30 by bobjacobsen