JMRI/JMRI #759: Problem dropping configured connections

bobjacobsen, 2016-02-07:
I haven't been able to reliably recreate it, but the current (Feb 6) HEAD of master has a problem with dropping / deleting previously configured connections.

In the UI, it shows up as the "Save" button in the Configuration window not displaying the usual restart dialog box, apparently because there's been a NPE during the store.

It's hard to track down, though, because the problem seems to result in configuration data spread across several files.  More to follow.

Has anybody seen this?

bobjacobsen, 2016-02-07:
I haven't been able to recreate this reliably.

It seems to have something to do with (settings)/(profile)/ProfileConfig.xml and (settings)/(profile)/profile/profile.xml getting out of synch, but I don't know whether that's cause or effect.

rhwood, 2016-02-07:
The configuration data is stored as described in https://github.com/JMRI/JMRI/issues/69#issuecomment-155385101. We continue to write ProfileConfig.xml in an attempt to provide a backwards compatible Profile configuration for earlier versions of JMRI, but it should not be read if ```profile:profile/profile.xml``` exists.

If you can post an NPE stack trace, that would be helpful.

bobjacobsen, 2016-02-07:
````
2016-02-06 14:21:07,593 managers.ManagerDefaultSelector       WARN  - Connection ECoS has been removed, we shall remove it as the default for interface jmri.ThrottleManager [AWT-EventQueue-0]
2016-02-06 14:21:13,122 managers.ManagerDefaultSelector       WARN  - Connection LocoNet has been removed, we shall remove it as the default for interface jmri.PowerManager [AWT-EventQueue-0]
2016-02-06 14:21:13,123 managers.ManagerDefaultSelector       WARN  - Connection LocoNet has been removed, we shall remove it as the default for interface jmri.GlobalProgrammerManager [AWT-EventQueue-0]
2016-02-06 14:21:13,123 managers.ManagerDefaultSelector       WARN  - Connection LocoNet has been removed, we shall remove it as the default for interface jmri.ThrottleManager [AWT-EventQueue-0]
2016-02-06 14:21:19,351 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
        at jmri.jmrix.loconet.hexfile.configurexml.ConnectionConfigXml.store(ConnectionConfigXml.java:40)
        at jmri.jmrix.configurexml.AbstractConnectionConfigXml.store(AbstractConnectionConfigXml.java:33)
        at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:515)
        at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:495)
        at jmri.configurexml.ConfigXmlManager.addPrefsStore(ConfigXmlManager.java:254)
        at jmri.configurexml.ConfigXmlManager.storePrefs(ConfigXmlManager.java:407)
        at jmri.configurexml.ConfigXmlManager.storePrefs(ConfigXmlManager.java:401)
        at jmri.implementation.JmriConfigurationManager.storePrefs(JmriConfigurationManager.java:128)
        at jmri.jmrix.ecos.swing.preferences.PreferencesPane.updateButtonPressed(PreferencesPane.java:444)
        at jmri.jmrix.ecos.swing.preferences.PreferencesPane.savePreferences(PreferencesPane.java:557)
        at apps.gui3.TabbedPreferences.invokeSaveOptions(TabbedPreferences.java:230)
        at apps.gui3.TabbedPreferences.lambda$init$35(TabbedPreferences.java:147)
        at apps.gui3.TabbedPreferences$$Lambda$45/1143042988.actionPerformed(Unknown Source)
        at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
````

rhwood, 2016-02-07:
Can you provide the complete log and does this only happen with LocoNet devices?

pabender, 2016-02-08:
Sent from my iPad

On Feb 7, 2016, at 6:17 PM, Randall Wood <notifications@github.com> wrote:
> Can you provide the complete log and does this only happen with LocoNet devices?
> 

I saw something similar last week on an XBee connection, but it was at about the same time I attempted ( and failed ) to use that connection with the pure java serial code, so I attributed it to that bit of broken code....

Paul

rhwood, 2016-02-08:
When these failures happen, does profile:profile/profile.xml appear correct afterwards, or is it also missing the connections?

bobjacobsen, 2016-02-08:
Sorry, I don't have any more information on the problem above.

I just hit another one when converting a LocoNet connection to an ESU connection.  Clicking save caused the attached log.  Top of the exception trace:
````
     [java] 2016-02-07 22:26:33,960 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
     [java] 	at jmri.jmrix.loconet.hexfile.configurexml.ConnectionConfigXml.store(ConnectionConfigXml.java:40)
     [java] 	at jmri.jmrix.configurexml.AbstractConnectionConfigXml.store(AbstractConnectionConfigXml.java:33)
     [java] 	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:515)
     [java] 	at jmri.configurexml.ConfigXmlManager.elementFromObject(ConfigXmlManager.java:495)
     [java] 	at jmri.configurexml.ConfigXmlManager.addPrefsStore(ConfigXmlManager.java:254)
     [java] 	at jmri.configurexml.ConfigXmlManager.storePrefs(ConfigXmlManager.java:407)
     [java] 	at jmri.configurexml.ConfigXmlManager.storePrefs(ConfigXmlManager.java:401)
     [java] 	at jmri.implementation.JmriConfigurationManager.storePrefs(JmriConfigurationManager.java:128)
     [java] 	at jmri.jmrix.ecos.swing.preferences.PreferencesPane.updateButtonPressed(PreferencesPane.java:444)
     [java] 	at jmri.jmrix.ecos.swing.preferences.PreferencesPane.savePreferences(PreferencesPane.java:557)
     [java] 	at apps.gui3.TabbedPreferences.invokeSaveOptions(TabbedPreferences.java:230)
````
It looks pretty similar to the last one. ProfileConfig.xml and profile.xml (attached) are now inconsistent. Their timestamps indicate that profile.xml was written, but ProfileConfig.xml was not.

[profile.xml.txt](https://github.com/JMRI/JMRI/files/121194/profile.xml.txt)
[log.txt](https://github.com/JMRI/JMRI/files/121196/log.txt)
[ProfileConfig.xml.txt](https://github.com/JMRI/JMRI/files/121195/ProfileConfig.xml.txt)

rhwood, 2016-05-01:
JMRI do not attempt to write ProfileConfig.xml anymore (Startup Items preferences are not compatible with it), so this is no longer an issue.

closed 2016-05-01 by rhwood