JMRI/JMRI #760: Should NamedBeans be more JavaBean-like?

bobjacobsen, 2016-02-07:
(See the discussion in #753)

@rhwood [posted a while back](http://sourceforge.net/p/jmri/mailman/message/28456975/):
````
In NamedBean, the key parameter for getProperty and setProperty is an Object instead of a String. Does anyone recall why? Is there a reason not to change property keys to be Strings?

I'm asking because the JavaBeans introspection and property manipulation methods all use Strings for property names, and wonder if the NamedBean should match the JavaBeans design patterns.
````

Should we change this?  The  [bobj-bean-arg branch](https://github.com/JMRI/JMRI/tree/bobj-bean-arg) contains a first cut at what it would take (checks are done in the branch):

- [x] Change the NamedBean interface
- [x] Change the AbstractNamedBean implementation, along with ones in CatalogTreeFS and CatalogTreeIndex (which should be refactored, but that's another job)
- [x] Remove the code that attempts to create an Object key when reading a ConfigXML file (in AbstractNamedBeanManagerConfigXML) and a signal definition file (in DefaultSignalSystemManager). I searched for any uses of a key other than String type, and didn't find any.  It's possible that some user code has one, though, and that'll be a future problem.

Those changes are in the branch. It runs the JUnit tests OK.

In addition to fixing that, there's a bit of work still to be done:

- [ ] change the ConfigureXML code to not emit that attribute
- [ ] migrate the schema to a new one that doesn't have a 'class' attribute on the key element

Should we continue on this?

SteveRawlinson, 2016-02-07:
Speaking only for myself a clarification in the documentation that JMRI NamedBeans are not related to JavaBeans would be sufficient.

rhwood, 2016-02-07:
I would prefer that these classes become JavaBean compliant. This would remove a barrier to potentially using third-party JavaBean-capable systems within a JMRI application to manipulate layout elements.

bobjacobsen, 2016-03-20:
#778 implements this now.

closed 2016-03-20 by bobjacobsen