JMRI/JMRI #763: Signal Mast Logic not reacting on block occupancy

kjlisby, 2016-02-07:
Starting with this pull request  https://github.com/JMRI/JMRI/pull/682, signal mast logic no longer reacts on block occupancy changes.

bobjacobsen, 2016-02-07:
Could you provide a panel file that demonstrates this?  It seems to be working for the test files.

kjlisby, 2016-02-08:
I am using this panel file: https://xa.yimg.com/df/jmriusers/ECOS_OBlock_ControlPanel.xml?token=sNsQvJ4pH6mRCjzv_NiLZsoo-yI3rVLvOcDa0oIdTUZC5eHKbaX_2h4TR0OS2wSKQ_uPhE5CTc0tYphXG7eIWIE9DtpC95AJQtIyJeM5lZ2XV_UQJ6wszDm7B3Q-oh6vUYjBT3aNCJX95Q&type=download

BR
Lisby

bobjacobsen, 2016-02-09:
@kjlisby sorry, but I haven't been able to get that to work well enough to fail. Without an ECos simulator, I tried to adapt it to another connection type without success. 

Could you please try to create a simple panel file that shows the problem?  Ideally with internal devices (the internal sensor problem should be fixed now, see #764), but LocoNet simulator or another simulator connection OK too.

kjlisby, 2016-02-10:
@bobjacobsen 

I made this file: https://xa.yimg.com/df/jmriusers/Lisby_Layout_Sim.xml?token=SM-nHGCVR0Q2to34j9V1fBQQQfyQBjWIbGB7w_xS2D7LTDyBlnMBv0ahOrQVXJkCHdVc4toDSNANHxiVS9J-UV_svF7zRRZZZnuQkBJby3_fMhwoQOoHCMIO8akpP2Hu_aY&type=download

I did so on a clean Windows 7 machine. I could not get it to load my Danish Signals until I consistently named them "DanishSimplified" in the signal definition XML files (for which I just submitted a pull request) and in the layout file itself.

The signal mast logic seems to be entirely dead when I load that layout file - even more dead than when I run with the ECOS, i.e. in the simulated setup, there is not even any reaction to turnout settings.

Not really sure what tells you, but I am seeing the same exceptions as I reported earlier when I activate/deactivate sensors via the sensor table window. My favourite test case is to simulate a train running back and forth between B14 and B15. That is what I did in the physical layout.

And just to repeat the exceptions:
2016-02-10 14:18:34,713 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast.checkStateDetails(DefaultSignalMastLogic.java:1948)
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast$1.run(DefaultSignalMastLogic.java:1841)
	at java.lang.Thread.run(Thread.java:745)
 [B10-B09 Bom 23 Mast]
2016-02-10 14:18:34,714 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast.checkStateDetails(DefaultSignalMastLogic.java:1948)
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast$1.run(DefaultSignalMastLogic.java:1841)
	at java.lang.Thread.run(Thread.java:745)
 [B10-B09 B08-B24]
2016-02-10 14:18:34,714 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast.checkStateDetails(DefaultSignalMastLogic.java:1948)
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast$1.run(DefaultSignalMastLogic.java:1841)
	at java.lang.Thread.run(Thread.java:745)
 [B08-B09 B10-B11]
2016-02-10 14:18:34,713 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast.checkStateDetails(DefaultSignalMastLogic.java:1948)
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast$1.run(DefaultSignalMastLogic.java:1841)
	at java.lang.Thread.run(Thread.java:745)
 [B03-B09 B10-B11]
2016-02-10 14:18:34,713 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception: java.lang.NullPointerException
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast.checkStateDetails(DefaultSignalMastLogic.java:1948)
	at jmri.implementation.DefaultSignalMastLogic$DestinationMast$1.run(DefaultSignalMastLogic.java:1841)
	at java.lang.Thread.run(Thread.java:745)
 [B10-B09 B03-B24]

bobjacobsen, 2016-02-11:
Still working on this. Its more subtle than just "occupancy not working", because occupancy does sometimes work (for other files) with this code.  Current hypothesis, not confirmed, is that something goes wrong during the reading of the file that (silently) breaks the Block connections.

kjlisby, 2016-02-14:
While testing the internal sensor and internal turnout issue, I compared my original layout XML to the one produced by master / HEAD from Github. And there are a number of differences in signal logic. Perhaps these differences can explain something about this signalling issue?
I have uploaded both the original file "ECOS_OBlock_ControlPanel.xml" and the file produced by master / HEAD (with a "2" added to its filename) to ProblemsBeingWorkedOn/Lisbys files on the Yahoo filestore.

A file compare should reveal the differences in signal logic.

BR
Lisby

PR #853 (2016-02-21): Improvements to equals, hashCode operators
Fix problems with equals, hashCode operators in Block and Path classes.  These were using mutable values as part of the comparison, which meant that they weren't living up to the "hashCode is immutable" contract.

Closes #763
closed 2016-02-21 by bobjacobsen