JMRI/JMRI #764: Internal sensors and turnout not stored in XML files

kjlisby, 2016-02-07:
Staing with this pull request: https://github.com/JMRI/JMRI/pull/733, JMRI is no longer able to store internal sensors and turnuts in XML files.

bobjacobsen, 2016-02-07:
The "InternalSensorManager" is no longer being registered for persistence. (This is also true of several other internal managers)

In v4.3.2, this happened like this:
````
     [java] 2016-02-07 15:02:13,388 managers.AbstractManager              DEBUG - register for config of type class jmri.managers.InternalSensorManager [main]
     [java] java.lang.Exception: traceback
     [java] 	at jmri.managers.AbstractManager.registerSelf(AbstractManager.java:43)
     [java] 	at jmri.managers.AbstractManager.<init>(AbstractManager.java:27)
     [java] 	at jmri.managers.AbstractSensorManager.<init>(AbstractSensorManager.java:18)
     [java] 	at jmri.managers.InternalSensorManager.<init>(InternalSensorManager.java:13)
     [java] 	at jmri.managers.ProxySensorManager.makeInternalManager(ProxySensorManager.java:24)
     [java] 	at jmri.managers.AbstractProxyManager.getInternal(AbstractProxyManager.java:75)
     [java] 	at jmri.managers.AbstractProxyManager.getMgr(AbstractProxyManager.java:43)
     [java] 	at jmri.managers.AbstractProxyManager.addVetoableChangeListener(AbstractProxyManager.java:274)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutBlockManager.<init>(LayoutBlockManager.java:39)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
     [java] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
     [java] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
     [java] 	at jmri.InstanceManager.getDefault(InstanceManager.java:141)
     [java] 	at jmri.managers.DefaultSignalMastLogicManager.<init>(DefaultSignalMastLogicManager.java:45)
     [java] 	at jmri.managers.DefaultInstanceInitializer.getDefault(DefaultInstanceInitializer.java:132)
     [java] 	at jmri.InstanceManager.getDefault(InstanceManager.java:151)
     [java] 	at jmri.InstanceManager.signalMastLogicManagerInstance(InstanceManager.java:452)
     [java] 	at jmri.jmrit.signalling.EntryExitPairs.<init>(EntryExitPairs.java:664)
     [java] 	at apps.Apps.<init>(Apps.java:234)
     [java] 	at apps.PanelPro.PanelPro.<init>(PanelPro.java:48)
     [java] 	at apps.PanelPro.PanelPro.main(PanelPro.java:110)
````

This doesn't appear if running at current HEAD of master.  There is an internal sensor created at HEAD:
````
     [java] 2016-02-07 15:10:33,520 managers.AbstractSensorManager        DEBUG - newSensor:ISCLOCKRUNNING;null [main]
     [java] 2016-02-07 15:10:33,522 managers.AbstractSensorManager        DEBUG - Internal Sensor "ISCLOCKRUNNING", "null" created [main]
````

If I go on and create an internal IS1 sensor from the table, that also works:
````
     [java] 2016-02-07 15:12:43,160 managers.AbstractSensorManager        DEBUG - newSensor:IS1;null [AWT-EventQueue-0]
     [java] 2016-02-07 15:12:43,161 managers.AbstractSensorManager        DEBUG - Internal Sensor "IS1", "null" created [AWT-EventQueue-0]
````

but the InternalSensorManager wasn't registered for persistence when it was first created.

(As an aside, there are now two InternalSensorManager classes:  jmri.managers.InternalSensorManager and jmri.jmrix.internal.InternalSensorManager
Does this indicate some partial migration?

bobjacobsen, 2016-02-07:
In master HEAD, the InternalSensorManager is created quite early, but isn't registered:

````
     [java] 2016-02-07 15:30:16,775 apps.Apps                             INFO  - PanelPro version 4.3.3ish-201602072330-jake-Rbc3e5e7 starts under Java 1.8.0_25 at Sun Feb 07 15:30:16 PST 2016 [main]
     [java] 2016-02-07 15:30:22,486 apps.Apps                             INFO  - Starting with profile PanelPro.3ecefb62 [main]
     [java] 2016-02-07 15:30:22,569 managers.InternalSensorManager        DEBUG - InternalSensorManager ctor [main]
     [java] java.lang.Exception: traceback
     [java] 	at jmri.managers.InternalSensorManager.<init>(InternalSensorManager.java:16)
     [java] 	at jmri.managers.ProxySensorManager.makeInternalManager(ProxySensorManager.java:24)
     [java] 	at jmri.managers.AbstractProxyManager.getInternal(AbstractProxyManager.java:75)
     [java] 	at jmri.managers.AbstractProxyManager.getMgr(AbstractProxyManager.java:43)
     [java] 	at jmri.managers.AbstractProxyManager.addVetoableChangeListener(AbstractProxyManager.java:274)
     [java] 	at jmri.managers.AbstractTurnoutManager.<init>(AbstractTurnoutManager.java:24)
     [java] 	at jmri.managers.InternalTurnoutManager.<init>(InternalTurnoutManager.java:13)
     [java] 	at jmri.managers.ProxyTurnoutManager.makeInternalManager(ProxyTurnoutManager.java:29)
     [java] 	at jmri.managers.AbstractProxyManager.getInternal(AbstractProxyManager.java:75)
     [java] 	at jmri.managers.AbstractProxyManager.getMgr(AbstractProxyManager.java:43)
     [java] 	at jmri.managers.ProxyTurnoutManager.getValidOperationTypes(ProxyTurnoutManager.java:195)
     [java] 	at jmri.TurnoutOperationManager.loadOperationTypes(TurnoutOperationManager.java:159)
     [java] 	at jmri.TurnoutOperationManager.getInstance(TurnoutOperationManager.java:138)
     [java] 	at jmri.managers.AbstractTurnoutManager.<init>(AbstractTurnoutManager.java:23)
     [java] 	at jmri.managers.InternalTurnoutManager.<init>(InternalTurnoutManager.java:13)
     [java] 	at jmri.managers.ProxyTurnoutManager.makeInternalManager(ProxyTurnoutManager.java:29)
     [java] 	at jmri.managers.AbstractProxyManager.getInternal(AbstractProxyManager.java:75)
     [java] 	at jmri.managers.AbstractProxyManager.getMgr(AbstractProxyManager.java:43)
     [java] 	at jmri.managers.ProxyTurnoutManager.getClosedText(ProxyTurnoutManager.java:120)
     [java] 	at jmri.jmrit.beantable.RouteTableAction.<clinit>(RouteTableAction.java:1989)
     [java] 	at java.lang.Class.forName0(Native Method)
     [java] 	at java.lang.Class.forName(Class.java:260)
     [java] 	at apps.startup.StartupActionModelUtil.lambda$new$67(StartupActionModelUtil.java:26)
     [java] 	at apps.startup.StartupActionModelUtil$$Lambda$18/289639718.accept(Unknown Source)
     [java] 	at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(ForEachOps.java:183)
     [java] 	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
     [java] 	at java.util.HashMap$KeySpliterator.forEachRemaining(HashMap.java:1540)
     [java] 	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:512)
     [java] 	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:502)
     [java] 	at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(ForEachOps.java:150)
     [java] 	at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(ForEachOps.java:173)
     [java] 	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
     [java] 	at java.util.stream.ReferencePipeline.forEach(ReferencePipeline.java:418)
     [java] 	at apps.startup.StartupActionModelUtil.<init>(StartupActionModelUtil.java:24)
     [java] 	at apps.startup.AbstractActionModelFactory.initialize(AbstractActionModelFactory.java:53)
     [java] 	at apps.StartupActionsManager.<init>(StartupActionsManager.java:45)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
     [java] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
     [java] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
     [java] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:408)
     [java] 	at java.lang.Class.newInstance(Class.java:438)
     [java] 	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
     [java] 	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
     [java] 	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
     [java] 	at jmri.implementation.JmriConfigurationManager.<init>(JmriConfigurationManager.java:41)
     [java] 	at apps.Apps.<init>(Apps.java:220)
     [java] 	at apps.PanelPro.PanelPro.<init>(PanelPro.java:48)
     [java] 	at apps.PanelPro.PanelPro.main(PanelPro.java:110)
````

rhwood, 2016-02-08:
Are you sure it's not registered? I think you stack race is being triggered at the wrong point. What happens if you place that trace in ```jmri.managers.AbstractManager.registerSelf```? (The first three lines of the <=4.3.2 stack trace are within the construction of the InternalSensorManager (using, I think, an implicit super constructor).)

bobjacobsen, 2016-02-08:
The problem seems to be that the internal managers are being created before there's a configureManagerInstance.  This trace shows them being created early:
````
-normal-jmri-application:
     [echo] Launch normally (no debugger support)
     [java] 2016-02-07 23:45:00,415 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
     [java] 2016-02-07 23:45:01,024 util.Log4JUtil                        INFO  - This log is appended to file: /Users/jake/Library/Preferences/JMRI/log/messages.log [main]
     [java] 2016-02-07 23:45:01,025 util.Log4JUtil                        INFO  - This log is stored in file: /Users/jake/Library/Preferences/JMRI/log/session.log [main]
     [java] 2016-02-07 23:45:01,034 apps.Apps                             INFO  - PanelPro version 4.3.3ish-201602080744-jake-R22c7f99 starts under Java 1.8.0_25 at Sun Feb 07 23:45:01 PST 2016 [main]
     [java] 2016-02-07 23:45:03,974 apps.Apps                             INFO  - Starting with profile PanelPro.3ecefb62 [main]
     [java] 2016-02-07 23:45:04,041 managers.AbstractManager              DEBUG - registerSelf for config of type class jmri.managers.InternalTurnoutManager [main]
     [java] 2016-02-07 23:45:04,042 managers.AbstractManager              DEBUG - registerSelf for config of type class jmri.managers.InternalTurnoutManager [main]
     [java] 2016-02-07 23:45:04,044 managers.AbstractManager              DEBUG - registerSelf for config of type class jmri.managers.InternalSensorManager [main]
     [java] 2016-02-07 23:45:04,044 managers.InternalSensorManager        DEBUG - InternalSensorManager constructed [main]
     [java] 2016-02-07 23:45:04,105 anagers.DefaultUserMessagePreferences DEBUG - start load user pref file: /Users/jake/Library/Preferences/JMRI/PanelPro/UserPrefsProfileConfig.xml [main]
     [java] 2016-02-07 23:45:04,105 plementation.JmriConfigurationManager DEBUG - loading file:/Users/jake/Library/Preferences/JMRI/PanelPro/UserPrefsProfileConfig.xml ... [main]
     [java] 2016-02-07 23:45:04,196 implementation.SignalSpeedMap         DEBUG - _interpretation= 1 [main]
````

The code that's showing that is:
````
    protected void registerSelf() {
        log.debug("registerSelf for config of type {}", getClass());
        if (InstanceManager.configureManagerInstance() != null) {
            InstanceManager.configureManagerInstance().registerConfig(this, getXMLOrder());
            log.debug("registering for config of type {}", getClass());
        }
    }
````
so "registerSelf" is called, but it's not registering.  (These debug statements are slightly different from the earlier ones; in the prior traces, just the 2nd log.debug was present)

Apparently, those are now created before the configuration manager is created, or at least put in the InstanceManager.   That apparently happens before or during the read of the preferences.

N.B.:  The load/store tests don't detect this because they don't go through the full app initialization. Not sure whether we should have a JUnit test that catches this or not.

rhwood, 2016-02-08:
That's the hint we need. I need to delay initializing the StartupActionFactoryUtil object's list of actions

rhwood, 2016-02-08:
Hit comment too soon.
...StartupActionModelUtil objects list of actions until last possible moment (i.e. when the list is needed).

rhwood, 2016-02-08:
Thinking about this, I think delaying the creation of the list of actions in StartupActionModelUtil is merely masking bad behavior in some Action (I don't know which, just that it's one of the Actions in apps.ActionListBundle.properties) where the poorly behaved Action is initializing the InternalWhateverManager too soon.

bobjacobsen, 2016-02-08:
The internal SensorManager is created when the internal TurnoutManager (but it would have been any turnout manager) creates it's "operation" handlers, which in turn asks for the SensorManager, which in turn creates a proxy Sensor Manager.

The internal TurnoutManager is being created because RouteTableInstance has lines like:
```` Java
    private static String SET_TO_CLOSED = rbx.getString("Set") + " "
            + InstanceManager.turnoutManagerInstance().getClosedText();
    private static String SET_TO_THROWN = rbx.getString("Set") + " "
            + InstanceManager.turnoutManagerInstance().getThrownText();
````

That's probably getting the wrong text, because it's meant to get the language used by the default _real_ TurnoutManager, which is the layout's terminology, not the internal one. Doing that at class initialization time isn't going to be right. Fixed in #768

bobjacobsen, 2016-02-09:
It appears that #767 and #768 have doubly-fixed this in master. 

@kjlisby could you try this please?

closed 2016-02-14 by bobjacobsen

kjlisby, 2016-02-14:
Yes. It works.