JMRI/JMRI #771: Install to /usr/bin

jhdulaney, 2016-02-08:
I would like to package jmri for Fedora (and, RHEL/Centos, as well).  In order to do so, I will need to do a system-wide install (ie, /usr/bin).  Is this possible?  How would I go about doing a build from source to do this?

rhwood, 2016-02-08:
There is no need to build from source.

Extract the .tgz distribution into /usr/lib/jmri
Symlink DecoderPro, PanelPro, and SignalPro from that directory into /usr/bin
Optionally create .desktop files for those executables in /usr/share/applications (IIRC)
On Fedora, you may want to depend upon the rxtx RPM and remove it from /usr/lib/jmri/lib, since the Fedora version is identical to ours, but avoids the /var/lock issues; IIRC CentOS/RHEL do not have an rxtx RPM, so those OSes will need to use the included rxtx.
You may want to make the application scripts setgid for a group who can write to /var/lock.

You may want to include a udev rule set that sets permissions for USB and serial devices that connect to DCC systems; one is started at http://rhwood.github.com/paducah/devices/70-jmri.rules

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

jhdulaney, 2016-02-09:
Unfortunately, to meet packaging guidelines, I do have to build from source.

WHat I am doing to build is:

ant remakedecoderindex
ant dist
ant decoderpro
ant panelpro

Is there something I can do to then copy out the built parts?

rhwood, 2016-02-09:
"ant clean && ant package-linux" creates the tar file ; currently it's the only way to get a usable set of materials for distribution.

If that cannot be used, then you must write your own build target in ant to accomplish this. Note that since the guidelines state that no upstream jars can be included in the sources, it may be that JMRI can not be packaged within Fedora without writing your build target.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

pabender, 2016-02-09:
On Feb 8, 2016 7:04 PM, "John Dulaney" <notifications@github.com> wrote:
>
> Unfortunately, to meet packaging guidelines, I do have to build from
source.
>
> WHat I am doing to build is:
>
> ant remakedecoderindex
> ant dist
> ant decoderpro
> ant panelpro
>
> Is there something I can do to then copy out the built parts?
>

Those last two don't do what you think they do.  They run the decoderpro
and panelpro applications.

I would suggest you look at the release build instructions on the website (
I am on my phone right now, and don't have the link handy ).  The build
instructions will give you a guidline to follow in the process.

What would be really nice is if you can automate the process of building
the rpm and contribute that back to the project.  We could use that to
build rpms as part  of our release build process.

Paul

bobjacobsen, 2016-02-09:
> to meet packaging guidelines, I do have to build from source

Could you provide a link for those guidelines, please?  Thanks.

rhwood, 2016-02-09:
- https://fedoraproject.org/wiki/Packaging:Guidelines
- https://fedoraproject.org/wiki/Packaging:Java

jhdulaney, 2016-02-09:
Okay, ant package-linux was what I needed.  I do not, however, recall seeing that in the documentation?

Also, is there any way to specify the install location of, say, /usr/share/java/jmri.jar ?

jhdulaney, 2016-02-09:
Okay, never mind that last part, got that sorted.

jhdulaney, 2016-02-09:
So, a rough rpm spec file:

%define debug_package %{nil}
Name:    jmri
Summary: Java Model Railroad Interface
Version: 4.2.1
Release: 1%{?dist}
License: GPLv2
URL:     http://jmri.sourceforge.net/
Source0: http://github.com/JMRI/JMRI/archive/v%{version}.tar.gz

BuildRequires:  java
BuildRequires:  ant
BuildRequires:  xmvn-install

Requires:  java

%description
Jumping-off point for hobbyists who want to control their layouts
with a computer without having to create an entire system
from scratch.

%prep
%setup -q -n JMRI-%{version}

%build
ant clean
ant package-linux

%install
mkdir -p %{buildroot}%{_javadir}/jmri/
install -p -m 644 jmri.jar %{buildroot}%{_javadir}/jmri.jar
install -pd -m 444 xml/ %{buildroot}%{_javadir}/jmri/xml/
#cp -r jython// $RPM_BUILD_ROOT%{_libdir}/jmri/
#cp -r lib/ $RPM_BUILD_ROOT%{_libdir}/jmri/
#cp -r web/ $RPM_BUILD_ROOT%{_libdir}/jmri/
#cp -r resources/ $RPM_BUILD_ROOT%{_libdir}/jmri/
mkdir -p %{buildroot}%{_pkgdocdir}/
cp -r help/ %{buildroot}%{_pkgdocdir}/

%files
%doc %{_docdir}/jmri
%{_javadir}/jmri.jar

Anything y'all see here that I may be doing wrong?

I will gladly give y'all the spec file I wind up with.

pabender, 2016-02-09:
On 02/08/2016 10:44 PM, John Dulaney wrote:
> Anything y'all see here that I may be doing wrong?

I don't see any reference to the javadoc.

Paul

rhwood, 2016-02-09:
We don't normally ship JavaDocs, so it's a different ant target.

jhdulaney, 2016-02-09:
okay, got that sorted.

I have noticed, however, that one folder appears to not have any permissions set on any of its files:

jdulaney@gefjon:~/rpmbuild/BUILD$ ls -l JMRI-4.2.1/xml/
ls: cannot access JMRI-4.2.1/xml/config: Permission denied
ls: cannot access JMRI-4.2.1/xml/mfginfo: Permission denied
ls: cannot access JMRI-4.2.1/xml/DTD: Permission denied
ls: cannot access JMRI-4.2.1/xml/decoderIndex.xml: Permission denied
ls: cannot access JMRI-4.2.1/xml/XSLT: Permission denied
ls: cannot access JMRI-4.2.1/xml/catalog.xml: Permission denied
ls: cannot access JMRI-4.2.1/xml/defaultPanelIcons.xml: Permission denied
ls: cannot access JMRI-4.2.1/xml/demoOperations: Permission denied
ls: cannot access JMRI-4.2.1/xml/signals: Permission denied
ls: cannot access JMRI-4.2.1/xml/digitrax-cs-config.xml: Permission denied
ls: cannot access JMRI-4.2.1/xml/jinputDevices: Permission denied
ls: cannot access JMRI-4.2.1/xml/names.xml: Permission denied
ls: cannot access JMRI-4.2.1/xml/README: Permission denied
ls: cannot access JMRI-4.2.1/xml/decoders: Permission denied
ls: cannot access JMRI-4.2.1/xml/schema: Permission denied
ls: cannot access JMRI-4.2.1/xml/samples: Permission denied
ls: cannot access JMRI-4.2.1/xml/programmers: Permission denied
ls: cannot access JMRI-4.2.1/xml/COPYING: Permission denied
ls: cannot access JMRI-4.2.1/xml/layout: Permission denied
total 0
-????????? ? ? ? ?            ? catalog.xml
d????????? ? ? ? ?            ? config
-????????? ? ? ? ?            ? COPYING
-????????? ? ? ? ?            ? decoderIndex.xml
d????????? ? ? ? ?            ? decoders
-????????? ? ? ? ?            ? defaultPanelIcons.xml
d????????? ? ? ? ?            ? demoOperations
-????????? ? ? ? ?            ? digitrax-cs-config.xml
d????????? ? ? ? ?            ? DTD
d????????? ? ? ? ?            ? jinputDevices
d????????? ? ? ? ?            ? layout
d????????? ? ? ? ?            ? mfginfo
-????????? ? ? ? ?            ? names.xml
d????????? ? ? ? ?            ? programmers
-????????? ? ? ? ?            ? README
d????????? ? ? ? ?            ? samples
d????????? ? ? ? ?            ? schema
d????????? ? ? ? ?            ? signals
d????????? ? ? ? ?            ? XSLT

bobjacobsen, 2016-05-08:
Is there anything we can do to help with this?

rhwood, 2016-05-09:
> Is there anything we can do to help with this?
> 
Not easily. There is at least one jar we rely on that can't ship from Fedora (SerialIO.jar) because of a known non-free license.

Dylan-M, 2016-05-14:
Actually, the part about rxtx not being available for CentOS and RHEL isn't strictly true. It just appears in the EPEL (Extra Packages for Enterprise Linux) repository.  So you would need to make mention that your RPM requires that.

Another thing I would suggest is looking at FPM to make multi-distro packages:
https://github.com/jordansissel/fpm

Much easier than making SPEC files (As I can well attest to from prior experience packaging things for Fedora, CentOS, RHEL, Mandriva, and OpenSUSE) yourself.

pabender, 2016-05-14:
Sent from my iPad

On May 13, 2016, at 11:09 PM, Dylan Myers <notifications@github.com> wrote:

> Actually, the part about rxtx not being available for CentOS and RHEL isn't strictly true. It just appears in the EPEL (Extra Packages for Enterprise Linux) repository. So you would need to make mention that your RPM requires that.
> 
SerialIO.jar that Randall mentioned is not RxTx.
> 

This is a closed source library that we have a license to distribute.  If I am not mistaken, we use it primarily for handling the screwy baud rate of the Digitrax MS100.

Paul

Dylan-M, 2016-05-14:
@pabender I was responding to comments older than that, that did mention rxtx ;)

I know nothing about SerialIO, so that's a whole different ball of wax.

bobjacobsen, 2016-05-16:
@pabender is correct about SerialIO.  We have a license to distribute (for build time and run time) the SerialIO library.  It's used in the jmrix.loconet.ms100 package, and also in jmrix.direct, to handle the odd baud rates of the MS100.

There are a couple ways to handle a distribution that doesn't (can't) include a closed-source .jar:

- Reimplement a couple files to access the necessary classes via reflection, so that people can build without it, but will get an error if they try to configure to use the MS100
- Remove the MS100 support, including SerialIO, entirely.  Our licensed SerialIO only works on Windows up to and including Windows Vista; it doesn't work with more modern ones. The MS100 support on those Windows versions has to go at some point. (With the current SerialIO organization, I doubt we can get a license for the current version with acceptable terms in cost or restriction) This would also include removing it from jmrix.direct, which was never fully implemented anyway.
- Wait until we move away from RXTX (our other serial library, see #936) and combine removing SerialIO at the same time.

Dylan-M, 2016-05-16:
With that final option, what would replace it? From looking at that PR it looks like RXTX would be replaced by PureJavaComm. Can PJC handle what SerialIO is currently doing as well?

bobjacobsen, 2016-05-16:
> Can PJC handle what SerialIO is currently doing as well?

That's an experimental question.  PJC is supposed to be able to access the capability of the underlying system, so it should work for the MS100.  But it will take somebody with the MS100 hardware and suitable Windows system(s) to know for sure.

Dylan-M, 2016-05-16:
Okay, thanks for the information. I've never dealt with low-level IO in Java, so it's pretty much a complete mystery to me. Sadly, I don't use Digitrax, so I can't help you there. I'm Lenz for my home layout and NCE at my club.

KenC57, 2017-03-06:
Is the current status of this waiting for how we do with the PJC integration?? I was reviewing the progress and I think that's where this stopped. Now that we are getting around to real testing with the PJC, this might be worth looking at again.

closed 2021-04-14 by stale[bot]