JMRI/JMRI #773: Return of Travis Tests Crashing VM

bobjacobsen, 2016-02-09:
(This is the return of the previous #369 issue)

Travis is (again) crashing during ci-test, generally in the schema test section.

Debugging is complicated by inability to see logs (suggestions welcome) but it appears that the process is being "kill"ed for violating some system limit.  

There's a bit of recent discussion in #753 and #768.

mattharris, 2016-02-09:
Looks like we might not be the only ones:

https://github.com/travis-ci/travis-ci/issues/5597

bobjacobsen, 2016-02-09:
Still don't know what is causing this. Others are seeing it, so it might be a real bug somewhere.

- checked ulimit, only -s was even conceivably relevant, and setting huge made no change
- changing ant's fork type, including not forking, cleaned up messages but didn't stop crash
- ant is from the 1.8 series, old but not that old
- running with the ant java task, not the junit task, passed. 

Is ant provoking a crash?  Is it just that much bigger or slower to trip a limit? Don't know yet.

bobjacobsen, 2016-02-10:
The travis-ci/travis-ci#5597 shows that this is hitting others. It also had a clue:  Moving from the Travis container infrastructure to standalone VMs might fix it.  Perhaps the available memory in a container isn't well controlled?  Whatever. #775 is a possible fix.

rhwood, 2016-03-04:
travis-ci/travis-ci#5597 suggests that the real issue is that the containers are running tests with the memory setting ```Xmx32m``` do we want to perhaps force our own memory settings for the CI Tests?

bobjacobsen, 2016-03-04:
> travis-ci/travis-ci#5597 suggests that the real issue is that the containers are running tests with the memory setting Xmx32m do we want to perhaps force our own memory settings for the CI Tests?

The situation was confusing, because it looked like a limit problem, but running out of Java heap (usually?) gives different errors and ulimit was unlimited.

It probably wouldn’t hurt to experiment with putting Xmx in the Ant target.  We do that for other targets.  And it would be nice to get back to the other Travis configuration, which starts up faster.

rhwood, 2016-03-30:
Periodically updating the branch #1044 is based on suggests that this might be fixed by setting the JVM max ram reservation to something between 128 and 640 MB.

Failures in that branch appear to no more prevalent in that branch than in the master branch.

PR #775 (2016-02-10): Travis runs standalone
This changes the Travis YML control file to run the Travis CI tests in a standalone VM, instead of a container.  This _may_ fix #773.

Note: because the standalone VM is slower to start up, this form may take about 90 seconds more clock time to run.

No changes to Appveyor infrastructure.
closed 2016-06-06 by bobjacobsen