JMRI/JMRI #809: Recommended form for Logger definition

bobjacobsen, 2016-02-15:
(See threads in #806 and #787 for background; see also the ["JMRI: How to Use the Logging Facilities"](http://jmri.orgt/help/en/html/doc/Technical/Logging.shtml) page and the ["JMRI: Recommended Practices"](http://jmri.org/help/en/html/doc/Technical/RP.shtml) page.)

This issue is for discussion of our conventions for adding the logger definition to JMRI Java source files.

The current language is:

```
To log messages from a class named MyClass, add this to the bottom of the class's .java file:
    private static final Logger log = LoggerFactory.getLogger(MyClass.class());
and add imports for org.slf4j.Logger and org.slf4j.LoggerFactory in your imports section:
    import org.slf4j.Logger;
    import org.slf4j.LoggerFactory;
```

- The reference to "MyClass.class()", with the (), seems to be a typo.  The code contains "MyClass.class" and "MyClass.class.getName()". In #806, @rhwood recommends the "MyClass.class" form.

- Note "_private_ static final".  The code was recently bulk-translated to this (thanks @rhwood @DanielBoudreau), although the recommendation had been around for almost two years.  Lots of classes had logger definitions with protected access which weren't being used in the class.  Setting them private caused over 800 "unused" warnings (protected access doesn't trigger that because it's part of the class signature; it could be intended for use by a sub-class). Those definitions were removed, which removed the warnings and saved some memory and start-up time for those classes.

  Now, if somebody wants to add some logging to e.g. provide info on a problem, they have to add the Logger definition back.  That's just a one-line copy and paste (you have to remember the change the class name, but that's just part of the idiom, like adding a default copyright line or JavaDoc comment) for the logger definition itself, plus adding the import statements.

  I'd prefer the line to be:
```
private static final org.slf4j.Logger log = org.slf4j.LoggerFactory.getLogger(MyClass.class);
```
which is a bit longer, but doesn't require moving around in (sometimes very large) files to copy the imports, etc.  It also has the advantage that, if you comment it out should it be no longer needed (e.g. if you're done with temporary logging), there's no import statements left hanging.  But I don't think there's a strong reason to change the existing forms which use the import statements, either.

- Finally, @rhwood has suggested that Logger definitions should be at the front of files:
```
The coding conventions we follow stipulate that variable definitions are to be the first things in a class definition. I have been following those conventions when I make a major edit on a file, which is why the Logger is not defined at the bottom of the file.
```
The ["JMRI: Recommended Practices"](http://jmri.org/help/en/html/doc/Technical/RP.shtml) page does point to Java Coding Conventions that says (section 3.1.3) variables should be declared prior to method declarations.  But "log" isn't a _variable_; note the "private final static" definition.  It's a constant reference to an object (a tool, effectively), and it's not going to vary.  That makes the recommendation on the ["How to Use the Logging Facilities"](http://jmri.orgt/help/en/html/doc/Technical/Logging.shtml) page quite reasonable:  This is a consistent idiom in our code, putting it at the bottom gets it out of the way, etc.

So I propose making these changes:
 - fix the extra () in .class() in the existing recommendation
 - add to the recommendation that using fully-qualified names is OK too.
 - add a recommendation that it's OK to comment out a Logger definition if it's not being used (and it's also OK to remove it; @DanielBoudreau did a service by removing a few hundred of those that are unlikely to ever come back)
 - leave the recommendation on the ["How to Use the Logging Facilities"](http://jmri.orgt/help/en/html/doc/Technical/Logging.shtml) page to define the Logger at the bottom on the Java file, and add some "but we have local conventions, c.f. Logging(link)" language to the ["Recommended Practices"](http://jmri.org/help/en/html/doc/Technical/RP.shtml) page (are there other places where we have those? We could add them all at the same time)

Normally, I'd just put these in a PR, and we could discuss them there.  But this seems likely to engender enough discussion that starting an Issue seems better.

rhwood, 2016-03-10:
I have found that I cannot configure Eclipse, NetBeans, or IntelliJ to, when organizing fields, to leave just the one field ```log``` at the end of the class while placing all other private final static fields at the beginning of the class.

I have also found that although the CheckStyle tool can be configured to require that all fields be at the beginning of the class, it can also be configured to allow individual fields at different locations, but only by adding special comments to each file. (CheckStyle can be used for style checking in a CI environment).

bobjacobsen, 2016-03-10:
> I have found that I cannot configure Eclipse, NetBeans, or IntelliJ to, when organizing fields, to leave just the one field log at the end of the class while placing all other private final static fields at the beginning of the class.

Do those all do the same thing with "organizing fields"? In other words, is there a defacto standard there, or would applying different "organizing fields" tools move things around?

closed 2016-03-11 by bobjacobsen

rhwood, 2016-03-11:
Eclipse and NetBeans default to the same thing with "organizing code" (fields and methods); changes to this in NetBeans can be saved as property of the project, while changes to this in Eclipse only appear to be personal only.

PR #928 (2016-03-11): Documentation updates
Closes #809